simbody/simbody #331: undefined references to ~PIMPLHandle()

xantares, 2015-01-22:
seems some instanciated templates of PIMPLHandle are missing the export of ~PIMPLHandle:

Linking CXX shared library ../../libSimTKmath.dll
CMakeFiles/SimTKmath.dir/objects.a(CMAESOptimizer.cpp.obj): In function `~ParallelExecutor':
/home/xantares/projects/simbody/SimTKcommon/./include/SimTKcommon/internal/ParallelExecutor.h:65: undefined reference to `SimTK::PIMPLHandle<SimTK::ParallelExecutor, SimTK::ParallelExecutorImpl, false>::~PIMPLHandle()'
/home/xantares/projects/simbody/SimTKcommon/./include/SimTKcommon/internal/ParallelExecutor.h:65: undefined reference to `SimTK::PIMPLHandle<SimTK::ParallelExecutor, SimTK::ParallelExecutorImpl, false>::~PIMPLHandle()'
CMakeFiles/SimTKmath.dir/objects.a(ContactGeometry_TriangleMesh.cpp.obj): In function `ZNK5SimTK15ContactGeometry12TriangleMesh19createPolygonalMeshEv':
/home/xantares/projects/simbody/SimTKcommon/Geometry/include/SimTKcommon/internal/PolygonalMesh.h:76: undefined reference to `SimTK::PIMPLHandle<SimTK::PolygonalMesh, SimTK::PolygonalMeshImpl, true>::~PIMPLHandle()'
CMakeFiles/SimTKmath.dir/objects.a(ContactGeometry_TriangleMesh.cpp.obj): In function `~PolygonalMesh':
/home/xantares/projects/simbody/SimTKcommon/Geometry/include/SimTKcommon/internal/PolygonalMesh.h:76: undefined reference to `SimTK::PIMPLHandle<SimTK::PolygonalMesh, SimTK::PolygonalMeshImpl, true>::~PIMPLHandle()'
/home/xantares/projects/simbody/SimTKcommon/Geometry/include/SimTKcommon/internal/PolygonalMesh.h:76: undefined reference to `SimTK::PIMPLHandle<SimTK::PolygonalMesh, SimTK::PolygonalMeshImpl, true>::~PIMPLHandle()'
CMakeFiles/SimTKmath.dir/objects.a(ContactGeometry_TriangleMesh.cpp.obj): In function `ZN5SimTK13PolygonalMeshD1Ev':
/home/xantares/projects/simbody/SimTKcommon/Geometry/include/SimTKcommon/internal/PolygonalMesh.h:76: undefined reference to `SimTK::PIMPLHandle<SimTK::PolygonalMesh, SimTK::PolygonalMeshImpl, true>::~PIMPLHandle()'
CMakeFiles/SimTKmath.dir/objects.a(ContactGeometry_TriangleMesh.cpp.obj): In function `~PolygonalMesh':
/home/xantares/projects/simbody/SimTKcommon/Geometry/include/SimTKcommon/internal/PolygonalMesh.h:76: undefined reference to `SimTK::PIMPLHandle<SimTK::PolygonalMesh, SimTK::PolygonalMeshImpl, true>::~PIMPLHandle()'
CMakeFiles/SimTKmath.dir/objects.a(ContactGeometry_SmoothHeightMap.cpp.obj):/home/xantares/projects/simbody/SimTKcommon/Geometry/include/SimTKcommon/internal/PolygonalMesh.h:76: more undefined references to `SimTK::PIMPLHandle<SimTK::PolygonalMesh, SimTK::PolygonalMeshImpl, true>::~PIMPLHandle()' follow
collect2: error: ld returned 1 exit status
SimTKmath/sharedTarget/CMakeFiles/SimTKmath.dir/build.make:4301: recipe for target 'libSimTKmath.dll' failed
make[2]: *** [libSimTKmath.dll] Error 1
CMakeFiles/Makefile2:402: recipe for target 'SimTKmath/sharedTarget/CMakeFiles/SimTKmath.dir/all' failed
make[1]: *** [SimTKmath/sharedTarget/CMakeFiles/SimTKmath.dir/all] Error 2
Makefile:117: recipe for target 'all' failed

sherm1, 2015-01-22:
On what compiler version and platform is this failure occurring?

xantares, 2015-01-22:
crosscompiling for win32 on linux with mingw-gcc 4.9.2

sherm1, 2015-01-22:
Did you already try adding `SimTK_COMMON_EXPORT` to `PIMPLHandle` and `PIMPLImplementation` classes?

sherm1, 2015-01-22:
I just tried that and got a bunch of warnings like this:
```
ParallelExecutor.h(37): warning C4910: 'SimTK::PIMPLHandle<SimTK::ParallelExecutor,SimTK::ParallelExecutorImpl,false>' : '__declspec(dllexport)' and 'extern' are incompatible on an explicit instantiation (Parallel2DExecutor.cpp)
```
The `extern` it is referring to is an [extern template instantiation]
(http://www.stroustrup.com/C++11FAQ.html#extern-templates) that occurs [here](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/ParallelExecutor.h#L37). That construct intentionally suppresses implicit instantiation (which would fail due to hidden implementation), promising that instantiation will occur in a source file somewhere.

It appears that everything but the destructor is getting instantiated properly in your build (unless only the destructor is being called?). I wonder if there is a problem with handling of extern templates in the compiler you're using? If so maybe there is another way to accomplish this without extern templates.

sherm1, 2015-08-18:
I don't have a way to reproduce this on our supported platforms. If you have a fix that could be incorporated into the main repo, please provide a PR. I'm going to close this issue.

closed 2015-08-18 by sherm1