simbody/simbody #333: Need DecorativeMeshFile & visualizer support for it and other things

sherm1, 2015-01-29:
Other things: torus, arrow and/or cone, quadrants

Per opensim-org/opensim-gui#1

aseth1, 2019-01-08:
@aymanhab in [this commit](https://github.com/simbody/simbody/commit/3ddf3502d9b50c0a75eee846f69104121ab84171) "stubs" for `DecorativeArrow`, `DecorativeCone`, and `DecorativeTorus` were added but were never implemented. Now `OpenSim` visualization through the API cannot render all the `OpenSim::Geometry` that the user can add programmatically/scripting.

aymanhab, 2019-01-08:
Some of these are easy to do (Cone similar to Cylinder but with two radii) others will require major rework  since the meshes are cached by the visualizer. This will require upgrading simbody anyway. For the short term you can use a mesh file representing the shapes you need as a workaround.

sherm1, 2019-01-08:
Would definitely be nice to have those shapes in Simbody! Any volunteers?