simbody/simbody #343: Mechanism to invalidate Index member variables upon copy

chrisdembia, 2015-02-20:
Say you have a MeasureIndex as a member variable in your class. When you copy your class, the index should become invalid. This can be done by writing a copy constructor in which you invalidate the index. Sherm suggested a clean mechanism to do this automatically: create a class that will invalidate your index for you: InvalidateIndexOnCopy<T>, ReferenceIndex<T> (since the functionality is similar to ReferencePtr<T>), or InvalidateOnCopyIndex<T>.

sherm1, 2015-07-30:
This should get fixed by #359.

sherm1, 2015-08-18:
@chrisdembia, this issue should also have been fixed by PR #426. Please close if you agree.

chrisdembia, 2015-08-20:
I'll close the issue once I make use of the new mixin class for indices.

closed 2015-09-03 by chrisdembia