simbody/simbody #349: VectorIterator has broken assignment operator

sherm1, 2015-03-16:
The `VectorIterator` class keeps a reference to the `Vector` which prevents copy assignment from working. Nevertheless it contains a copy assignment operator which ends up making a deep copy of the Vector, rather then setting the new reference to be the same as the old one.

This is often a self-copy which is weird and may be slow but possibly harmless which is probably why this wasn't noticed. Valgrind caught it. I'm not sure if this causes any noticeable bugs.

sherm1, 2015-03-16:
Fixed in [this commit](https://github.com/simbody/simbody/commit/4abb5c83eda7ce228ae5451028f86ec0058cf0f4).

closed 2015-03-16 by sherm1

chrisdembia, 2015-03-16:
FYI IIRC github will close this issue even if you have `Fixed #issue` in your commit message.

sherm1, 2015-03-16:
Thanks -- actually I meant to push this to a separate branch and make a PR but my click finger got away from me and I ended up pushing it right into the master branch by accident. Didn't seem worth undoing it at that point.