simbody/simbody #350: Building from source on OSX with GNU compiler

nishbo, 2015-03-24:
For some reason, when building simbody with a gcc compilerinstead of a native one on OSX, unrecognized option `-stdlib=libc++` produces an error instead of a warning. This is a quick fix: replace line 270 in `CMakeList.txt` with:
```
            IF(${CMAKE_CXX_COMPILER_ID} MATCHES "GNU")
                SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
            ELSE()
                SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
                SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++")
            ENDIF()
```

scpeters, 2015-03-24:
Is this equivalent to what you mean?
~~~
diff --git a/CMakeLists.txt b/CMakeLists.txt
index f41a9a4..6c01822 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -268,7 +268,9 @@ IF(${CMAKE_CXX_COMPILER_ID} MATCHES "GNU" OR
             SET(CMAKE_XCODE_ATTRIBUTE_CLANG_CXX_LIBRARY "libc++")
         ELSE()
             SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
-            SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++")
+            IF(${CMAKE_CXX_COMPILER_ID} MATCHES "Clang")
+              SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++")
+            ENDIF()
         ENDIF()
     ELSE() # not APPLE
         SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
~~~

nishbo, 2015-03-24:
@scpeters Yes, it is. If `-stdlib=libc++` style is only used by Clang, than it is a solution.

scpeters, 2015-03-24:
`libc++` is used by clang (maybe others?) but gcc uses `libstdc++`

nishbo, 2015-03-24:
Typo?: gcc uses `-std=c++11`, not `-stdlib=libc++` on every system, I believe.

scpeters, 2015-03-24:
I was referring to its name, not the compiler argument: https://gcc.gnu.org/libstdc++/

chrisdembia, 2015-03-25:
Great catch. @nishbo would you be interested in submitting a PR for this?

nishbo, 2015-03-25:
Thanks @chrisdembia, hope I made correct PR.

nishbo, 2015-03-25:
Issue number in PR name did not get referenced.
https://github.com/simbody/simbody/pull/352

sherm1, 2015-06-03:
I'm assuming this is fixed now, by PR #352. @nishbo, please re-open if not.

closed 2015-06-03 by sherm1