simbody/simbody #359: Create ClearOnCopy mixin to clear indices upon copying

chrisdembia, 2015-04-10:
This came up in https://github.com/opensim-org/opensim-core/issues/398

@sherm1 said the following:

...A "mix in" template class that would replace a class's copy constructor with its default constructor and its copy assignment with a call to its clear() method. The data member could then be:
```c++
ClearOnCopy<SimTK::Measure> m_delayMeasure;
```
The implementation could be something like
```c++
template <class T> ClearOnCopy : public T {
    ClearOnCopy(const ClearOnCopy&) : T() {} // copy constructor does default construct
    ClearOnCopy& operator=(const ClearOnCopy&) {clear();} // copy assignment
    // or maybe:
    ClearOnCopy& operator=(const ClearOnCopy&) {
        static_cast<T>(*this) = T();
    }
};
```

I'm not sure that would be enough to make it useful though. Anything that would achieve the goal of not requiring a copy constructor to be written would be good.

aymanhab, 2015-04-10:
@chrisdembia  Most ModelComponents already keep an index into the system, why would that be different or their solution wouldn't work in this case?  is this issue specific to measures or a placeholder for a larger across-the-board issue?

chrisdembia, 2015-04-10:
I'm not sure about the scope of the issue. In #398, the issue is that if I create a copy of my ModelComponent, it contains indices that were appropriate for the System it came from but that are now meaningless, potentially causing unexpected behavior. This is more of a hypothetical issue, since the copied ModelComponent can't really be used until an initSystem call, in which case its index will be updated.

chrisdembia, 2015-07-30:
If we were to dynamically allocate handles, the new `NullOnCopyUniquePtr` class would solve this problem. But I don't think that's the right solution.

sherm1, 2015-07-30:
> If we were to dynamically allocate handles, the new NullOnCopyUniquePtr class would solve this problem. But I don't think that's the right solution.

I agree. Better to have something like `ClearOnCopy<MobilizedBodyIndex>` as above that would derive from `MobilizedBodyIndex` but override its copy & copy assignment methods. We would also have to add move methods to the index classes so that the copy methods won't be used in their place.

sherm1, 2015-07-30:
I see a few quirks with using ClearOnCopy; I'll take a crack at this unless you are already doing it Chris.

sherm1, 2015-07-30:
Also see #343; that should get fixed by this mechanism.

chrisdembia, 2015-07-30:
I am not already doing this.

sherm1, 2015-07-31:
Crap. I wrote a beautiful little mixin class but it wouldn't compile on VS2015 (works on gcc & clang). I filed a [bugreport](https://connect.microsoft.com/VisualStudio/feedback/details/1610850/inherit-constructors-doesnt-work-for-mixin-class-works-in-gcc-and-clang) with Microsoft.

For a good general-purpose `ClearOnCopy<T>` we want the class to behave just like T except when copying. So the basic mixin should look like this:
```c++
template <class T> class ClearOnCopy : public T {
public:
    using T::T;
    // other stuff
};
```
That `using T::T` is ANSI-sanctioned C++11 black magic for "inherit all the constructors of T". But in VS2015 when I instantiate like this:
```c++
ClearOnCopy< std::unique_ptr<int> > mp;
```
I get the error: `'T' : is not a member of 'std::unique_ptr<int,std::default_delete<_Ty>>`. I can't think of a good workaround for this -- any ideas?

/cc @klshrinidhi, @aseth1

klshrinidhi, 2015-07-31:
Like you mention in the bug report, I can confirm this is required by standard to work. The only neat solution I can think of is to `default` all the constructors except one that needs different behavior. This is the old-fashioned way.
I would like to try another (not so neat) solution based on the fact that they closed [this other bug](https://connect.microsoft.com/VisualStudio/feedback/details/988442) but I don't have the compiler. What if we write a struct like this:
```
template<typename T>
struct Identity : public T {};
```
and use it as the base class of `ClearOnCopy` like this:
```
template<typename T>
class ClearOnCopy : public Identity<T> {
    using Identity<T>::Identity;

    // Other stuff.
};
```
This compiles on gcc-5 and clang-3.6. Maybe worth a try.

sherm1, 2015-07-31:
Thanks, Shrinidhi. I saw that too and tried that approach. It didn't work either :angry:.

klshrinidhi, 2015-07-31:
Thats probably because the struct Identity in my example doesn't inherit *all* the constructors itself so its derivatives can't do much depending on it.
Too bad they don't support the language yet.

klshrinidhi, 2015-07-31:
Explicitly defaulting all the constructors will not get everything from the base class either, which is why the feature was probably added in the first place.
Until they completely support it, it is probably better to not have a generalized ClearOnCopy but write it for each case separately.

sherm1, 2015-07-31:
Oops. I wrongly accused VS2015 of feebleness. I have both 2013 and 2015 installed and they have identical GUIs. Turns out I had switched to 2013 and the constructor inheritance problem was there. I switched to 2015 and `using T::T;` works fine :innocent:. So we'll be able to use a good `ClearOnCopy<T>` as soon as we can switch over. Hopefully we can do that next week once AppVeyor is updated. I apologized to Microsoft and withdrew my bug report.

sherm1, 2015-07-31:
BTW, it occurs to me that we could use this class for built-in types too. I tried specializing it for pointers which works fine so these all behaved correctly:
```c++
ClearOnCopy<unique_ptr<T>> ownedThing; // like NullOnCopyUniquePtr<T>
ClearOnCopy<MobilizedBodyIndex>> referencedByIndex;
ClearOnCopy<Thing*> unownedReference; // same as ReferencePtr<Thing>
```
I think it could easily be extended to at least integral types like:
```c++
ClearOnCopy<bool> someFlag; // initialized to false on copy
ClearOnCopy<int> counter; // initialized to zero on copy
ClearOnCopy<int,-1> index; // initialized to -1 on copy
```
This would let us get rid of more explicit constructors since we wouldn't need them just to initialize and clear built in types. We should also make use of C++11's ability to initialize data members for builtins that have ordinary copy semantics.

sherm1, 2015-08-18:
Chris, `ResetOnCopy<MobilizedBodyIndex>>` from PR #426 should solve this problem. Please close this issue if you agree.

chrisdembia, 2015-08-20:
@sherm1 before I start updating OpenSim by making changes like `MobilizedBodyIndex` -> `ResetOnCopy<MobilizedBodyIndex>`: are there cases in which you *don't* want to reset the index upon copy? Should we have a variant of the macro that creates these index classes that employs the new mixin class?

sherm1, 2015-08-21:
> are there cases in which you don't want to reset the index upon copy? 

Yes, at least in Simbody. These classes are supposed to behave exactly like ints except that (a) they prevent type mixing, and (b) they have a recognizable illegal value (like a NaN), to which they initialize themselves. I would rather leave them that way and only reset them if there is an explicit ResetOnCopy on the data member.

chrisdembia, 2015-08-21:
Okay yeah I just realized also that in OpenSim, we make copies of indices *everywhere*. Lots of methods take indices by value or return them by value.

I'm actually kinda scared of using ResetOnCopy now; we have to be pretty careful.

sherm1, 2015-08-21:
> Lots of methods take indices by value or return them by value. I'm actually kinda scared of using ResetOnCopy now; we have to be pretty careful.

That shouldn't be any problem, Chris. It is only `ResetOnCopy<T>` that has the bizarre copy constructor. When you pass one to a `T` you are using the native behavior of class `T`.

chrisdembia, 2015-08-21:
Oh... k. You mean the following is fine (doens't reset inside foo)?

```
void foo(MyIndex);

ResetOnCopy<MyIndex> ix;
foo(ix);
```

sherm1, 2015-08-21:
> You mean the following is fine (doens't reset inside foo)?

Right! In order to get it to reset you would have to write:
```c++
void foo(ResetOnCopy<MyIndex>);

ResetOnCopy<MyIndex> ix;
foo(ix);
```

closed 2015-09-03 by chrisdembia