simbody/simbody #369: Generate a Vector using a linspace-like method, as one would do in MATLAB.

chrisdembia, 2015-05-05:
I'd like the following static method:

```cpp
static SimTK::Vector_<T>::linspace(const T& firstValue, const T& secondValue, int numValues)
```

or an analogous convenience constructor.

sherm1, 2015-05-05:
Sounds useful. I notice that Matlab `linspace` returns a row vector rather than a column vector. Is that a desirable feature for some reason? I suppose we should have methods in both `Vector_` and `RowVector_` classes. I'm assuming you mean for this method to return a `Vector_ as its functional return, like
```c++
template <class T> class SimTK::Vector_ {
    static Vector_ linspace(const T& firstValue, const T& secondValue, int numValues);
};
```
I don't think it should be done as a constructor because there are already way too many constructors with various barely-different signatures. With move construction there is no performance cost to these:
```c++
    Vector myVec(Vector::linspace(1.,10.,1000));
    Vector myVec = Vector::linspace(1.,10.,1000);
```

chrisdembia, 2015-05-05:
> Is that a desirable feature for some reason?

I don't know. Usually when I use it in matlab, I don't care if it's a row or column vector.

I agree with everything you said. I could do this too; seems straightforward enough.