simbody/simbody #381: Visualizer anti-aliasing

chrisdembia, 2015-06-03:
Antialiasing can be achieved with 2 lines

```
glutInitDisplay(... | GLUT_MULTISAMPLE);
glEnable(GL_MULTISAMPLE);
```

Here's before:

[image]

and after:
[image]

Is this worthwhile?

sherm1, 2015-06-03:
Sweet! Yes, that would be great especially for making nice stills.

It should be optional though, I think, because it is likely to have a large performance impact on complicated scenes. @peastman says it operates by generating multiple renderings of the same scene and combining them. It would be easy to add an antialiasing option to the Visualizer -- could even add it to the "View" button that is always present so it wouldn't have to be done programmatically.

peastman, 2015-06-03:
It's not exactly that it generates multiple renderings of the same scene.  It's more sophisticated than that.  But there is a performance cost.  Given the sorts of scenes you're rendering, though, that's likely to be a minor issue.  Modern GPUs are designed to handle *somewhat* more complicated scenes than the one shown above. :)

chrisdembia, 2015-06-03:
It wasn't obvious to me how to get Visualizer settings into simbody-visualizer.cpp:main, which is where I tried adding the new lines. I think having the option in the window is good but I think a programmatic option would be good as well.

sherm1, 2015-06-03:
> It wasn't obvious to me how to get Visualizer settings into simbody-visualizer.cpp:main

Maybe that's not necessary. What if in main you put:
```c++
glutInitDisplayMode(... | GLUT_MULTISAMPLE);
glDisable(GL_MULTISAMPLE);
shouldAntialias = false;
```
(I'm assuming that disabling multisample avoids the performance cost -- do you know, Peter?)

Then add an "Antialias" entry to the View menu, a toggle similar to the "Show shadows" entry (see MENU_SHOW_SHADOWS in the visualizer code). If a user selects that you could execute:
```c++
shouldAntialias = !shouldAntialias;
if (shouldAntialias) glEnable(GL_MULTISAMPLE);
else glDisable(GL_MULTISAMPLE);
```

If that works, programmatic control could be provided also, but it would be separate. You would add a new message to the many already supported by the VisualizerProtocol and have it execute the same code that the View menu item would do.

peastman, 2015-06-03:
I think that should be fine.  By the way, you can find a discussion of this antialiasing method at https://www.opengl.org/wiki/Multisampling.

chrisdembia, 2015-06-03:
Thanks both of you. @sherm1, if I test this on Linux and OSX can you test on Windows?

sherm1, 2015-06-03:
> By the way, you can find a discussion of this antialiasing method at https://www.opengl.org/wiki/Multisampling.

I noticed in there that they use `glEnable(GL_MULTISAMPLE_ARB)` rather than just `GL_MULTISAMPLE`. Not sure what that means. Also it would be good to check when multisample was added to OpenGL to make sure we don't accidentally prevent the visualizer from working if someone has an older OpenGL.

sherm1, 2015-06-03:
> if I test this on Linux and OSX can you test on Windows

Yes. I only have OpenGL 4.4 so can't check anything earlier. I *think* multisample was added in OpenGL 1.5 so we are probably safe for everything but raw Microsoft opengl with which the visualizer doesn't work anyway.

peastman, 2015-06-03:
According to https://stackoverflow.com/questions/4207506/where-is-gl-multisample-defined, it became a standard feature in 1.3.  Extensions start out being proposed by one vendor, and they have their own name for it.  Then if everyone agrees it's useful, the OpenGL Architecture Review Board standardizes it and gives it a name with the _ARB suffix.  This is sort of an intermediate state.  It's still an extension, so implementations don't have to include it.  But if they do, they're supposed to use the standard interface.  And then if it later becomes a core feature, they define a new name for it without the _ARB extension, but also keep supporting the old name.

sherm1, 2015-06-03:
Ah, thanks Peter. So no "_ARB" for us.

Chris, I would suggest just doing the View menu option since that is entirely local to the visualizer. That would be all that's needed for making nice pictures. You could leave the programmatic access until someone has a need for it.

chrisdembia, 2015-06-19:
This documentation might help: This page http://docs.gl/ might