simbody/simbody #389: Trying to build the examples separately using Simbody's static libraries generates linker errors

chrisdembia, 2015-06-08:
VS 2013
```
2>  ExampleKneeJoint.cpp
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??0MatrixStructure@SimTK@@QEAA@W4Structure@01@W4Position@01@W4DiagValue@01@@Z (public: __cdecl SimTK::MatrixStructure::MatrixStructure(enum SimTK::MatrixStructure::Structure,enum SimTK::MatrixStructure::Position,enum SimTK::MatrixStructure::DiagValue)) imported in function "public: static class SimTK::SplineFitter<double> __cdecl SimTK::SplineFitter<double>::fitFromGCV(int,class SimTK::Vector_<double> const &,class SimTK::Vector_<double> const &)" (?fitFromGCV@?$SplineFitter@N@SimTK@@SA?AV12@HAEBV?$Vector_@N@2@0@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??0MatrixStorage@SimTK@@QEAA@XZ (public: __cdecl SimTK::MatrixStorage::MatrixStorage(void)) imported in function "public: __cdecl SimTK::MatrixCommitment::Vector::Vector(void)" (??0Vector@MatrixCommitment@SimTK@@QEAA@XZ)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??0MatrixCommitment@SimTK@@IEAA@AEBVMatrixStructure@1@AEBVMatrixStorage@1@AEBVMatrixOutline@1@AEBVMatrixCondition@1@@Z (protected: __cdecl SimTK::MatrixCommitment::MatrixCommitment(class SimTK::MatrixStructure const &,class SimTK::MatrixStorage const &,class SimTK::MatrixOutline const &,class SimTK::MatrixCondition const &)) imported in function "public: __cdecl SimTK::VectorBase<double>::VectorBase<double>(int)" (??0?$VectorBase@N@SimTK@@QEAA@H@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??0State@SimTK@@QEAA@AEBV01@@Z (public: __cdecl SimTK::State::State(class SimTK::State const &)) imported in function main
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??1State@SimTK@@QEAA@XZ (public: __cdecl SimTK::State::~State(void)) imported in function main
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??1?$MatrixHelper@N@SimTK@@QEAA@XZ (public: __cdecl SimTK::MatrixHelper<double>::~MatrixHelper<double>(void)) imported in function "int `public: __cdecl SimTK::MatrixBase<double>::MatrixBase<double>(class MatrixBase<double>::MatrixCommitment const &,int,int,double const &)'::`1'::dtor$0" (?dtor$0@?0???0?$MatrixBase@N@SimTK@@QEAA@AEBVMatrixCommitment@1@HHAEBN@Z@4HA)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??0?$MatrixHelper@N@SimTK@@QEAA@HHAEBVMatrixCommitment@1@HH@Z (public: __cdecl SimTK::MatrixHelper<double>::MatrixHelper<double>(int,int,class SimTK::MatrixCommitment const &,int,int)) imported in function "public: __cdecl SimTK::MatrixBase<double>::MatrixBase<double>(class SimTK::MatrixCommitment const &,int,int)" (??0?$MatrixBase@N@SimTK@@QEAA@AEBVMatrixCommitment@1@HH@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??0?$MatrixHelper@N@SimTK@@QEAA@AEBVMatrixCommitment@1@AEBV01@AEBVDeepCopy@01@@Z (public: __cdecl SimTK::MatrixHelper<double>::MatrixHelper<double>(class SimTK::MatrixCommitment const &,class SimTK::MatrixHelper<double> const &,class SimTK::MatrixHelper<double>::DeepCopy const &)) imported in function "class SimTK::Vector_<double> __cdecl SimTK::operator*<double>(class SimTK::VectorBase<double> const &,double const &)" (??$?DN@SimTK@@YA?AV?$Vector_@N@0@AEBV?$VectorBase@N@0@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?copyAssign@?$MatrixHelper@N@SimTK@@QEAAAEAV12@AEBV12@@Z (public: class SimTK::MatrixHelper<double> & __cdecl SimTK::MatrixHelper<double>::copyAssign(class SimTK::MatrixHelper<double> const &)) imported in function "public: class SimTK::MatrixBase<double> & __cdecl SimTK::MatrixBase<double>::operator=(class SimTK::MatrixBase<double> const &)" (??4?$MatrixBase@N@SimTK@@QEAAAEAV01@AEBV01@@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?updElt@?$MatrixHelper@N@SimTK@@QEAAPEANHH@Z (public: double * __cdecl SimTK::MatrixHelper<double>::updElt(int,int)) imported in function "public: class SimTK::MatrixBase<double> & __cdecl SimTK::MatrixBase<double>::elementwiseAddScalarInPlace<double>(double const &)" (??$elementwiseAddScalarInPlace@N@?$MatrixBase@N@SimTK@@QEAAAEAV01@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?getElt@?$MatrixHelper@N@SimTK@@QEBAPEBNH@Z (public: double const * __cdecl SimTK::MatrixHelper<double>::getElt(int)const ) imported in function "public: double const & __cdecl SimTK::VectorBase<double>::operator[](int)const " (??A?$VectorBase@N@SimTK@@QEBAAEBNH@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?updElt@?$MatrixHelper@N@SimTK@@QEAAPEANH@Z (public: double * __cdecl SimTK::MatrixHelper<double>::updElt(int)) imported in function "public: double & __cdecl SimTK::VectorBase<double>::operator[](int)" (??A?$VectorBase@N@SimTK@@QEAAAEANH@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?fillWith@?$MatrixHelper@N@SimTK@@QEAAXPEBN@Z (public: void __cdecl SimTK::MatrixHelper<double>::fillWith(double const *)) imported in function "public: __cdecl SimTK::MatrixBase<double>::MatrixBase<double>(class SimTK::MatrixCommitment const &,int,int,double const &)" (??0?$MatrixBase@N@SimTK@@QEAA@AEBVMatrixCommitment@1@HHAEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?copyInByRowsFromCpp@?$MatrixHelper@N@SimTK@@QEAAXPEBN@Z (public: void __cdecl SimTK::MatrixHelper<double>::copyInByRowsFromCpp(double const *)) imported in function "public: __cdecl SimTK::MatrixBase<double>::MatrixBase<double>(class SimTK::MatrixCommitment const &,int,int,double const *)" (??0?$MatrixBase@N@SimTK@@QEAA@AEBVMatrixCommitment@1@HHPEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?scaleBy@?$MatrixHelper@N@SimTK@@QEAAXAEBN@Z (public: void __cdecl SimTK::MatrixHelper<double>::scaleBy(double const &)) imported in function "class SimTK::Vector_<double> __cdecl SimTK::operator*<double>(class SimTK::VectorBase<double> const &,double const &)" (??$?DN@SimTK@@YA?AV?$Vector_@N@0@AEBV?$VectorBase@N@0@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?nrow@?$MatrixHelper@N@SimTK@@QEBAHXZ (public: int __cdecl SimTK::MatrixHelper<double>::nrow(void)const ) imported in function "public: class SimTK::MatrixBase<double> & __cdecl SimTK::MatrixBase<double>::elementwiseAddScalarInPlace<double>(double const &)" (??$elementwiseAddScalarInPlace@N@?$MatrixBase@N@SimTK@@QEAAAEAV01@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?ncol@?$MatrixHelper@N@SimTK@@QEBAHXZ (public: int __cdecl SimTK::MatrixHelper<double>::ncol(void)const ) imported in function "public: class SimTK::MatrixBase<double> & __cdecl SimTK::MatrixBase<double>::elementwiseAddScalarInPlace<double>(double const &)" (??$elementwiseAddScalarInPlace@N@?$MatrixBase@N@SimTK@@QEAAAEAV01@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?nelt@?$MatrixHelper@N@SimTK@@QEBA_JXZ (public: __int64 __cdecl SimTK::MatrixHelper<double>::nelt(void)const ) imported in function "public: virtual double __cdecl SimTK::Function_<double>::Linear::calcValue(class SimTK::Vector_<double> const &)const " (?calcValue@Linear@?$Function_@N@SimTK@@UEBANAEBV?$Vector_@N@3@@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?getCharacterCommitment@?$MatrixHelper@N@SimTK@@QEBAAEBVMatrixCommitment@2@XZ (public: class SimTK::MatrixCommitment const & __cdecl SimTK::MatrixHelper<double>::getCharacterCommitment(void)const ) imported in function "class SimTK::Vector_<double> __cdecl SimTK::operator*<double>(class SimTK::VectorBase<double> const &,double const &)" (??$?DN@SimTK@@YA?AV?$Vector_@N@0@AEBV?$VectorBase@N@0@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?deleteRepIfOwner@?$MatrixHelper@N@SimTK@@QEAAXXZ (public: void __cdecl SimTK::MatrixHelper<double>::deleteRepIfOwner(void)) imported in function "class SimTK::Vector_<double> __cdecl SimTK::operator*<double>(class SimTK::VectorBase<double> const &,double const &)" (??$?DN@SimTK@@YA?AV?$Vector_@N@0@AEBV?$VectorBase@N@0@AEBN@Z)
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ??1Subsystem@SimTK@@QEAA@XZ (public: __cdecl SimTK::Subsystem::~Subsystem(void)) imported in function main
2>ExampleKneeJoint.obj : warning LNK4217: locally defined symbol ?setUseUniformBackground@System@SimTK@@QEAAAEAV12@_N@Z (public: class SimTK::System & __cdecl SimTK::System::setUseUniformBackground(bool)) imported in function main
[... cut ...]
2>ExampleKneeJoint.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl SimTK::Force::Custom::Implementation::realizeVelocity(class SimTK::State const &)const " (?realizeVelocity@Implementation@Custom@Force@SimTK@@UEBAXAEBVState@4@@Z)
2>ExampleKneeJoint.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) const SimTK::Force::Custom::Implementation::`vftable'" (__imp_??_7Implementation@Custom@Force@SimTK@@6B@) referenced in function "public: virtual __cdecl MyStop::~MyStop(void)" (??1MyStop@@UEAA@XZ)
2>C:\Users\chris\Documents\simbodyex-build\RelWithDebInfo\ExampleKneeJoint.exe : fatal error LNK1120: 27 unresolved externals
```

sherm1, 2015-06-08:
Oh, I think that's because any client that wants to use static libraries has to define the preprocessor symbol `SimTK_USE_STATIC_LIBRARIES`. Here is a [comment about it](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/common.h#L158).

If you compile with that defined does the problem go away?

chrisdembia, 2015-06-08:
Ohh thanks. I already tore down my files for testing this, so I'll just close this issue.

closed 2015-06-08 by chrisdembia