simbody/simbody #400: Building version 3.3.1 from source on Ubuntu 14.04 causing single test failure

ndhir, 2015-06-15:
When building simbody 3.3.1 from source (as required by OpenSim 3.2) one of the tests fails. 

Following these instructions https://github.com/simbody/simbody, and when it comes to running the tests via `ctest` (item three under the Linux **Build and Install** section) in the build directory, the following error message is given:

```
99% tests passed, 1 tests failed out of 97

Total Test time (real) =  17.55 sec

The following tests FAILED:
	 85 - TestCustomConstraints (Failed)
Errors while running CTest
 ```
The install directory is the default recommended `/usr/local/`.

sherm1, 2015-06-15:
Hi, Neil. I just tried downloading the simbody 3.3.1 zip from [here](https://github.com/simbody/simbody/releases/tag/Simbody-3.3.1) to my Ubuntu 14.04 (64 bit) VM and building it using all the default CMake settings, except I chose which compiler to use. I built it once with clang 3.4 and once with gcc 4.8.2. I got a few warnings during the build (different from each compiler) but the builds completed successfully. With either compiler the tests all passed. 

`TestCustomConstraints` is a touchy test and has caused some spurious failures in the past. So I suspect this OK. But it would be good to know how your build environment or toolchain differs from mine. Are you building 32 bit or using different compilers? Did you change any of the optimization settings or compiler flags? 

Sherm

ndhir, 2015-06-16:
Hi Sherm, 

I am building 64 bit on a 64 bit Ubuntu 14.04 LTS, same as yours I would imagine. 

Below is the info from my cmake Unix Makesfiles generator:

```
The C compiler identification is GNU 4.9.2
The CXX compiler identification is GNU 4.9.2
Check for working C compiler: /usr/bin/cc
Check for working C compiler: /usr/bin/cc -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working CXX compiler: /usr/bin/c++
Check for working CXX compiler: /usr/bin/c++ -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
Looking for gethostbyname
Looking for gethostbyname - found
Looking for connect
Looking for connect - found
Looking for remove
Looking for remove - found
Looking for shmat
Looking for shmat - found
Looking for IceConnectionNumber in ICE
Looking for IceConnectionNumber in ICE - found
Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
Found GLUT: /usr/lib/x86_64-linux-gnu/libglut.so  
Configuring done
```

sherm1, 2015-06-16:
The only difference I notice is gcc 4.9.2 in your build vs. gcc 4.8.2 in mine. I wonder if that is causing the difference in behavior. If so, it might be a harmless change in instruction selection affecting roundoff errors (meaning the test is too sensitive to trivia), or could be a bug, possibly in the 4.9.2 optimizer. One thing you could try is to reduce the optimization level from `-O3` to `-O2`, or try a Debug build to see if the test still fails.

moorepants, 2016-07-18:
I'm getting the same test failure when building with openblas on linux:

https://circleci.com/gh/conda-forge/staged-recipes/6288

Not sure what the fix is.

sherm1, 2016-07-18:
Hmmm. I see random numbers being used in that test :-(. That is a recipe for flakiness, so it is very possible this is just a flake.

chrisdembia, 2016-07-18:
The test could be modified to use the same seed to make it more reproducible.

wagglefoot, 2016-07-25:
Compiler version:

`g++ (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3
`

stefano2734, 2025-05-18:
Actual version 3.8 and Ubuntu 24.
So question. Here: actual bug or obsolete?