simbody/simbody #401: Matrix resizeKeep() should be smarter about space allocation

sherm1, 2015-06-18:
This gets used sometimes to build up a Matrix row by row or column by column. In that case it should act like std::vector and preallocate extra space when it runs out. Currently it just allocates what is asked for then copies the old data so naive use has O(n^2) complexity for adding n rows or columns.

@klshrinidhi ran into this for the DataTable implementation in OpenSim.

ianhamilton117, 2018-01-16:
@sherm1 Hi there, I'm new to Simbody and am looking for an issue to start contributing. Can I take this one?

chrisdembia, 2018-01-16:
Hey @ianhamilton117. No one is working on this issue currently, so if you think you know how to do this, we would welcome the contribution.

sherm1, 2018-01-17:
Agreed, thanks @ianhamilton117! Please be sure to include a unit test that can run in our automated builds, verifies that the code is working, and ensures that it continues to work after some future programmer mangles it.

chrisdembia, 2018-01-17:
@ianhamilton117 to ensure a swift review of your changes, you may want to look over [CONTRIBUTING.md](https://github.com/simbody/simbody/blob/master/CONTRIBUTING.md).

ianhamilton117, 2018-01-18:
Thanks @chrisdembia, will do.