simbody/simbody #406: Constructing a slider constraint, which enforces -a < body.x() < a

ndhir, 2015-06-30:
Hello!

I am trying to simulate the [inverted pendulum example](https://en.wikipedia.org/wiki/Inverted_pendulum), using a slider constraint. But I cannot seem to be able to figure out (I know very embarrassing) how to constrain the cart such that it is only allowed to move between say +/-1m from the origin ``Vec3(0)`` in the x-direction (as prescribed by the slider joint).

I.e. I want to constrain this construction:

```c++
    Body::Rigid cartInfo(MassProperties(1.0, Vec3(0), UnitInertia(1)));
    cartInfo.addDecoration(
        Transform(),
        DecorativeBrick(Vec3(0.2,0.1,0.1)).setColor(Red));

    // Create the linear slider object
    MobilizedBody::Slider cart(
        matter.Ground(), 
        Transform(Vec3(0)),
        cartInfo,
        Transform(Vec3(0.0,0.0,0.0)));
```

which has an inverted pendulum attached. 

Why do we want to do this? The real experimental cart does unfortunately not have an infinite slider.

Thanks

sherm1, 2015-06-30:
Hi, Neil. Please use the forum for support -- GitHub issues are for software problems and enhancement requests; they are seen by a very different audience. Regards, Sherm

closed 2015-06-30 by sherm1

ndhir, 2015-06-30:
Apologies.