simbody/simbody #409: Support for Contiuous Collision Detection (CCD)?

Skylion007, 2015-07-06:
So I am attempting to use simulate laminated robots in the robot simulation software Gazebo. So far, two engines DART and Simbody have produced the most accurate results for my simulation. Unfortunately, both physics engines fail on one key issue, they lack support (default) support for continuous collision detection. I was wondering if this physics currently has support for or plans on adding support for this feature within the next few weeks.

I have [linked](https://github.com/dartsim/dart/issues/426) the previous discussion on how this issue relates to DART to further illustrate the issues I am having with collision detection. If you have any ideas for potential workarounds, any help would be appreciated.

sherm1, 2015-07-06:
If you can use a half space for ground rather than a mesh, that would avoid any pass-through problem with ground. That only works for a flat ground surface, though. Otherwise use a thick object.

It occurs to me that you might be misdiagnosing this problem in Simbody though if you are using a mesh. In that case you are using Simbody's Elastic Foundation (Winkler) contact model which is *nothing* like a typical rigid contact mesh. Contact is represented essentially by springs located only at the center point of each mesh triangle, acting on the other body's triangles. If one mesh has large triangles and the other small ones, then you can easily get pass-through because the small triangles don't hit the center points. The problem in that case would not be fixed by CCD; it requires denser meshes.

Skylion007, 2015-07-06:
Ah, so you are saying it's because my meshes are too optimized? If that is the case, why does it pass through even when I scale the mesh to more or less match the width of the floor plane/block? Moreover, Gazebo generates the planes and boxes within the software itself, although I could easily import meshes to replace them. Furthermore, if it requires denser meshes why do "thicker" meshes work even when I only scale the Z axis of the mesh? As seen below in this example: 
[image] This mesh has triangles that are the exact same size as the problematic mesh on the faces normal to the plane. Yet it works, while the other mesh (below) does not. How would this scenario explain why the meshes are passing through each other in the second screenshot, but not the first? 
[image]

sherm1, 2015-07-06:
In that case the mesh is probably dense enough. If the thicker surface works is there a reason you can't use it?

Skylion007, 2015-07-06:
I can, but it's much lower fidelity. I am designing an exporter feature for popupCad that allows you to run physics simulations on laminated robot designs. Ideally, it would be nice if we could model the exported models with the appropriate thickness instead of having to hack a solution together that may make the model behave in a way the differs from it's behaves in reality. While it does not matter so much in this laminated model, in other much more complicated ones, it can potentially damage the fidelity of the simulation (through incorrectly modelling contact collisions within the own mesh and such). Here is a classic example where it could affect the simulation of even this simple mesh: 
[image]

Skylion007, 2015-07-06:
Any other ideas what might be causing it? Or a possible workaround besides changing the thickness of a mesh?

sherm1, 2015-07-06:
Oh, I see. Your robot really does have thin parts. If the meshes are already dense enough (I'm not sure how dense Gazebo is making them), then the only other workaround I can think of is to reduce the step size so you don't get pass-through between steps. But that will cost speed.

We are currently reworking Simbody's contact system entirely to add a rigid contact option in addition to its current compliant models. I hope to add pass through prevention also but at the moment it is probably a long way off.

Skylion007, 2015-07-06:
I tried reduce the step size, but it makes a simulation that should take seconds, take hours. Is there anyway to change the SI unit of Simbodies and have everything scale properly? That might be a proper work around, but I am worried the difference between the thickness and the width will make the simulation very difficult to visualize and possibly less accurate.

Skylion007, 2015-07-06:
Hmm, hold on I realized that Gazebo was actual defaulting to Dart because of some error I had after scaling the mesh. 

 what():  SimTK Exception thrown at MassProperties.h:538:
  Error detected by Simbody method Inertia::operator-=(): Diagonals of an Inertia matrix must be nonnegative; got 6.73672e-08,-3.50725e-08,-3.94697e-08.
  (Required condition 'd >= -SignificantReal' was not met.)

Which is odd because I left as the default Inertial Matrix and Gazebo was suppose to calculate the values (or use it's default ones).

You might be on to something with the collision sides, but it didn't seem to help when I scaled the mesh. I can try making a the floor smaller and seeing what happens.
[image]

Skylion007, 2015-07-06:
I solved this issue the same way I solved it on the Dart thread, alas without continuous collision detection, it will be difficult to have a model that will phase through the floor if it falls even one foot. So continuous collision detection would really help in this case.

sherm1, 2015-07-06:
> Diagonals of an Inertia matrix must be nonnegative; got 6.73672e-08,-3.50725e-08,-3.94697e-08.
> Which is odd because I left as the default Inertial Matrix and Gazebo was suppose to calculate the values.

That sounds like a Gazebo bug -- please report it to them; negative inertia diagonals are definitely wrong!

Skylion007, 2015-07-06:
Already done, I would really appreciate if you could try to implement Continuous Collision Detection soon though as it would dramatically increase the speed and stability of my simulation. Thank you for your help though.

sherm1, 2015-07-06:
OK. I'll keep this issue open until it is done.

hsu, 2015-07-06:
@Skylion007 Thank you for posting the issue, [here's a link to the Gazebo bug on negative inertia due to scaling](https://bitbucket.org/osrf/gazebo/issue/1650/scaling-a-mesh-in-use-the-parameter-in-sdf), p.s. I formatted the issue slightly more legible.