simbody/simbody #415: Add mesh collision support for Gazebo

Skylion007, 2015-07-24:
I am curious if you have an plans to add mesh support using Gazebo's API? I am in desperate need of a engine that supports closed loops as well as Simbody, but also support mesh collisions. Are there any plans to implement this any time soon? I noticed you added an STL reader last year, but I have not seen any activity on this front since.

sherm1, 2015-07-24:
Simbody does support mesh contact using the Elastic Foundation contact model (essentially a bed of springs), and I know Gazebo uses EF meshes for some contact using Simbody (blocks and cylinders at least). I don't know whether Gazebo also supports more general meshes when running with Simbody. But John @hsu will know.

xanderdunn, 2016-03-31:
I'm running into the same difficulty.  Using the simbody built into Gazebo 6.5, the models I've defined using a mesh for their collision structure don't actually produce any collisions.  They immediately fall through the ground plane.  

The [Gazebo API](https://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/group__gazebo__physics__simbody.html) does have types `gazebo::physics::SimbodyMesh` described as "Triangle mesh collision helper class" and `gazebo::physics::SimbodyMeshShape` described as "Triangle mesh collision."  This would seem to indicate potential support.  However, if [this issue](https://bitbucket.org/osrf/gazebo/issues/883/simbodycollision-doesnt-support-meshes-or) is up-to-date, then it sounds like Simbody does not yet support either mesh or polyline collisions, so this does not work.

There are also [examples](http://www.osrfoundation.org/wordpress2/wp-content/uploads/2015/04/roscon2014_scpeters.pdf) of very complex models being run in Gazebo with simbody, so it seems contradictory that there are no supported collisions in simbody with Gazebo other than cylinder and box.

sherm1, 2016-03-31:
@xanderdunn, I think this would be easy to add to Gazebo for closed meshes, since those can be used directly in Simbody's Elastic Foundation contact element. General meshes with holes aren't supported in Simbody. That said, I don't think anyone has yet modified Gazebo to ask Simbody to make EF contact surfaces. I'm sure they would appreciate a PR to do so, but short of that perhaps some pressure could get it on the to-do list?

xanderdunn, 2016-03-31:
It [turns out](https://twitter.com/scpeters/status/715648372359565313) that the Atlas model I referenced above used boxes for its feet and everything else was a joint, so this is inline with the lack of Gazebo+Simbody collision for meshes.

@sherm1 thanks a lot for your feedback.  I won't have the time to work on modifying Gazebo, but I'll mention it on their issue tracker.