simbody/simbody #422: SimTK::ThreadLocal crashes on thread-destruction cleanup

sherm1, 2015-08-03:
The `ThreadLocal<T>` class tries to maintain several file-scope static variables [here](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/ThreadLocal.h#L34), and also registers a destructor with pthreads via `pthread_key_create()`. That destructor ([cleanUpThreadLocalStorage()](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/ThreadLocal.h#L38)) references those file-scope statics. However, since that destructor doesn't get invoked until pthread end-of-program cleanup, the file-scope statics can get destructed first, causing `cleanUpThreadLocalStorage()` to throw an exception. This was observed in PR #414 when running on Windows (see discussion there).

/cc @thomasklau, @chrisdembia

sherm1, 2015-08-03:
Thomas and Chris, I can't see any reason for `ThreadLocal<T>` to maintain those static variables at all. It appears to me that it goes to a lot of trouble to create and destruct them, even with explicit locks, but then no method in the class provides access to them. So I think this could be fixed just by deleting all that junk. Can you see why it has to be there? @peastman do you remember why [these statics](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/ThreadLocal.h#L34) are present in `ThreadLocal<T>`?

thomasklau, 2015-08-03:
Do we even need `cleanUpThreadLocalStorage()`? Even with no call to `delete &system;` on Linux and Mac, `cleanUpThreadLocalStorage()` doesn't get called. When it does on Windows, a segfault occurs.

peastman, 2015-08-03:
It's been a long time since I looked at that code, but I believe it's necessary to ensure correct memory handling.  Suppose you create a ThreadLocal.  It then gets accessed from several threads, and each one creates a new value for it on the heap.  Still later, that ThreadLocal object gets deleted.  At that point, you need to make sure every value that was created for it on any thread gets deleted.  That's what's happening here:

```C++
        std::set<void*>& instances = keyInstances[key];
        for (std::set<void*>::const_iterator iter = instances.begin(); iter != instances.end(); ++iter) {
            instanceMap.erase(*iter);
            delete (T*) *iter;
        }
        keyInstances.erase(key);
```

It can also happen in a different way.  If a thread exits, that's when it should delete that thread's value for every ThreadLocal object.  That's what cleanUpThreadLocalStorage() is for.

sherm1, 2015-08-03:
Thanks, Peter. I think that's right. But it may not work because the destruction occurs on the main thread rather than on the thread that allocated the memory. The `pthread_key_delete()` documentation [here](http://pubs.opengroup.org/onlinepubs/009695399/functions/pthread_key_delete.html) says: "In general, there is no safe mechanism whereby an implementation could free thread-specific data at key deletion time."

So the `cleanUpThreadLocalStorage()` method is probably the only safe way to tidy up thread-local heap space. Currently rumor has it that that method isn't being invoked on Linux, which would be bad!

sherm1, 2015-08-03:
@thomasklau, how does C++11 thread local support deal with thread local cleanup? Maybe this should be the first thing you replace.

thomasklau, 2015-08-03:
http://stackoverflow.com/questions/6021273/how-to-allocate-thread-local-storage

See the second response.

sherm1, 2015-08-04:
So for either pthreads or C++11 thread_local there are two very distinct objects: the `thread_local<T>` object, typically created in the main thread, and a bunch of objects of type `T`, each allocated and used by one thread. Those latter objects have "thread scope", meaning their destructors are called only when the thread terminates (with destructors called in the same thread as allocation occurred). The former object (the `thread_local<T>`, or "key" in pthreads) is destructed when it goes out of scope. But that doesn't invoke destructors on any of the thread-local objects of type `T`.

chrisdembia, 2015-08-04:
I knew that `std::thread_local` is somewhat intended for global variables, but I wasn't sure if that was the caase for `SimTK::ThreadLocal`.

sherm1, 2015-08-04:
> I knew that std::thread_local is somewhat intended for global variables, but I wasn't sure if that was the case for SimTK::ThreadLocal.

(I learned yesterday that `thread_local` is a C++11 storage-duration specifier keyword analogous to `static` -- it is not an `std::` class.)

`thread_local` is usable for namespace (global) scope or block-scope variables or static data members of classes. That is slightly different from `SimTK::ThreadLocal` which doesn't have to be a static data member. I don't think that will matter in practice but I'm not sure yet.

chrisdembia, 2015-08-04:
Oh oops; thanks for informing.

PR #423 (2015-08-05): Fix ThreadLocal so it doesn't try to destruct thread objects from the main thread
Fixes #422. Addresses a problem discovered during review of PR #414.

@thomasklau, @chrisdembia please review. I marked the class as deprecated in the class documentation but I didn't make it generate warnings. That should wait until Thomas reimplements it in terms of `thread_local` and then we can replace all uses of it in Simbody to avoid those warnings.

I got rid of all the unnecessary "static" infrastructure the class had before.
closed 2015-08-05 by sherm1