simbody/simbody #424: Trouble building with cmake (native and via IDE) in OS X Yosemite (10.10.4)

ndhir, 2015-08-10:
Greetings, 

I've been experimenting a bit with the install options now available with simbody. I ended up trying with the install version via `brew install simbody`. So far so good. 

I tried running the basic pendulum example 

```cpp
// Simulate and visualise a 2-link chain:
#include "Simbody.h"

using namespace SimTK;
using std::cout; using std::endl;

class PositionReporter : public PeriodicEventReporter {
public:
    PositionReporter(const MultibodySystem& system, const MobilizedBody& body, Real interval) :
        PeriodicEventReporter(interval), system(system), body(body) {
    }

    // Show x-y position of the pendulum weight as a function of time.
    void handleEvent(const State& state) const {
        system.realize(state, Stage::Position);
        Vec3 pos = body.getBodyOriginLocation(state);
        std::cout<<state.getTime()<<"\t"<<pos[0]<<"\t"<<pos[1]<<std::endl;
    }
private:
    const MultibodySystem& system;
    const MobilizedBody& body;
};

int main() {
try{// If anything goes wrong, an exception will be thrown.

    // Define the system.
    MultibodySystem system;
    SimbodyMatterSubsystem matter(system);
    GeneralForceSubsystem forces(system);
    Force::Gravity gravity(forces, matter, -YAxis, 9.8);

    // Describe mass and visualization properties for a generic body.
    Body::Rigid bodyInfo(MassProperties(1.0, Vec3(0), UnitInertia(1)));
    bodyInfo.addDecoration(Transform(), DecorativeSphere(0.1));

    // Create the moving (mobilized) bodies of the pendulum.
    MobilizedBody::Pin pendulum1(
        matter.Ground(), 
        Transform(Vec3(0)),
        bodyInfo,
        Transform(Vec3(0, 1.0, 0)));

    // MobilizedBody::Pin pendulum2(
    //     pendulum1, 
    //     Transform(Vec3(0)),
    //     bodyInfo, 
    //     Transform(Vec3(0, 0.5, 0)));

    // Set up visualization.
    Visualizer viz(system);
    system.addEventReporter(new Visualizer::Reporter(viz, 0.01));

    // Add event reporter
    system.addEventReporter(new PositionReporter(system, pendulum1, 0.5));

    // Initialize the system and state.
    State state = system.realizeTopology();
    pendulum1.setRate(state, 1.5); // The initial angular velocity of pendulum: ms^-1

    // Simulate for n seconds.
    RungeKuttaMersonIntegrator integ(system);
    TimeStepper ts(system, integ);
    ts.initialize(state);
    ts.stepTo(50.0);
} 
catch (const std::exception& e) {
    std::cout << "ERROR: " << e.what() << std::endl;
    return 1;
} 
catch (...) {
    std::cout << "UNKNOWN EXCEPTION\n";
    return 1;
}
    return 0;
}
```

using both CLion and stand-alone CMake 3.3.0 with simbody 3.5.1. I am consistently running into troubles with, what I believe, is the `SimbodyConfig.cmake` file.

CLion returns as such:

```
Error:The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Simbody_INCLUDE_DIR
   used as include directory in directory /Users/path/to/project/files/Pendulum
```

Similiarly the cmake-gui comes up with something similar:

```
The C compiler identification is AppleClang 6.1.0.6020053
The CXX compiler identification is AppleClang 6.1.0.6020053
Check for working C compiler using: Xcode
Check for working C compiler using: Xcode -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler using: Xcode
Check for working CXX compiler using: Xcode -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
/usr/local;/Developer
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.3/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Simbody (missing: Simbody_INCLUDE_DIR)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.3/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindSimbody.cmake:271 (find_package_handle_standard_args)
  CMakeLists.txt:18 (find_package)

Configuring incomplete, errors occurred!
See also "/Users/path/to/project/files/Pendulum/build/CMakeFiles/CMakeOutput.log".
```

My `CMakeLists.txt` looks as such:

```
# CMakeLists.txt for making a Simbody-using executable.
# This shows how to use the provided SimbodyConfig.cmake to locate a Simbody
# installation on your machine so you can use it from your own code.

cmake_minimum_required(VERSION 2.8)
project(Pendulum)

# Main source (and) header files
set(my_source_files Pendulum.cpp)
#set(my_header_files myexe.h)

# This depends on SimbodyConfig.cmake being located somewhere predictable
# on your machine. If you have installed it somewhere that CMake won't be
# able to guess, you'll need to tell find_package where to look.
list(APPEND CMAKE_MODULE_PATH "${CMAKE_SOURCE_DIR}/cmake")
find_package(Simbody REQUIRED)

include_directories(${Simbody_INCLUDE_DIR})
link_directories(${Simbody_LIB_DIR})

add_executable(Pendulum ${my_source_files})
target_link_libraries(Pendulum ${Simbody_LIBRARIES})
```

chrisdembia, 2015-08-10:
@ndhir I'm happy to hear you're using CLion; I like CLion as well.

It seems that CMake is electing to use the FindSimbody.cmake (module file) file instead of SimbodyConfig.cmake (config file); what if in your `find_package(Simbody REQUIRED)` line you force the use of the config file instead of the module file: `find_package(Simbody REQUIRED CONFIG)` (see http://www.cmake.org/cmake/help/v3.0/command/find_package.html)?

ndhir, 2015-08-10:
Hey @chrisdembia 

It is pretty sweet one must admit!

So I tried what you suggested, not much luck I am afraid:

```
# CMakeLists.txt for making a Simbody-using executable.
# This shows how to use the provided SimbodyConfig.cmake to locate a Simbody
# installation on your machine so you can use it from your own code.

cmake_minimum_required(VERSION 2.8)
project(InvertedPendulum)

# Main source (and) header files
set(my_source_files Pendulum.cpp)
#set(my_header_files myexe.h)

# This depends on SimbodyConfig.cmake being located somewhere predictable
# on your machine. If you have installed it somewhere that CMake won't be
# able to guess, you'll need to tell find_package where to look.
list(APPEND CMAKE_MODULE_PATH "${CMAKE_SOURCE_DIR}/cmake")
find_package(Simbody REQUIRED CONFIG)

include_directories(${Simbody_INCLUDE_DIR})
link_directories(${Simbody_LIB_DIR})

add_executable(Pendulum ${my_source_files})
target_link_libraries(Pendulum ${Simbody_LIBRARIES})
```

Can you make simbody examples work on your CLion IDE?

scpeters, 2015-08-10:
Is the prefix where you installed simbody (ie `/usr/local`) in your `CMAKE_PREFIX_PATH` for CLion?

chrisdembia, 2015-08-10:
> Can you make simbody examples work on your CLion IDE?

I haven't tried, but OpenSim works with Simbody, and OpenSim is just a huge Simbody example.

I'm confused though..is this a CLion issue? Can we first investigate stand-alone CMake?

chrisdembia, 2015-08-10:
I'm not convinced that CMake started using the SimbodyConfig.file. Can you start fresh (and as @scpeters said, set `CMAKE_PREFIX_PATH`)?

chrisdembia, 2015-08-11:
Neil, I just used homebrew to install simbody 3.5.3 and I used the SampleCMakeLists without changing anything related to the `find_package` line. CMake (3.3.0) automatically found the simbody that homebrew installed, since it's in `/usr/local`.

I think your `list(APPEND CMAKE_MODULE_PATH "${CMAKE_SOURCE_DIR}/cmake")` line might be the culprit. If not, could you try simbody 3.5.3?

Btw, I'm curious, how did you get simbody 3.5.1 from homebrew? Did you get it a while ago? I'm not too familiar with how homebrew works and I'm curious about how one gets old versions of a package...seems like the `brew versions` feature was removed.

ndhir, 2015-08-11:
I am not sure why it threw up so many errors to start with but I managed to build it with the pure cmake using your approach, i.e. do nothing. Now question is why it refuses to work on CLion (I am getting the same error as before).

Shoot; that was a typo, I never got 3.5.1, I used 3.5.3 as well but I want to get 3.5.1 so I opened a new issue here regarding this: https://github.com/osrf/homebrew-simulation/issues/44

chrisdembia, 2015-08-11:
> Now question is why it refuses to work on CLion (I am getting the same error as before).

Even when you remove the `CMAKE_MODULE_PATH` line? Can you try starting completely fresh?

ndhir, 2015-08-11:
Yeah even when I removed that; it will not budge, still the same issue. 

```
Error:The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Simbody_INCLUDE_DIR
   used as include directory in directory /path/to/project/folder
```

chrisdembia, 2015-08-11:
I really think that error message means you're using the module file.

Are the other variables set? In your project, after the `find_package` line, try `message("HELLO ${Simbody_LIB_DIR}") etc.

chrisdembia, 2015-08-11:
These are the variables that should be defined: https://github.com/simbody/simbody/blob/master/cmake/SimbodyConfig.cmake.in#L4

ndhir, 2015-08-12:
I'm just confused now. I wrote this now as a new test:

```
#include <iostream>
#include "Simbody.h"

using namespace std;
using namespace SimTK;

int main() {
    cout << "Hello, World!" << endl;
    return 0;
}
```

with `CMakeLists.txt` as

```
cmake_minimum_required(VERSION 3.2)
project(HelloTest)

# Simbody
find_package(Simbody REQUIRED)
include_directories(${Simbody_INCLUDE_DIR})
link_directories(${Simbody_LIB_DIR})

set(SOURCE_FILES main.cpp)
add_executable(HelloTest ${SOURCE_FILES})
target_link_libraries(HelloTest ${Simbody_LIBRARIES})
```
and it returns 

```
/Users/user/Library/Caches/clion10/cmake/generated/c1d0f54d/c1d0f54d/Debug/HelloTest
Hello, World!

Process finished with exit code 0
```
but I cannot for my life understand why that did not work before. Nothing changed except for the required cmake version.

chrisdembia, 2015-08-12:
Well I'm happy something is working. I think it's unlikely to be the version number. Perhaps corrupted cache files?

closed 2015-08-12 by ndhir

chrisdembia, 2015-08-12:
@ndhir let me know if you have suggestions for how we could improve documentation, etc.

ndhir, 2015-08-12:
Certainly! Though it would seem to me this must just have been some quirk of CLion that sorted itself out. 

I am happy to contribute some stuff, once I've got my current project up and running.

chrisdembia, 2015-08-12:
I've had some trouble with using cmake in CLion. I don't like how they try to manage cmake for you...I especially don't like that they don't really let you choose the build directory.

ndhir, 2015-08-12:
Aha, but if you go to: `CLion>Preferences>Build, Execution, Deployment>CMake`I believe you can edit that.