simbody/simbody #439: Simbody and CMake changes needed to build using MinGW on Windows

sherm1, 2015-09-22:
@gjacquenot reported success in building Simbody with MinGW and sent an email summarizing his changes; see comment below.

It would be great to get a pull request for the necessary changes.

sherm1, 2015-09-22:
From: Guillaume Jacquenot

I succeed in compiling Simbody with MinGW 4.9 64 bits with posix threads and seh exceptions handling. The main things I have done
- Replaced the dll in the platform dependent directory with mines (those found in MinGW/bin directory). I did so since they are many versions of MinGW with different threads mechanism and exceptions handling. This step can be automated in CMake (See below)
- Removed the header files in the platform dependent directory (MINGW alreay provides everything needed)
- Fixed the EXPORT macros in common.h files to remove the `__declspec` instructions with MINGW (See below)

For the visualizer, I made the following changes
- Made use of freeglut
- Fixed one command in CMakeLists.txt that performed copy and are bat specific

I am not sure all the changes I made are needed, so for instance I do not plan to commit these. I want to look at my problem, that is using SimTK as an external library to evaluate contact effort!

MSYS specific CMake instructions.
```
1) DLL copy
One could copy all the dll located in MinGW/bin
IF(MSYS)
    GET_FILENAME_COMPONENT(MINGW_BIN_PATH ${CMAKE_CXX_COMPILER} PATH)
    MESSAGE(STATUS "MINGW_BIN_PATH = ${MINGW_BIN_PATH}")
    LIST(APPEND MINGW_ADDITIONAL_LIB "${MINGW_BIN_PATH}/libgcc_s_dw2-1.dll")
    LIST(APPEND MINGW_ADDITIONAL_LIB "${MINGW_BIN_PATH}/libstdc++-6.dll")
    LIST(APPEND MINGW_ADDITIONAL_LIB "${MINGW_BIN_PATH}/libgfortran-3.dll")
    LIST(APPEND MINGW_ADDITIONAL_LIB "${MINGW_BIN_PATH}/libwinpthread-1.dll")
    IF(CMAKE_SIZEOF_VOID_P EQUAL 8) # If on a 64 bit machine
        LIST(APPEND MINGW_ADDITIONAL_LIB "${MINGW_BIN_PATH}/libgcc_s_seh-1.dll")
    ENDIF()
    FOREACH(f ${MINGW_ADDITIONAL_LIB})
        IF(EXISTS ${f})
            INSTALL(FILES ${f} DESTINATION ${LIBRARY_OUTPUT_DIRECTORY})
        ENDIF()
    ENDFOREACH()
ENDIF()

2) Example of modification done for EXPORT macros
    #if defined(SimTK_SimTKCOMMON_BUILDING_SHARED_LIBRARY)
        #ifdef _MSC_VER
        #define SimTK_SimTKCOMMON_EXPORT __declspec(dllexport)
        /* Keep MS VC++ quiet when it tries to instantiate incomplete template classes in a DLL. */
        #pragma warning(disable:4661)
        #else
        #define SimTK_SimTKCOMMON_EXPORT
        #endif
    #elif defined(SimTK_SimTKCOMMON_BUILDING_STATIC_LIBRARY) || defined(SimTK_USE_STATIC_LIBRARIES)
        #define SimTK_SimTKCOMMON_EXPORT
    #else
        #ifdef _MSC_VER
        #define SimTK_SimTKCOMMON_EXPORT __declspec(dllimport) /*i.e., a client of a shared library*/
        #else
        #define SimTK_SimTKCOMMON_EXPORT
        #endif
    #endif
```

sherm1, 2015-09-22:
@gjacquenot, a PR would be most welcome when you have a chance. The benefit to you would be that future versions of Simbody would then contain your changes so you wouldn't have to do them again if you want to upgrade. And of course the community would benefit since there are likely to be other Simbody users who would like to use MinGW.

Gjacquenot, 2015-09-28:
I have created a fork of simbody where I added compilation support for MinGW
https://github.com/Gjacquenot/simbody

The main modifications are the updates of the EXPORT macros to disable __declspec(dllexport) and __declspec(dllimport)

Compilation works fine, however I encounter crash when running tests.

These crashes occur when using the DLL provided in the platform specific directory (At least Lapack).
If using my DLLs generated with my version of GCC and the MinGW DDL, I get rid of these crashes and everything works as a charm.

For the MinGW DLL, one can use the ones located in the MINGW bin directory.
But for the Lapack DLL, I am wondering how we could do to use user specified libraries. The ideal would be to give a path to Lapack libraries with a variable like `LAPACK_ROOT` or `LAPACK_DIR`. Or use the `BULD_USING_OTHER_LAPACK` variable?

Below is a log of gdb, showing that crash occurs in dgetrf_ for `testMatInverse`

    $ gdb MatVecTest.exe 
    GNU gdb (GDB) 7.8.1
    This GDB was configured as "x86_64-w64-mingw32".
    Reading symbols from MatVecTest.exe...done.
    (gdb) r
    Starting program: d:\Z\simbody_gj_build\MatVecTest.exe 
    [New Thread 2860.0x16a4]
    Starting test MatVecTest ...
    ...
    ...
    ...
    ...
    ...
      testMatInverse ...
    
    Program received signal SIGSEGV, Segmentation fault.
    0x000000006627e560 in dgetrf_ () from d:\Z\simbody_gj_build\libSimTKcommon.dll
    (gdb) bt
    #0  0x000000006627e560 in dgetrf_ ()
       from d:\Z\simbody_gj_build\libSimTKcommon.dll
    #1  0x00000000662f4067 in SimTK::RegularFullHelper<double>::lapackInvertInPlace() () from d:\Z\simbody_gj_build\libSimTKcommon.dll
    #2  0x00000000662ad870 in SimTK::MatrixHelper<double>::invertInPlace() ()
       from d:\Z\simbody_gj_build\libSimTKcommon.dll
    #3  0x000000000040480b in testMatInverse() ()
    #4  0x0000000000419d44 in main ()
    (gdb) q
    A debugging session is active.
    
            Inferior 1 [process 2860] will be killed.
    
    Quit anyway? (y or n) [answered Y; input not from terminal]

sherm1, 2015-09-28:
Have you tried `BULD_USING_OTHER_LAPACK `? The purpose of that is to allow use of a non-default lapack and blas. I am not sure it allows a full pathname though; it may have to be combined with the Windows `PATH` variable to find libraries with given names.

It is surprising to me though that the Simbody-supplied lapack and blas don't work. Those were created with MinGW.

Gjacquenot, 2015-09-28:
Well, I made a test with MinGW 32 bits GCC/G++ 5.2.0 and test `MatVecTest` succeeds, however it fails on MinGW 64 bits GCC/G++ 4.9.2. 
I have to check the output of the command `g++ --version` on all platforms...

On the 32 bit platform, I have the following output:

`g++.exe (i686-posix-sjlj-rev0, Built by MinGW-W64 project) 5.2.0`

sherm1, 2015-09-28:
I wonder if this could be a problem of mixing 64- and 32-bit libraries. Simbody provides lapack/blas DLLs in both sizes (see Platforms/Windows/lib_x64 and /lib_x86). These were produced using MinGW 64 and MinGW 32 respectively.

Gjacquenot, 2015-09-28:
I will investigate tomorrow and look at the link command to see which one is used... I will also look at differences between the x64 lapack libraries (the one I have generated and the one provided).
I am happy to see that the fork still compiles with MSVC(hopefully ;-))

chrisdembia, 2015-09-28:
@Gjacquenot If you want to save yourself some effort, you can use Simbody's automated testing of Simbody on MSVC through Appveyor when/if you open a pull request.

Gjacquenot, 2015-09-29:
@chrisdembia : Thanks for the tip: I have tried it and my modifications do not work for MSVC! This is in fact not so trivial!

Apparently one of my commit creates linkage error on MSVC
I guess this is the following commit
https://github.com/Gjacquenot/simbody/commit/af390df2a3271d1cc40a6ceefe4f65dd48f25a7d
This commit should enable MSVC and MinGW compilation and linkage. I thought it was working on both platform. I remember testing it on MSVC successfully, but something must have gone wrong.

When I will have fixed it and look at the DLL issues, I will open a pull request.

The log error from AppVeyor is here:
https://ci.appveyor.com/project/Gjacquenot/simbody/build/1.0.4

As I don't know if it is public or not, I put the log message

    Build started
    Fetching repository commit (515685b)...OK
    Total: 67.7 MB in 1112 files
    mkdir build
    cd build
    cmake .. -G"Visual Studio 14 2015 Win64" -DBUILD_VISUALIZER=OFF -DCMAKE_INSTALL_PREFIX=C:\simbody
    -- The C compiler identification is MSVC 19.0.23026.0
    -- The CXX compiler identification is MSVC 19.0.23026.0
    -- Check for working C compiler using: Visual Studio 14 2015 Win64
    -- Check for working C compiler using: Visual Studio 14 2015 Win64 -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working CXX compiler using: Visual Studio 14 2015 Win64
    -- Check for working CXX compiler using: Visual Studio 14 2015 Win64 -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
    -- Configuring done
    -- Generating done
    -- Build files have been written to: C:/projects/simbody/build
    cmake --build . --target ALL_BUILD --config Release -- /maxcpucount:4 /verbosity:quiet
    Microsoft (R) Build Engine version 14.0.23107.0
    Copyright (C) Microsoft Corporation. All rights reserved.
     
    pendEs.obj : error LNK2019: unresolved external symbol CPodeCreate referenced in function main [C:\projects\simbody\build\SimTKmath\Integrators\src\CPodes\sundials\tests\pendEs.vcxproj]
    pendEs.obj : error LNK2019: unresolved external symbol CPodeInit referenced in function main [C:\projects\simbody\build\SimTKmath\Integrators\src\CPodes\sundials\tests\pendEs.vcxproj]

Gjacquenot, 2015-09-29:
I have fixed the EXPORT macros problem. So the modifications brought to the code allows to compile with MSVC and MinGW.

https://github.com/Gjacquenot/simbody/commit/49525a942a23d8150c0ee669a1032db681d52018

Tests run fine with MSVC and MinGW 32 bits with the DLL provided by the project.
I still have to investigate why tests crash with MinGW 64 bits with the DLL provided here.
With user specified DLL everything compiles and runs fine.

chrisdembia, 2015-09-29:
With MinGW it might make sense to use the MinGW lapack/blas instead of the ones provided with Simbody. What do you think @sherm1 ?

sherm1, 2015-09-29:
Is there an "official" lapack/blas for MinGW? If so we should probably use it! Or did you have to make one @Gjacquenot?

Gjacquenot, 2015-09-30:
I found out the problem.

The problem with the 64 bit of MinGW come from the exception mechanism.
When downloading MinGW, one has the choice between SEH, SLJL.

I use the SEH version.
However, the Lapack and Blas libraries shipped with Simbody have been linked against a SLJL version of MinGW (One can see it from the libgcc_s_sjlj-1.dll file shipped with it).

Thus the program I have compiled depends on a libgcc_s_seh-1.dll and a libgcc_s_sjlj-1.dll
Apparently you can not mix the two.

If I had chosen a SJLJ version of MinGW, I might have not notice the potential danger.

So with MSVC, the solution proposed to embed Lapack and Blas is fine.
With MinGW, one should add tests to see:

- the version of MinGW (32 or 64 bits)
- if 64 bits, one has to look at the version used (SEH, SLJL)
- if SEH, stop compilation, request user to give its version Lapack. OR commit the SEH version of Lapack?

Here is the output of a `G++ --version` command

    $ g++ --version
    g++.exe (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 4.9.2

One could parse the result of this command with a CMake

    EXECUTE_PROCESS(COMMAND ${CMAKE_C_COMPILER} --version OUTPUT_VARIABLE GCC_VERSION)
    ....

Some Lapack and Blas DLL are available here
http://icl.cs.utk.edu/lapack-for-windows/lapack
But here also, they depend on the libgcc_s_sjlj-1.dll, that one has to find somewhere...

Informations about SJLJ, Dwarf and SEH can be found here:
http://stackoverflow.com/questions/15670169/what-is-difference-between-sjlj-vs-dwarf-vs-seh

sherm1, 2015-09-30:
Thanks, Guillaume -- great detective work. I understand now that Simbody's supplied 32 bit lapack uses Dwarf and 64 bit lapack uses SJLJ. The libraries were built using gfortran not g++. I wonder if SEH is available for Fortran also?

I like your idea for checking compatibility. It would also be reasonable though to handle that in the MinGW build instructions in Simbody's `README.md`. We could simply say that sjlj is required for use with the provided lapack and that if someone is using seh they need to provide their own lapack.

Gjacquenot, 2015-10-01:
Yes I agree on the conditions. We could use CMake to check that these conditions are met.

I was thinking parsing the `g++ --version` command with the following code

    CMAKE_MINIMUM_REQUIRED(VERSION 2.8.8)
    PROJECT(MINGW_CONFIGURATION C)
    
    IF(CMAKE_SIZEOF_VOID_P EQUAL 8)
       SET(ARCH64_BITS TRUE)
    ELSE()
       SET(ARCH64_BITS FALSE)
    ENDIF()
    IF(MINGW AND ARCH64_BITS)
        EXECUTE_PROCESS(COMMAND ${CMAKE_C_COMPILER} --version OUTPUT_VARIABLE MINGW_GCC_VERSION)
        # Split multi-line string into list
        STRING(REPLACE "\r\n" ";" MINGW_GCC_VERSION "${MINGW_GCC_VERSION}")
        STRING(REPLACE "\n" ";" MINGW_GCC_VERSION "${MINGW_GCC_VERSION}")
        # Only keep first line
        LIST(GET MINGW_GCC_VERSION 0 MINGW_GCC_VERSION)
        MESSAGE(STATUS "MINGW_GCC_VERSION     :${MINGW_GCC_VERSION}")
        # Extract the content between parenthesis
        STRING(REGEX REPLACE "^.*\\((.*)\\)" "\\1" MINGW_GCC_VERSION "${MINGW_GCC_VERSION}")
        # Extract the content between parenthesis
        STRING(REGEX REPLACE "^([^-]+)[-]([^-]+)[-]([^-]+)[-].*$" "\\1" MINGW_GCC_ARCH      "${MINGW_GCC_VERSION}")
        STRING(REGEX REPLACE "^([^-]+)[-]([^-]+)[-]([^-]+)[-].*$" "\\2" MINGW_GCC_THREAD    "${MINGW_GCC_VERSION}")
        STRING(REGEX REPLACE "^([^-]+)[-]([^-]+)[-]([^-]+)[-].*$" "\\3" MINGW_GCC_EXCEPTION "${MINGW_GCC_VERSION}")
        MESSAGE(STATUS "MINGW_GCC_ARCH        :${MINGW_GCC_ARCH}")
        MESSAGE(STATUS "MINGW_GCC_THREAD      :${MINGW_GCC_THREAD}")
        MESSAGE(STATUS "MINGW_GCC_EXCEPTION   :${MINGW_GCC_EXCEPTION}")
    ENDIF()

However, these instructions are not robust at all.

    # $ /c/MinGW32_452/bin/gcc.exe --version
    # gcc.exe (GCC) 3.4.5 (mingw-vista special r3)
    
    # $ /c/MinGW32_472/bin/gcc.exe --version
    # gcc.exe (GCC) 4.7.2
    
    # $ /c/MinGW64_481/bin/gcc.exe --version
    # gcc.exe (rev5, Built by MinGW-W64 project) 4.8.1
    
    # $ /c/MinGW64_492_Posix_SEH/bin/gcc.exe --version
    # gcc.exe (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 4.9.2
    
    # $ /c/MinGW64_520_Posix_SJLJ/bin/gcc.exe --version
    # gcc.exe (x86_64-posix-sjlj-rev0, Built by MinGW-W64 project) 5.2.0
    

Therefore, I was thinking using the dll install with MinGW. This solution is more robust

    CMAKE_MINIMUM_REQUIRED(VERSION 2.8.8)
    PROJECT(MINGW_CONFIGURATION C)
    IF(CMAKE_SIZEOF_VOID_P EQUAL 8)
       SET(ARCH64_BITS TRUE)
    ELSE()
       SET(ARCH64_BITS FALSE)
    ENDIF()
    IF(MINGW AND ARCH64_BITS)
        GET_FILENAME_COMPONENT(MINGW_BIN_PATH ${CMAKE_C_COMPILER} PATH)
        MESSAGE(STATUS "MINGW_BIN_PATH = ${MINGW_BIN_PATH}")
        FILE(GLOB allddls ${MINGW_BIN_PATH}/*.dll)
        SET(MINGW_GCC_THREAD "")
        SET(MINGW_GCC_EXCEPTION "")
        FOREACH(f ${allddls})
            IF(${f} MATCHES "libgcc")
                MESSAGE(STATUS "res = ${f}")
                IF(${f} MATCHES "dw2")
                    SET(MINGW_GCC_EXCEPTION "dw2")
                ELSEIF(${f} MATCHES "seh")
                    SET(MINGW_GCC_EXCEPTION "seh")
                ELSEIF(${f} MATCHES "sjlj")
                    SET(MINGW_GCC_EXCEPTION "sjlj")
                ENDIF()
            ENDIF()
        ENDFOREACH()
    ENDIF()

sherm1, 2015-10-01:
I like the more-robust solution but I'm still not clear on what happens afterwards. Is this just to warn people that they can't use the provided lapack?

Minor comments: 
- is `allddls` a typo? I expected `alldlls`.
- the if-elseif block should have a case for when nothing matches (issue a fatal message?) That way if more changes occur in the future at least the CMake script will protect itself.

Gjacquenot, 2015-10-01:
Well it is up to us to decide what happen...
I would propose:

    IF user does not provide a Lapack and Blas implementation
        Check MinGW installation
        IF (Installation OK)
            Continue
        ELSE
            Error
        ENDIF
    ELSE
            Copy MinGW DLL in the bin directory
            Copy Lapack and Blas DLL in the bin directory
    ENDIF

Ok on the comments remarks

sherm1, 2015-10-02:
I'm not sure I follow this exactly but it looks reasonable. It will be useful to see the user instructions in the README.md file in the PR I think. Any thoughts @chrisdembia?

chrisdembia, 2015-10-02:
I think these changes are worth including in Simbody regardless if we add a section to the README about it, though if we don't add a section to the README, the changes are very likely to get stale. @Gjacquenot would you be willing to add a section to the readme that mimics the current build instructions? I'd say the MinGW instructions go at the bottom, as it's probably not as common as the others.

chrisdembia, 2015-12-31:
@sherm1 @Gjacquenot can this be closed?

sherm1, 2016-01-01:
I think PR #441 addressed this issue.

Gjacquenot, 2016-05-07:
@sherm1 : This discussion can be closed!

closed 2016-05-07 by sherm1