simbody/simbody #443: SimTK Exception utilities should be usable in OpenSim too

sherm1, 2015-10-16:
The `SimTK::Exception` classes and macros provide good printf-like message formatting with file and line number inserted. Currently the formatting includes Simbody-specific verbiage, preventing its use from higher-level applications like OpenSim. This should be fixed so that the tools are available but the app-specific part of the formatting is adjustable. Would be nice to add in `__func__` too.

See [Exception.h](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/Exception.h) and [ExceptionMacros.h](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/internal/ExceptionMacros.h).

One thought would be to leave only app-agnostic formatting in SimTK::Exception::Base, then derive SimbodyException and OpenSimException from Base to add in the app-specific text, and derive the rest of the exception subclasses from the appropriate one of those. Not sure if that would work.

/cc @klshrinidhi, @aseth1, @chrisdembia, @tkuchida, @msdemers, @jenhicks

sherm1, 2015-10-16:
From @chrisdembia: 
Just a small point…The SimTK utilities are already used in some places in OpenSim, at least by Matt Millard.

klshrinidhi, 2015-10-16:
For reference:
http://www.boost.org/community/error_handling.html

sherm1, 2015-10-16:
> For reference: http://www.boost.org/community/error_handling.html

Those guidelines are great for boost-like libraries, but definitely *not* for OpenSim. For example, they are mostly opposed to formatting any messages at all, which I agree is appropriate for boost. But that is not our model -- be sure you don't apply the wrong model when thinking about OpenSim! OpenSim has very different goals and targets a very different set of users than boost does.