simbody/simbody #453: MinGW on Appveyor

chrisdembia, 2015-11-05:
@Gjacquenot said in #452 :

> By the way, do you know if it is possible to have Appveyor CI with MinGW? It would be really nice to have CI with MinGW, since MSVC 2015 seems really restrivtive to me.

> If Appveyor does not provide a MinGW compiler, and maybe we can download a file and unzip the compiler.

I'm starting to try this out in https://github.com/simbody/simbody/tree/appveyor-mingw. I'm not sure though if @sherm1 would want appveyor to build mingw.

I think requiring MSVC 2015 on Windows is as bad now as it used to be, since MSVC Community is free.

chrisdembia, 2015-11-05:
@Gjacquenot the MinGW used by appveyor has the wrong threading: https://ci.appveyor.com/project/opensim-org/simbody/build/712/job/uyi1l9k2uh8yut9k.

We could download a different version of MinGW in the appveyor script, as you suggested, or we could try using msys?

chrisdembia, 2015-11-05:
Actually I think the issue is that the MinGW on AppVeyor is 32bit and I was trying to build as 64bit.

Gjacquenot, 2015-11-06:
@chrisdembia : You shloud have a look at https://github.com/gtk-rs/examples/blob/master/appveyor.yml. This appveyor script downloads a MinGW archive, unzip it and use it. I think it is a good start. You may use the URL I have referenced in [Readme.md](https://github.com/simbody/simbody/blob/master/README.md)

chrisdembia, 2015-11-06:
Good find! Thanks.

sherm1, 2015-11-06:
I'm not sure whether it would be helpful to build MinGW on AppVeyor. The main issues seem to be with the variety of MinGW installations that a user might have (for example, mine didn't work until I replaced it with one that Guillaume recommended). Compilation issues should be caught on Travis since we use gcc there also. So I would think the best we can do on AppVeyor is manage to get one version functional which would not guarantee that it will work for users in general.

There would be some value in making sure we didn't break MinGW by some build system change, but those are infrequent compared to code changes. So we would consume lots of AppVeyor build time, and have to maintain a more complicated AppVeyor build environment, for little gain. I'm not sure that's a worthwhile tradeoff. 

What do you think?

Gjacquenot, 2015-11-08:
I agree with @sherm1 : compilation errors with GCC should be caught on Travis. Setting up a MinGW appveyor job will require to download each time MinGW will be costly.

For myself, I have setup a CI job with Jenkins and MinGW.

@sherm1 and @chrisdembia : Have you thought about creating package for Simbody to have an installer artefact?

chrisdembia, 2015-11-08:
Yes; we've thought about it. For CI purposes, we have simbody binary artifacts for linux, osx, and windows online. However, we're not sure we want to post binaries for distribution. On OSX, people can get simbody through homebrew. On Ubuntu, people should be able to get simbody through apt-get. The binaries would be most useful on Windows.

Sherm said this:

> Support for binaries is nice when they work; a nightmare as soon as one fails. So at the moment I’m not sure if we would be doing anyone a favor by releasing Simbody binaries. 
> On Windows people would probably expect an installer if we provided binaries. There might be similar expectations on Linux and OSX too.

How easy is Jenkins to use? Do you pay?

Gjacquenot, 2015-11-09:
@chrisdembia and @sherm1 : For Windows installer, I use CPack and NSIS with CMake. With the install targets already present, it should be really easy to do. I can start it if you want.

@chrisdembia: Jenkins is a CI server, it can be run on any OS and execute any task on a Jenkins slave machine. It is free and open source. It is a little more complicated to use, since one has to install all required plugins to run a job (Git client, doxygen client). But once done, it is as easy as Travis

chrisdembia, 2015-11-09:
Thanks. Do you need your own server for Jenkins? If not, how fast are the machines?

Gjacquenot, 2015-11-09:
You need to access a server. If you can not have access to a distant server, you can set one really easily on your machine, either on Linux or on Windows. You basically need 500Mo of RAM on your machine. Once set up , you need to create a job with the bat or sh commands to configure and compile Simbody.

sherm1, 2015-11-09:
> For Windows installer, I use CPack and NSIS with CMake. With the install targets already present, it should be really easy to do. I can start it if you want.

That is the method we used to use to make Simbody binary installations for Windows a few years ago. They are very nice to have when they work, but we have had more success with people building from source which ensures that their own code does not have compiler conflicts with Simbody. For example, people would run the installer then try to use Visual Studio Express and fail, or they would try to mix Debug and Release libraries. This might be less of a problem now than it was a few years ago since only VS 2015 will work and VS Community is compatible with VS Pro. Are you thinking about creating MinGW installers also? It would be hard to cover all the variations I think.

chrisdembia, 2015-11-09:
Maybe some of the issues could be addressed with documentation that doesn't promise much for the binary distribution.

chrisdembia, 2015-11-09:
We could also distribute the binaries through the Windows chocolately package manager to match the fact that Simbody is distributed through package managers on other platforms.

Gjacquenot, 2015-11-10:
In the past, I have created installers for applications, that were standalone. I used to embed all required DLL. I did not try to create libraries and distribute them. 

I understand that it may lead to DLL conflicts for people who do not have the same dev environnment.
I do not have enough experience with MSVC compilation, so I do not know what to recommend.