simbody/simbody #456: Should add @Gjacquenot's elastic foundation theory test case

sherm1, 2015-11-20:
@Gjacquenot wrote a test comparing a Hertz sphere with an equivalent Elastic Foundation sphere and measuring the dependency of force with depth. I'm filing this issue with the discussion to date copied in so we have a place to track it.

sherm1, 2015-11-20:
From Guillaume Jacquenot 11/20/2015:
I have made some convergence check on the problem of a sphere in contact with an infinite plane with ElasticFoundation.
The result was quite surprising to me, since I was expecting a convergence like for a Hertz problem (Each time we double the penetration depth, the force is multiplied by 2**1.5=2.82
But theoretical background is different...
But after checking it is ok.
Doubling the penetration depth of a sphere on a plane result in multiplying the force normal by 4.
I dont know if this should be in simbody repository?

```c++
#include "SimTKsimbody.h"

using namespace SimTK;
using namespace std;

void testForcesIn();
void testForcesIn()
{
    const double radius = 1.0;
    const double initialPenetration = 0.0005;
    const int maxLevel = 10;
    const Real stiffness = 1e9;
    const Real dissipation = 0.0, us = 0.0, ud = 0.0, uv = 0.0, vt = 0.0;
    vector<double> F;
    for (int i=0;i<maxLevel;++i)
    {
        const double penetration = initialPenetration * pow(2.0,(double)i);
        MultibodySystem system;
        SimbodyMatterSubsystem matter(system);
        GeneralContactSubsystem contacts(system);
        GeneralForceSubsystem forces(system);

        PolygonalMesh sphereMesh;
        sphereMesh = PolygonalMesh::createSphereMesh(radius,6);
        const Body::Rigid body(MassProperties(1.0, Vec3(0), Inertia(1)));
        const ContactSetIndex setIndex = contacts.createContactSet();
        const MobilizedBody::Translation mesh(matter.updGround(), Transform(), body, Transform());
        contacts.addBody(setIndex, mesh, ContactGeometry::TriangleMesh(sphereMesh), Transform());
        contacts.addBody(setIndex, matter.updGround(), ContactGeometry::HalfSpace(), Transform(Rotation(-0.5*Pi, ZAxis), Vec3(0.0,penetration-radius,0.0))); // y < 0
        ElasticFoundationForce ef(forces, contacts, setIndex);
        ef.setBodyParameters(ContactSurfaceIndex(0), stiffness, dissipation, us, ud, uv);
        ef.setTransitionVelocity(vt);
        State state = system.realizeTopology();
        system.realize(state, Stage::Dynamics);
        SpatialVec r = system.getRigidBodyForces(state, Stage::Dynamics)[mesh.getMobilizedBodyIndex()];
        cout<<"Effort for penetration : "<<penetration<<endl;
        cout<<r[1]<<endl;
        F.push_back(r[1][1]);
    }
    cout<<endl<<"Display the ratio between efforts each time penetration is double"<<endl;
    for (int i=0;i<maxLevel-1;++i)
    {
        cout<<(F[i+1]/F[i]) << endl;
    }
}

int main() {
    testForcesIn();
    return 0;
}
```

sherm1, 2015-11-20:
Sherm said:
That’s a great test and a very interesting result. I knew that Elastic Foundation theory is different from the linear elasticity theory represented by Hertz (basically, no coupling between adjacent elements). But I would have guessed that an EF sphere would be *softer* than a Hertz sphere. I’m glad to hear it is behaving correctly – much better than my intuition, I see!

It would be great to have this test in the Simbody repo. It could be an “adhoc” test as it is, but would be more valuable as a regression test where it would fail if someone breaks the contact code in the future. In that case it would need to include the expected result and throw an exception if the results didn’t match (to an appropriate tolerance). If you want to submit a PR for this that would be great – you should in that case include in the comments information about where you found the appropriate theory to test against. If you want to make a regression test, there are some test macros that make it easier. You could look at some of the existing tests like TestGravity, TestLinearBushing, or TestMassMatrix that use the test macros.

Ideally, this code should use the newer implementation of contact in Simbody – the one you are using is deprecated (same theory though so the test is still welcome). In the newer scheme one attaches ContactSurfaces (geometry and material properties) to bodies, and forces are generated when the surfaces interact. ExampleContactPlayground and TheoJansenStrandbeest use the new system which involves ContactTrackerSubsystem and CompliantContactSubsystem rather than the deprecated GeneralContactSubsystem. The new system also has much better force reporting than the old one.

Gjacquenot, 2015-11-26:
This topic can be closed and the associtaed PR containing two new test cases can be merged in `master`.

closed 2015-11-27 by sherm1