simbody/simbody #471: SimTK Exception thrown at TestGeo.cpp:99

worm2fed, 2016-02-15:
Hello. I can't pass TestGeo, there is output. Can u help me to fix this issue?

```
worm2fed@tarvos ~ $ /tmp/yaourt-tmp-worm2fed/aur-simbody/src/build/TestGeo 
Starting test TestGeo ...
  testMiscGeo ...
  testMiscGeo done. real/CPU ms: 0.0 / 0.0
  testBox ...
  testBox done. real/CPU ms: 0.0 / 0.0
  testTriMeshBoundingSphere ...
  testTriMeshBoundingSphere done. real/CPU ms: 33.9 / 33.9
Test failed due to exception: SimTK Exception thrown at TestGeo.cpp:99:
  Internal bug detected: Test condition failed.
  (Assertion '(radius <= ritter.getRadius()*(1.01))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

TestGeo done. real/CPU ms: 34.1 / 34.1
worm2fed@tarvos ~ $ 
```

chrisdembia, 2016-02-15:
What operating system (and version) are you using? Which compiler (and version)? Are you using the latest commit on the master branch?

worm2fed, 2016-02-15:
Oh, I'm sorry
I'm using Arch linux

> worm2fed@tarvos ~ $ uname -a
> Linux tarvos 4.4.1-2-ARCH #1 SMP PREEMPT Wed Feb 3 13:12:33 UTC 2016 x86_64 GNU/Linux
> worm2fed@tarvos ~ $ 
> 

I have installed gcc 5.3.0

I'm trying to install simbody via yaourt.
https://aur.archlinux.org/packages/simbody/
The last update was 2015-12-10 21:03

chrisdembia, 2016-02-15:
We have seen test failures using the lapack/blas that comes on RedHat Linux. So maybe it's the lapack/blas on Arch Linux? Can you try using a different lapack/blas? Perhaps ATLAS, OpenBLAS, etc.?

worm2fed, 2016-02-15:
It's strange that there is only one test failure. Okay, I'll try it.

chrisdembia, 2016-02-15:
Also, thank you for reporting this.

chrisdembia, 2016-02-15:
I found information about the lapack/blas issue that I've seen before: https://simtk.org/forums/viewtopic.php?f=47&t=4773&hilit=lapack&start=0.

Looks like it might be a different issue from yours, but perhaps trying a different lapack/blas is still a good idea.

worm2fed, 2016-02-15:
I tried to use openblas but it's still bad. Tomorrow I'll read the article and try to use ATLAS. Thank u

Bidski, 2019-01-19:
@worm2fed did you ever resolve this? I just came across this issue and am also on Arch Linux. Did using ATLAS make any difference?

worm2fed, 2019-01-19:
@Bidski as I can remember - no =(

it was kind of experiment and now I can't even remember what I wanted to do...

Bidski, 2019-01-19:
For more reference then
- Trying to install simbody 3.6
- TestGeo is the only test that is failing
- I am using OpenBLAS 0.3.4 with LAPACK 3.8.0
- Linux 4.20.3-arch1-1-ARCH #1 SMP PREEMPT x86_64 GNU/Linux

@chrisdembia do you have any other ideas about this problem?

sherm1, 2019-01-19:
Hi, Alex. I took a look at TestGeo to see what might be happening. There are few things I noticed:
- The test involves randomly-generated meshes. On a new platform it is of course likely that the random numbers are different than those generated on the other platforms that Simbody has run on.
- The test is claiming that Simbody's optimal bounding sphere algorithm (Welzl method) should always produce better results than the approximate Ritter method, except for some slight noise. However, you are seeing a case where the "optimal" Welzl radius is more than 10% larger. That does seem wrong.
- However ... I found [a comment](https://github.com/simbody/simbody/blob/master/SimTKmath/Geometry/src/Geo_Point.cpp#L1031) in the Welzl implementation that could possibly explain what's happening. (Click through to see the code.)

Here is an excerpt from the Welzl code:
```
TODO: there are still rare problems that crop up in random-
// point tests. With about 5 million random sets of 1000 points (see TestGeo)
// ... I have also seen the
// minimal sphere come out considerably larger than the crude Ritter sphere
// generated by calcApproxBoundingSphere(); this is also very rare but bears
// investigating. 
```

It is hard to say whether the random numbers you're getting happen to hit one of those cases. It would take careful investigation and capturing of the exact random mesh that caused the problem to find out. In that case it would be the test that is a problem -- it shouldn't have insisted on perfection if the algorithm can be made to fail. If anyone is interested in doing that diagnosis, please do! An easier check would be to note whether you are seeing a systematic problem, or whether just one or a few of the randomly-generated meshes produces the oversized Welzl sphere. Another check you should perform is to run the test in a Debug build (ensuring that the random number sequence is unchanged) and verify that the numerical results are the same (or very close within roundoff errors). If those checks pass, then the fault most likely lies with the test case which is too sensitive to be fully repeatable. 

A reasonable workaround might be to allow the Welzl sphere to be, say, 20% larger than Ritter rather than just 1%. That is, change the `1.01` in the test to `1.2`. If everything else in Simbody is working, that is likely a safe change. 

(By the way, getting a bounding sphere that is somewhat too big will cause slightly less-than-optimal performance but is otherwise harmless. Too small would be a problem!)

sherm1, 2019-01-19:
P.S. @Bidski Please report what you find here. If you determine that the test is too sensitive, we should fix it in the Simbody source so the problem does not recur.

Thanks and regards,
Sherm

Bidski, 2019-01-20:
I, unfortunately, neglected to check what seed was being used at that point in time.

Interestingly enough, depending on the seed chosen for the random number generator I can either get the tests to pass or get the tests to fail in a different way. 

However, I eventually found a seed that generated a fail in same location during the optimised build. Unfortunately, this same seed did not result in the debug build failing.

Bidski, 2019-01-24:
@sherm1 did you have any luck reproducing this issue?

sherm1, 2019-01-24:
Sorry, Alex I haven't been able to get to it yet. I did notice however that the next tests in the same file use a much higher threshold rather than 1%, but that the flaw in this test went unnoticed due to a "lucky" sequence of random problems that didn't cause problems. Will investigate further soon -- meanwhile I suggest updating this test to allow a looser tolerance.