simbody/simbody #477: Conflict when used with Qt

mitkof6, 2016-03-31:
When Simbody is included with Qt, there is a conflict with the definitions. Apparently the problem is the definition of qInfo:

Qt:
`#define qInfo QMessageLogger(QT_MESSAGELOG_FILE, QT_MESSAGELOG_LINE, QT_MESSAGELOG_FUNC).info`

Simbody::StateImpl.h
`Array_<ContinuousVarInfo>      qInfo, uInfo, zInfo;`

`2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C2059: syntax error: 'nullptr'
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C3484: syntax error: expected '->' before the return type
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C3613: missing return type after '->' ('int' assumed)
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C3646: 'info': unknown override specifier
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C3551: if a trailing return type is used then the leading return type shall be the single type-specifier 'auto' (not 'SimTK::Array_<SimTK::ContinuousVarInfo,unsigned int>')
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C2071: 'SimTK::PerSubsystemInfo::uInfo': illegal storage class
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(797): error C2071: 'SimTK::PerSubsystemInfo::zInfo': illegal storage class
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(690): error C2660: 'SimTK::PerSubsystemInfo::QMessageLogger': function does not take 3 arguments
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(690): error C2228: left of '.info' must have class/struct/union
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(690): error C2228: left of '.empty' must have class/struct/union
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(691): error C2660: 'SimTK::PerSubsystemInfo::QMessageLogger': function does not take 3 arguments
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(691): error C2228: left of '.info' must have class/struct/union
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(691): error C2228: left of '.back' must have class/struct/union
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(691): error C2530: 'last': references must be initialized
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(1070): error C2660: 'SimTK::PerSubsystemInfo::QMessageLogger': function does not take 3 arguments
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(1070): error C2228: left of '.info' must have class/struct/union
2>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKcommon/internal/StateImpl.h(1070): error C2228: left of '.push_back' must have class/struct/union`

sherm1, 2016-03-31:
OMG -- how could they define a macro that looks like a variable?? That's really impolite! Is there any way to get Qt to stop doing that? I would suggest that you first discuss this with Qt folks to see if others have hit this problem and if so what they did about it. One possibility would be `#undef qInfo` but I think it is worth investigating the correct solution since there are no doubt other bad macros in Qt.

scpeters, 2016-03-31:
qt has a bunch of macros, both in 4.8 and 5:

* http://doc.qt.io/qt-4.8/qtglobal.html
* http://doc.qt.io/qt-5/qtglobal.html

It's possible we never noticed this before because `qInfo` was added in qt5, and gazebo uses qt4.

I agree that it's a poor macro, but even if they listened to you and changed this, it's already in the qt4 and qt5 API, so it's not going to go away.

You could rename simbody's `qInfo` since it is a private class variable, but `#undef qInfo` is probably easier.

sherm1, 2016-03-31:
> qt has a bunch of macros, both in 4.8 and 5

Thanks, Steve. Wow, that is shockingly bad behavior on the part of Qt. From the links you sent they clearly know how macros are supposed to be named so they won't conflict with other code!

@mitkof6, can you file a bugreport with Qt? After that if you want to submit a PR to change that variable name to avoid trouble I'm OK with that but we have to make them stop adding unfriendly macros to Qt! You could change the Simbody stuff to something like `q_info` (with a note explaining why). There are some similar names there that probably ought to be changed to match even though they don't conflict (yet!).

mitkof6, 2016-04-01:
Hi,

If I change order and include OpenSim-Simbody stuff always on top, there is no problem. Also, if I undef qInfo, I get an error in SimTKlapack.h (SimTK_S_INOUT_ Macro):

`1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1282): error C2144: syntax error: 'char' should be preceded by ')'
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1282): error C2144: syntax error: 'char' should be preceded by ';'
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1282): warning C4091: '': ignored on left of 'char' when no variable is declared
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1282): error C2143: syntax error: missing ';' before ','
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1282): error C2059: syntax error: ','
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1282): error C2059: syntax error: ')'
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1985): error C2144: syntax error: 'char' should be preceded by ')'
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1985): error C2144: syntax error: 'char' should be preceded by ';'
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1985): warning C4091: '': ignored on left of 'char' when no variable is declared
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1985): error C2143: syntax error: missing ';' before ','
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1985): error C2059: syntax error: ','
1>d:\lib\simbody\install\master_3.6_x64_vs2015\include\SimTKlapack.h(1985): error C2059: syntax error: ')'`

If I `#undef small ` also it compiles.

I will go with a PR and will report this to Qt.

scpeters, 2016-04-01:
Does it work if you add the following cmake code to your project?

~~~
ADD_DEFINITIONS(-DQT_NO_KEYWORDS)
~~~

This disables many of the macros in qt5.

mitkof6, 2016-04-01:
> ADD_DEFINITIONS(-DQT_NO_KEYWORDS)

No it doesn't work.

closed 2016-04-07 by mitkof6