simbody/simbody #482: Compiler version check for gcc incorrecly fails

elen4, 2016-04-13:
I tried to compile Simbody on an Ubuntu 14.04 LTS machine, with gcc 4.8.4 installed, as checked with ``gcc --version``. Although the ``SIMBODY_REQUIRED_GCC_VERSION``

chrisdembia, 2016-04-13:
Thank you for reporting this.

Maybe we should try using the variable `CMAKE_CXX_COMPILER_VERSION` instead of using `gcc -dumpversion`

http://stackoverflow.com/questions/14933172/how-can-i-add-a-minimum-compiler-version-requisite

Could you quickly test if this also fixes the problem for you?

elen4, 2016-04-13:
Hi Chris, yes, this works for me as I get the full major.minor.patch version. I am using CMake 2.8.12.2, but I think that this variable was introduced in CMake 2.8.9, so I guess you would need to bump up the minimum required version, since it currently is 2.8.8 for non-MSVC compilers

chrisdembia, 2016-04-14:
Ah okay good point.

chrisdembia, 2016-04-14:
Thanks for checking.

Gjacquenot, 2016-04-18:
I have created a pull request to fix this.
I have not updated the CMake version. Maybe @chrisdembia , you want it to be updated?
We would move from 2.8.8 -> 2.8.10 for CMake

chrisdembia, 2016-04-18:
Ubuntu 14.04 provides cmake 2.8.12.2, and we already require gcc 4.9.1, which Ubuntu 14.04 does not require. So I suppose it would be okay to change the required CMake version.

chrisdembia, 2016-04-18:
I know this isn't directly related to this issue, but it would be good for us to clearly state the version(s) of Ubuntu that we support in the README.

elen4, 2016-04-18:
Isn't the required gcc version 4.8.1? This is stated both in the README and in the main CMakeLists, and Ubuntu 14.04 complies with this

chrisdembia, 2016-04-18:
Oh wow, good catch. Sorry about that. Yes, then the required version of Ubuntu is 14.04. Thanks, @elen4.

sherm1, 2016-04-18:
Didn't we decide we have to require gcc 4.9 though, for regular expression support?

chrisdembia, 2016-04-18:
Yes, you're right (sorry!); still, Ubuntu 14.04 provides 4.9.3.

Gjacquenot, 2016-04-28:
@chrisdembia, @sherm1 , I think this issue can be closed!

sherm1, 2016-04-28:
Can you verify and close please @elen4?

elen4, 2016-04-28:
Well, the original issue was causing problems with gcc < 4.9, therefore increasing the minimum gcc version fixed it automatically :smile:, although it required a manual gcc update on my side.
Anyway, I like this version of the compiler version check better, and it's working for me. I would add information on how to update gcc for ubuntu 14.04 in the README, right now I only see it in the changelog

closed 2016-04-28 by elen4

sherm1, 2016-04-28:
Thanks, @elen4!