simbody/simbody #483: AppVeyor failure in 32-bit on the master branch

chrisdembia, 2016-04-18:
PR #481 succeeded, but appveyor build on the master branch after the merge had one failure (https://ci.appveyor.com/project/opensim-org/simbody/build/822/job/4drq682gmlrgh0cy); TestMassProperties:

[image]

```
Starting test TestMassProperties ...
  testCrossProduct ...
  testCrossProduct done. real/CPU ms: 0.1 / 0.0
  testInertia ...
  testInertia done. real/CPU ms: 0.0 / 0.0
  testHalfCross ...
  testHalfCross done. real/CPU ms: 0.0 / 0.0
  testSpatialInertia ...
  testSpatialInertia done. real/CPU ms: 0.0 / 0.0
  testArticulatedInertia ...
  testArticulatedInertia done. real/CPU ms: 0.0 / 0.0
Test failed due to exception: SimTK Exception thrown at TestMassProperties.cpp:342:
  Internal bug detected: Test condition failed.
  (Assertion '(mabi(1,0) == ~massMoment)' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.
 
TestMassProperties done. real/CPU ms: 0.2 / 0.0
```

This means that the 32-bit binaries from this build were *not* uploaded to appveyor.  I'm now going to restart the appveyor build to see if it succeeds now (the link from above will likely become stale).

chrisdembia, 2016-04-18:
Oy...the same test failed again.

chrisdembia, 2016-04-18:
https://ci.appveyor.com/project/opensim-org/simbody/build/job/kqfhckmxid5cmcyt

tkuchida, 2016-04-18:
The last AppVeyor build on #481 ran 9 days ago and [all tests passed](https://ci.appveyor.com/project/opensim-org/simbody/build/821). There must have been commits to master in those 9 days that weren't pulled into the feature branch.

chrisdembia, 2016-04-18:
My examination of the git history indicates that there are no such commits:

[image]
(https://github.com/simbody/simbody/network)

I really don't know what happened.

tkuchida, 2016-04-18:
The test passes in my local (32-bit Windows) environment. The failure occurs [at TestMassProperties.cpp:342](https://github.com/simbody/simbody/blob/master/SimTKcommon/tests/TestMassProperties.cpp#L342), which is testing a _random_ articulated body inertia---though nothing looks :fish:y. I'll push a branch to investigate on AppVeyor… :mag:

chrisdembia, 2016-04-18:
good luck 🍀

tkuchida, 2016-04-18:
I think the problem might be an uninitialized value ([there's a `-nan` in one of the matrices being tested](https://ci.appveyor.com/project/opensim-org/simbody/build/824/job/pijlqhtcypxe90qj#L61)).

tkuchida, 2016-04-18:
The line `const SpatialMat mabi = abi.toSpatialMat();` [at testMassProperties.cpp:339](https://github.com/simbody/simbody/blob/master/SimTKcommon/tests/TestMassProperties.cpp#L339) triggers a small amount of calculation, but `mabi(1,0)` should simply be `~mabi(0,1)` (see [MassProperties.h:1522](https://github.com/simbody/simbody/blob/master/SimTKcommon/Mechanics/include/SimTKcommon/internal/MassProperties.h#L1522)). `mabi(0,1)` is correct, so I don't know how `mabi(1,0)` can have three bad entries. :confused:

chrisdembia, 2016-04-18:
Does `mabi(0, 1)` look uninitialized? Or did your expectation for `mabi(1, 0)` come from `~mabi(0, 1)`?

sherm1, 2016-04-18:
That looks as though the conversion from symmetric matrix (just 6 elements stored) to full matrix (all 9) somehow failed. But that conversion isn't at MassProperties.h:1522, it is at [MassProperties.h:1315](https://github.com/simbody/simbody/blob/master/SimTKcommon/Mechanics/include/SimTKcommon/internal/MassProperties.h#L1315) because that is an `ArticulatedBodyInertia` rather than a `MassProperties` object.

tkuchida, 2016-04-18:
> Does `mabi(0, 1)` look uninitialized? Or did your expectation for `mabi(1, 0)` come from `~mabi(0, 1)`?

The rest of `mabi` matches the values I get locally (also, the test failed at [TestMassProperties.cpp:342](https://github.com/simbody/simbody/blob/master/SimTKcommon/tests/TestMassProperties.cpp#L342) so `mabi(0,0)` and `mabi(0,1)` are correct). Lines 341 and 342 are
``` cpp
SimTK_TEST( mabi(0,1) == massMoment );
SimTK_TEST( mabi(1,0) == ~massMoment );
```
so `mabi(1,0)` should equal `~mabi(0,1)`.

tkuchida, 2016-04-18:
Thanks, @sherm1. Sorry, I pointed to the wrong line. Any idea why the conversion would fail on only 32-bit Windows?

sherm1, 2016-04-18:
> Any idea why the conversion would fail on only 32-bit Windows?

No. And when I tried making a 32 bit Windows build just now I saw the same thing as you -- all tests passed. Very strange. Does it fail *every time* on appveyor and *never* locally?

Hypotheses:
- something about generating random values makes the test unreliable (and these are not truly random so they might always generate the same bad value on appveyor)
- there is a compiler bug in the 32-bit compiler (which is quite different from the 64-bit one I believe); but why only on appveyor? out of date compiler?

tkuchida, 2016-04-18:
> Does it fail every time on appveyor and never locally?

Actually, it appears to be even more convoluted than that: it passes reliably on my local 32-bit build and it was passing on 32-bit AppVeyor on #481 but not after the merge to master.

> something about generating random values makes the test unreliable

I thought about that, but the numbers for `mass`, `massMoment`, `inertia`, and 33 of the 36 entries in `mabi` match the numbers I get locally.

> there is a compiler bug in the 32-bit compiler

Possibly… there have been a few [AppVeyor platform updates](https://www.appveyor.com/updates) since the last successful build (on #481). Have you applied **Visual Studio 2015 Update 2** locally? I have not.

tkuchida, 2016-04-18:
I'll try `os: Previous Visual Studio 2015`…

tkuchida, 2016-04-18:
[`os: Previous Visual Studio 2015` works](https://ci.appveyor.com/project/opensim-org/simbody/build/826/job/2hbjo40p4pa9m2if) so I guess there's a problem with [the AppVeyor update on April 16](https://www.appveyor.com/updates/2016/04/16).

chrisdembia, 2016-04-18:
Oh jeez.

sherm1, 2016-04-18:
I'm installing update 2 now and will report back.

sherm1, 2016-04-18:
Uh oh! It failed on my local machine with Update 2 :scream:

sherm1, 2016-04-18:
And I verified that testMassProperties succeeds in Debug build. It seems likely that they introduced an optimization bug in the 32 bit Update 2 compiler.

tkuchida, 2016-04-18:
> It seems likely that they introduced an optimization bug in the 32 bit Update 2 compiler.

:scream_cat: Sounds like Einstein's razor was violated (typically misquoted as "Everything should be made as simple as possible, but no simpler."). Should a bug be filed somewhere? I don't see anything relevant in the list of [Visual Studio 2015 Update 2 Known Issues](https://msdn.microsoft.com/en-us/library/mt668659.aspx).

The only other thought I had was that an optimization in Simbody could be taking advantage of a behavior that has changed. Do you see anything :fish:y in the list of [Breaking Changes in Visual C++ 2015 Update 2](https://msdn.microsoft.com/en-us/library/mt656697.aspx)?

sherm1, 2016-04-18:
> Do you see anything :fish:y in the list of Breaking Changes?

Nope. 
> Should a bug be filed somewhere?

Yes. With Microsoft [here](https://connect.microsoft.com/VisualStudio/Feedback/LoadSubmitFeedbackForm?FormID=6406).

I will submit a PR with a workaround in it in a few minutes. Who wants to file the bug report? :smile: Ideally we would find a smaller piece of code that can reproduce the problem but at least they should know it is there.

sherm1, 2016-04-18:
See PR #486 for workaround. Please review.

sherm1, 2016-04-18:
Can this be closed now?

closed 2016-04-18 by chrisdembia

tkuchida, 2016-04-19:
Thanks for investigating, @sherm1 :gift:

sherm1, 2016-04-19:
> Thanks for investigating, @sherm1

Thanks for discovering the problem, @tkuchida!