simbody/simbody #487: Describe how to get GCC 4.9 in Ubuntu 14.04 in the README

chrisdembia, 2016-04-28:
@elen4 suggested this in #482. The CHANGELOG gives instructions; we should copy them over to the README.

Gjacquenot, 2016-04-29:
I think one can find good instructions [here](http://askubuntu.com/questions/466651/how-do-i-use-the-latest-gcc-on-ubuntu-14-04)

chrisdembia, 2016-04-29:
Thanks Guillaume.

Gjacquenot, 2016-05-08:
@elen4 : I have created a [fork](https://github.com/Gjacquenot/simbody) that contain documentation on how to install GCC 4.9 on Ubuntu.
Can you read and check the paragraph "Check the compiler version" in this [commit](https://github.com/Gjacquenot/simbody/commit/c11863e744dd1cf4f9464acec86ac39376120b44)?

elen4, 2016-05-09:
Hi @Gjacquenot , I used a slighty different syntax for some commands (I "configured" separately gcc and g++), but I think these work the same. I also called ```sudo update-alternatives --remove-all gcc``` and ```sudo update-alternatives --remove-all g++``` before the ```sudo update-alternatives --install``` command, but I do not think this is necessary.
So it looks good, **but** you need to add the PPA with newer gcc releases to your system, with
```sudo add-apt-repository ppa:ubuntu-toolchain-r/test``` (this should go before line 256)

Gjacquenot, 2016-05-09:
Hi @chrisdembia , I have created a PR to describe how to get GCC 4.9 in Ubuntu 14.04 in the README.
Could you have a look?

chrisdembia, 2016-05-09:
Thanks for submitting that PR. I have made comments there.

chrisdembia, 2016-05-09:
Thank you @elen4 for taking a look also.

Gjacquenot, 2016-05-22:
Hi @chrisdembia , this issue can now be closed!

chrisdembia, 2016-05-22:
Great. Thank you

closed 2016-05-22 by chrisdembia