simbody/simbody #494: Using "<NONAME>" for sub/system names may not be great for XML format.

chrisdembia, 2016-05-15:
A few classes have the (default?) name `"<NONAME>"`, such as Guts, GutsRep (see [here](https://github.com/simbody/simbody/search?utf8=%E2%9C%93&q=%3CNONAME%3E)).

This causes elements to appear like so:
```xml
<Subsystem name="4" version="1">
<name>&lt;NONAME&gt;</name>
...
</Subsystem>
```

Also, newlines appear as `&#x0A;` when in the value of an element.

These might not be actual issues; just wanted to raise it though.

sherm1, 2016-05-15:
Those subsystem names don't mean anything -- at best they are useful for being able to tell which subsystem you're looking at. So we should change the names to something without problematic angle brackets and with a more useful name than "NONAME"!

I think the newline representation, though ugly, is OK. XML doesn't have anything nicer. What elements have newlines in their values?

chrisdembia, 2016-05-15:
> So we should change the names to something without problematic angle brackets and with a more useful name than "NONAME"!

Sounds good.

> What elements have newlines in their values?

Here are some examples (from my first shot at serializing SBIntanceVars):

```xml
<MassProperties version="1">
                                    <mass>11.776999999999999</mass>
                                    <comInB>-0.070699999999999999 0 0</comInB>
                                    <unitInertia_OB_B>&#x0A;[0.00872888,0,0]&#x0A;[0,0.0123943,0]&#x0A;[0,0,0.00991485]&#x0A;</unitInertia_OB_B>
</MassProperties>
```

```xml
<Array name="outboardMobilizerFrames" n="13" version="1">
    <value>&#x0A;[1,0,0,0]&#x0A;[0,1,0,0]&#x0A;[0,0,1,0]&#x0A;[0,0,0,1]&#x0A;</value>
    <value>&#x0A;[1,0,0,0]&#x0A;[0,1,0,0]&#x0A;[0,0,1,0]&#x0A;[0,0,0,1]&#x0A;</value>
...
</Array>
<Array name="inboardMobilizerFrames" n="13" version="1">
    <value>&#x0A;[1,0,0,0]&#x0A;[0,1,0,0]&#x0A;[0,0,1,0]&#x0A;[0,0,0,1]&#x0A;</value>
    <value>&#x0A;[1,0,0,0]&#x0A;[0,1,0,0]&#x0A;[0,0,1,0]&#x0A;[0,0,0,1]&#x0A;</value>
...
</Array>
```

chrisdembia, 2016-05-15:
Sidenote: tildes (`~`) show up fine in Xml.

sherm1, 2016-05-15:
Oh, that's bad. Those are formatted outputs but we want unformatted. We will need to write to/fromXmlElement for these types.

chrisdembia, 2016-05-15:
Or, write `writeUnformatted()` methods for those types?

sherm1, 2016-05-15:
Yes, but then also `readUnformatted()` (not sure how well developed that is).