simbody/simbody #495: Windows Install Instruction are incorrect

jlong29, 2016-05-20:
The recommended version for installing Simbody using Visual Studio 2015 should be 3.5.3 and not 3.5.1. 3.5.1 does not compile under Visual Studio 2015, though it does under Visual Studio 2013.

chrisdembia, 2016-05-20:
Thanks for reporting. Are you talking about this specific part:

[image]

?

jlong29, 2016-05-20:
My pleasure. That's the line. The second one should be "$git checkout
SImbody-3.5.3"

chrisdembia, 2016-05-20:
Would you be interested in submitting a PR to update the README? You can either just change `3.5.1` to `3.5.3` (there are multiple occurrences), or remove that line and the explanation of it, or give an explanation in the code block

```
$ git clone https://github.com/simbody/simbody.git C:/Simbody-source
$ git checkout Simbody-3.5.1 # For example, if you want a specific release. 
```

Gjacquenot, 2016-05-22:
Hi @jlong29 , @chrisdembia , I have created a PR #496, that updates this version number.

sherm1, 2016-05-22:
Resolved by #496.

closed 2016-05-22 by sherm1