simbody/simbody #507: Serialization of Ground body has NaNs in inertia instead of Infinity

chrisdembia, 2016-06-07:
[I think that the Ground body is created with an Infinity inertia](https://github.com/simbody/simbody/blob/d75d24cbe3f8a74bf7c3a0a7255eacd5a798777e/Simbody/src/BodyRep.h#L135). However, when serializing the MassProperties for ground, I get:

```xml
                                <MassProperties version="1">                                                                                                                   
                                    <mass>Inf</mass>                                                                                                                           
                                    <comInB>0 0 0</comInB>                                                                                                                     
                                    <UnitInertia name="unitInertia_OB_B" version="1">                                                                                          
                                        <I_OF_F>NaN 0 0&#x0A;0 NaN 0&#x0A;0 0 NaN</I_OF_F>                                                                                     
                                    </UnitInertia>                                                                                                                             
                                </MassProperties>
```

Is there some normalization or symmetrification procedure that is causing this? Is this OK? A potential problem is that constructing an Inertia matrix from NaNs throws an exception (in debug only, I think, from the `errChk()` method).

Note that if I change the Ground's MassProperties initialization from `infiniteMassProperties(Infinity, Vec3(0), Inertia(Infinity))` to `infiniteMassProperties(Infinity, Vec3(0), UnitInertia(Infinity))` (to use `UnitInertia`) then the `Inf`s appear correctly in Xml:

```xml
<MassProperties version="1">
    <mass>Inf</mass>
    <comInB>0 0 0</comInB>
    <UnitInertia name="unitInertia_OB_B" version="1">
        <I_OF_F>Inf 0 0&#x0A;0 Inf 0&#x0A;0 0 Inf</I_OF_F>
    </UnitInertia>
</MassProperties>
```

sherm1, 2016-06-07:
Yep, there's a bug [right here](https://github.com/simbody/simbody/blob/master/SimTKcommon/Mechanics/include/SimTKcommon/internal/MassProperties.h#L1389): Inf/Inf = NaN. When given an Inertia, MassProperties tries to divide it by the mass because internally it stores a UnitInertia. An easy fix would be to define Ground using an infinite UnitInertia rather than an Inertia (as you figured out); in that case no divide would occur.

In the XML you displayed the newlines embedded in the data are ugly. Is there a nicer way to output inertias? OSRF's sdf format puts them out as 3 moments and optionally 3 products of inertia.

chrisdembia, 2016-06-07:
> In the XML you displayed the newlines embedded in the data are ugly. Is there a nicer way to output inertias?

I agree they are ugly. This is an issue for all matrix-like types. They still deserialize correctly with the ugly newlines, so this is a lower priority for me. Also, changing the format for `writeUnformatted()` would be backwards-incompatible. We could instead implement `toXmlElement()` methods for all the matrix types, but then we are no longer leveraging the `writeUnformatted()` infrastructure.

sherm1, 2016-06-07:
Yeah I guess a few ugly newlines isn't the end of the world.

PR #508 (2016-06-07): Set Ground's Inertia to Infinity using UnitInertia instead of Inertia
Previously, Ground's inertia ended up being NaN as a result of
an Inf/Inf in Inertia's constructor (thanks @sherm1 for explaining). This PR changes Ground's inertia to be Inf by using the UnitInertia constructor instead.

Fixes #507.
closed 2016-06-07 by sherm1