simbody/simbody #510: simbody includes and links statically a forked IpOpt library, whose symbols conflicts with system IpOpt

traversaro, 2016-07-09:
* Debian bug report : https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=830506
* Gazebo bug report : https://bitbucket.org/osrf/gazebo/issues/2005/segfault-when-calling-ipopt-virtual-method

Quoting @drdanz from the Debian bug report : 
> Simbody includes and links statically its private version of the ipopt library
in libSimTKmath.so exports its symbols.

> This causes issues when trying to link Ipopt and libSimTKmath in the same
executable or when an executable linked to libSimTKmath opens a plugin linked
to Ipopt (this happens for example when opening a plugin linked to Ipopt from
gazebo).

As I am able to understand from https://github.com/simbody/simbody/issues/348 , the simbody internal IpOpt is a fork of an old version of IpOpt, so using directly the system version in simbody is not a viable alternative.
In that case I guess not exporting the IpOpt symbols anymore (or exporting them as different symbols, for example by wrapping them in a different namespace) is the best solution, but I guess that that is technically a change of ABI, even if it is impossible for a 3rd party project to use those symbols as the internal IpOpt headers are not installed by simbody. 

cc @scpeters @j-rivero @aerydna @drdanz

chrisdembia, 2016-07-09:
Ah! I've had this issue too!

Seems we could add the following to [this cmake file](https://github.com/simbody/simbody/blob/master/SimTKmath/Optimizers/src/IpOpt/CMakeLists.txt):

```cmake
if(${CMAKE_CXX_COMPILER_ID} MATCHES "Clang" OR 
    ${CMAKE_CXX_COMPILER_ID} MATCHES "GCC")
    set_source_files_properties(${src_files} ${incl_files}
        PROPERTIES COMPILE_FLAGS "-fvisibility=hidden")
endif()
```

sherm1, 2016-07-11:
Would that prevent the internal IpOpt from working properly? And is this also a problem on Windows?

chrisdembia, 2016-07-11:
> Would that prevent the internal IpOpt from working properly?

My understanding is no; `-fvisibility=hidden` only affects linking against other shared libraries; not visibility within the same shared library. Users of the `SimTKmath` library couldn't try to link against Simbody's IpOpt anyway; as @traversaro mentioned, the IpOpt headers aren't installed. I haven't tried this yet though.

> And is this also a problem on Windows?

I don't think so; from what I saw, there is no `__declspec(dllexport)` in the IpOpt declarations.

chrisdembia, 2016-07-11:
My scheme may not be the best idea...it could cause issues with exceptions arising from IpOpt:

https://gcc.gnu.org/wiki/Visibility

sherm1, 2016-07-11:
@chrisdembia, would it make sense to pull IpOpt out of Simbody and make it an external instead?

chrisdembia, 2016-07-11:
I think it is a great convenience to not have to build IpOpt on your own when using Simbody. On non-windows platforms, users can get IpOpt through package managers. Not so on Windows. Ideally, the chocolatey package manager would have an IpOpt package. Right now, it does not (I just checked). If @klshrinidhi created a "superbuild" procedure for Simbody, that could also allow us to remove IpOpt from Simbody.

I think a good middle ground would be to create the option to use an external IpOpt, just as with BLAS/LAPACK. Then Gazebo and Debian could choose to use an external IpOpt, but Windows users could still avoid building IpOpt on their own.

j-rivero, 2016-07-15:
Thanks @traversaro for handling the problem and @chrisdembia and @sherm1 for answering fast.

> I think the quickest fix is to simply put IpOpt inside the SimTK namespace. This should be fairly straightforward change. I agree that a more appropriate longer term fix is to remove IpOpt from Simbody (or at least allow using an external IpOpt).

I agree with the diagnostic and the proposed solutions. Removing the symbols will break the ABI so we will need a bump in the SONAME and the package name in debian. The best way of doing this would be to release a new version 3.6.x if possible. If this is not possible I will need to make some tricks with the debian packaging. Note that we are in a good time to get fixes/features that affects the ABI for the next Ubuntu Yaketty release before we get closer to the frozen period.

closed 2017-11-13 by sherm1

traversaro, 2017-11-13:
Thanks a lot @chrisdembia!

traversaro, 2017-12-18:
@chrisdembia Do you have any plan of doing a (ABI-breaking, if I am not wrong) release of the master-branch of Simbody in time to be included in Ubuntu 18.04 ? Thanks!

chrisdembia, 2017-12-18:
Our only plan is to make a release of Simbody just before OpenSim 4.0 is released in the next few months. If you need a release sooner, we can ask @sherm1 what he thinks. I'm happy to go through the steps of creating the release.

> ABI-breaking, if I am not wrong

That's my understanding, too.

chrisdembia, 2017-12-19:
@traversaro what's the deadine for getting into 18.04?

traversaro, 2017-12-19:
I have no idea, I think the mantainer of the simbody package @j-rivero can provide some information on this.

j-rivero, 2017-12-19:
> I have no idea, I think the mantainer of the Debian/Ubuntu simbody package @j-rivero can provide some information on this (thanks in advance!).

The end of the import from Debian is scheduled on 1st March. A major release of simbody should imply an update on package name (current is `libsimbody3.5`) which requires an extra process in Debian that could take weeks. I would suggest to release it before the end of January if possible. Feel free to write me when the release is read to be sure that we start the works as soon as possible.

PR #580 (2017-11-13): Change Ipopt namespace to SimTKIpopt
This PR fixes #510 by changing the `Ipopt` namespace to `SimTKIpopt`. This allows me to use Simbody with a different Ipopt in my direct collocation software.

An alternative solution is to create an option for users to "bring their own Ipopt" for use by Simbody. Changing the namespace is a quicker fix and prevents "branching" where Simbody is packaged differently on different platforms, etc. Also, Simbody's Optimizer classes provide dense Jacobian and Hessian matrices and Simbody's Ipopt uses a dense linear solver. I'm not sure that the sparse solvers that  [... cut]