simbody/simbody #511: Simbody 3.5.3 does not include the GLUT headers if stored somewhere other than `/usr/`.

moorepants, 2016-07-19:
Note from @chrisdembia:

> i think this is a bug in simbody's cmake. the following file should have `include_directories(${GLUT_INCLUDE_DIR})`: https://github.com/simbody/simbody/blob/3dd798950b7cd8ba83e9338f0875c9647c8d09d4/Simbody/Visualizer/simbody-visualizer/CMakeLists.txt
i think we have gotten away without that b/c usually GLUT headers are in `/usr/local`

You can see the failure in action here:

https://circleci.com/gh/conda-forge/staged-recipes/6341

chrisdembia, 2016-07-19:
Thanks for reporting.

moorepants, 2018-01-08:
Any chance this could get in the 3.6 release? It would fix the downstream conda builds.

chrisdembia, 2018-01-08:
For reference, here is more detail: 

```
[ 63%] Building CXX object Simbody/tests/CMakeFiles/TestCustomConstraints.dir/TestCustomConstraints.cpp.o
cd /opt/conda/conda-bld/work/simbody-Simbody-3.5.3/build/Simbody/tests && /opt/rh/devtoolset-2/root/usr/bin/g++   -DSIMBODY_VISUALIZER_INSTALL_DIR=\"/opt/conda/envs/_build/libexec/simbody/\" -DSimTK_SIMBODY_AUTHORS=\"Michael.Sherman_Peter.Eastman\" -DSimTK_SIMBODY_COPYRIGHT_YEARS=\"2005-15\" -DSimTK_SIMBODY_LIBRARY_NAME=SimTKsimbody -DSimTK_SIMBODY_MAJOR_VERSION=3 -DSimTK_SIMBODY_MINOR_VERSION=5 -DSimTK_SIMBODY_PATCH_VERSION=3 -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/Platform/Linux/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/./include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/Scalar/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/SmallMatrix/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/Mechanics/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/BigMatrix/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/Geometry/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/Simulation/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/Random/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKcommon/Polynomial/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKmath/./include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKmath/LinearAlgebra/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKmath/Integrators/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKmath/Optimizers/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/SimTKmath/Geometry/include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/Simbody/./include -I/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/Simbody/Visualizer/include  -DBOOST_MATH_DISABLE_FLOAT128 -m64  -std=c++11 -DNDEBUG -O2 -g      -o CMakeFiles/TestCustomConstraints.dir/TestCustomConstraints.cpp.o -c /opt/conda/conda-bld/work/simbody-Simbody-3.5.3/Simbody/tests/TestCustomConstraints.cpp
/opt/conda/conda-bld/work/simbody-Simbody-3.5.3/Simbody/Visualizer/simbody-visualizer/simbody-visualizer.cpp:84:25: fatal error: GL/glut.h: No such file or directory
     #include <GL/glut.h>
                         ^
compilation terminated.
```

@moorepants what is the easiest way to test a fix to this?

moorepants, 2018-01-09:
I think the easiest would be to submit a PR to the conda feedstock but pull the source code from the head of a branch where you have the fix.

chrisdembia, 2018-01-09:
Okay thanks.