simbody/simbody #512: Hard-code relative path from SimTKsimbody library to simbody-visualizer

chrisdembia, 2016-07-19:
Right now, we hard-code a full path to the installed location of the simbody-visualizer; see https://github.com/simbody/simbody/pull/64. It might be better to hard-code a relative path from the SimTKsimbody library file to the simbody-visualizer executable. This would allow us to relocate the installation and still find the visualizer without setting environment variables.

moorepants, 2016-07-19:
Here is a use case: https://github.com/conda-forge/staged-recipes/pull/1076#issuecomment-233785977

With conda we have to overwrite the hardcoded path to make things relocatable.

chrisdembia, 2017-04-10:
This has been addressed for most UNIX systems; #559

closed 2017-04-10 by chrisdembia