simbody/simbody #513: Errors while installing v. 3.5.3.

wagglefoot, 2016-07-25:
Hello!

Just wanted to share some errors encountered whilst trying to install v. 3.5.3 on an Ubuntu 14.04 machine. 

Setting the install directory to `~/simbody` and then getting as far as `ctest -j8` yields the following error:

```
.
.
100/101 Test  #95: GazeboReactionForceWithAppliedForceCompliant ...   Passed    9.12 sec
101/101 Test  #58: SemiExplicitEulerIntegratorTest ................   Passed   17.11 sec

99% tests passed, 1 tests failed out of 101

Total Test time (real) =  19.96 sec

The following tests FAILED:
	 88 - TestCustomConstraints (Failed)
Errors while running CTest
```
Ignoring the above and moving onto `sudo make -j8 install` yields:

```
.
.
.
Built target TwoPendulums
Built target TimsBoxHybrid
Built target UnilateralPointContactWithFriction
[100%] [100%] Built target UnilateralPointContact
Built target WristMomentArm
Install the project...
-- Install configuration: "RelWithDebInfo"
CMake Error at cmake_install.cmake:36 (FILE):
  file INSTALL destination: ~/simbody/share/doc/simbody/api is not a
  directory.

make: *** [install] Error 1
```
The structure of `~/simbody` upon receiving the above error is `~/simbody/share/doc/simbody/api` -  alas the directory has been made.

Thoughts and ideas?

moorepants, 2016-07-25:
Note that the first error your report is already reported in issue #400. Maybe you can comment there on your gcc version and blas/lapack version.

wagglefoot, 2016-07-25:
Oooops, yes that was a year ago. I'll post here and there.

wagglefoot, 2016-07-25:
Compiler version:

`g++ (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3`

chrisdembia, 2016-07-25:
But is `api` a file or a folder?

wagglefoot, 2016-07-25:
It's supposed to be a folder. It is a folder when the install dir is `/usr/local/`

chrisdembia, 2016-07-25:
Yes but the error message indicates that it's a file instead of a folder; is that the case?

wagglefoot, 2016-07-25:
It is a folder, not a file since I managed to `cd` into it.