simbody/simbody #514: Public interface for calcConstraintAccelerationErrors()

chrisdembia, 2016-07-26:
Right now, there is an internal method `SimbodyMatterSubsystemRep::calcConstraintAccelerationErrors()`; this method is not available to Simbody users. I'd like to add a method `SimbodyMatterSubsystem::calcConstraintAccelerationErrors()`. This would be very similar to the presence of `SimbodyMatterSubsystemRep::calcBodyAccelerationFromUDot()` and `SimbodyMatterSubsystem::calcBodyAccelerationFromUDot()`. For example, the user method would accept non-contiguous input/output Vectors.

I plan to add a test case in `TestConstraints`, based on some of the tests in `TestMassMatrix` and on `testConstraintMatrices()` in `TestConstraints`.

I plan to use this method in OpenSim alongside computing an implicit form of the dynamics.

chrisdembia, 2016-07-26:
Right now, the internal `calcConstraintAccelerationErrors()` takes `A_GB`, `udot`, and `qdotdot` as arguments. What do you think about having the public variant just take `udot` and use it to compute `A_GB` from the public `calcBodyAccelerationFromUDot()`, and compute `qdotdot` from the public `calcQDotDot()`?

sherm1, 2016-07-26:
I see there is already a method `Constraint::calcAccelerationErrorFromUDot(state,udot)`. But I went to look at the implementation and saw this:
```c++
Vector Constraint::calcAccelerationErrorFromUDot(const State&, const Vector& udot) const {
    SimTK_THROW1(Exception::UnimplementedMethod, "Constraint::calcAccelerationErrorFromUDot");
}
```
So I guess that's not very helpful!

sherm1, 2016-07-26:
> What do you think about having the public variant just take udot and use it to compute A_GB from the public calcBodyAccelerationFromUDot(), and compute qdotdot from the public calcQDotDot()?

Yes, that's definitely the way to do it. The A_GB, qdotdot, and udot must all be consistent so you don't want to trust users to supply those.

A few testing ideas:
- use forward dynamics on a constrained system to calculate udots; save those; then using a different State object that has been realized only to Velocity stage, call your new method with (a) zero udots and verify you get non-zero errors, and (b) the saved udots and verify you get zero errors.
- the acceleration error equations are G udot - bias(t,q,u) = 0. So you should be able to use your method to calculate G, which Simbody will also do for you with the calcG() method. You would call your method once with udot=0 to calculate -bias, then m times with one of the udots 1 and the rest 0 to pick up one column of G.

This is a great addition to Simbody -- thanks!

chrisdembia, 2016-07-26:
> So I guess that's not very helpful!

❗  ->  😞 

Thank you for those testing ideas.

I'll proceed on this now.

PR #517 (2016-07-28): SimbodyMatterSubsystem::calcConstraintAccelerationErrors().
Previously, `SimbodyMatterSubsystemRep` had an operator
`calcConstraintAccelerationErrors()`, but there was no related operator for use
by users. This PR introduces such an operator. The operator is tested
in TestConstraints to ensure it gives results that are consistent with
other methods/operators.

In the doxygen, I stated that the complexity of the operator is O(m+n) This was determined by inspecting the internal documentation for:
 * `SimbodyMatterSubsystemRep::calcConstraintAccelerationErrors()` (O(m+n)), 
 * `SimbodyMatterSubsy [... cut]
commit ce5fb7a4 (2016-07-28): SimbodyMatterSubsystem::calcConstraintAccelerationErrors().

Previously, SimbodyMatterSubsystemRep had an operator
calcConstraintAccelerationErrors(), but there was no related operators for use
by users. This commit introduces such an operator. The operator is tested
in TestConstraints to ensure it gives results that are consistent with
other methods.

Fixes #514.
closed 2016-07-28 by sherm1