simbody/simbody #516: Add version dependency for LAPACK in readme

jstaf, 2016-07-27:
I've compiled OpenSim 3.3 with Simbody 3.5.3 on CentOS 6. Anyhow number of `analyze` (OpenSim) runs were crashing with this extremely cryptic error (some, but not all runs would crash):

```
<rest of log appears normal>
**    on entry to DLASCL parameter number 4 had an illegal value
```

It turns out that this was caused by the CentOS 6 LAPACK libraries (3.2.1) being too old. Can you add a note in your Linux compilation instructions to indicate that a specific version of LAPACK is required? I ended up getting things to work by compiling against LAPACK 3.6.1.

LAPACK itself can be built pretty easily like this (you then compile simbody with the -DBUILD_USING_OTHER_LAPACK option with cmake):
```
cmake ../lapack-3.6.1 -DCMAKE_INSTALL_PREFIX=../lapack-3.6.1-install/ -DCMAKE_BUILD_TYPE=RELEASE -DBUILD_SHARED_LIBS=ON
make -j 4
make install
```

chrisdembia, 2016-07-27:
Ideally we would require the oldest version that works. Do you know which version of LAPACK fixes this bug?

chrisdembia, 2016-07-27:
Also, thank you for reporting this! And for testing a different version of LAPACK.

jstaf, 2016-07-27:
It looks like OpenSim supports Ubuntu versions 13.10 and newer, so I would assume that the version of LAPACK used by Ubuntu 13.10 and newer would work. Canonical doesn't have the package list for 13.10 up anymore, but it looks like [Ubuntu 14.04 uses LAPACK 3.5.0](http://packages.ubuntu.com/trusty/liblapack-dev). 

LAPACK 3.5.0 seems like a safe bet as to the minimum required version.

chrisdembia, 2016-07-27:
Okay that sounds good to me. Thank you for looking into that. It's unclear to me if `find_package(LAPACK)` supports a version number, given that this will probably use [`FindLAPACK.cmake`](https://github.com/Kitware/CMake/blob/master/Modules/FindLAPACK.cmake) and this file does not mention `version` anywhere.

https://cmake.org/cmake/help/v3.0/command/find_package.html

jstaf, 2016-07-27:
Honestly, just having a note about it in the Simbody compilation instructions would probably be enough. I'm not too well-versed with cmake so I'm not sure I can be that helpful in regards to having it search for a required version automatically.

chrisdembia, 2016-07-27:
Good point. Would you be interested in making a PR for that? If not, I'll try to do it later today.

jstaf, 2016-07-27:
Sounds good, I'll add a PR for things tomorrow morning.

chrisdembia, 2016-07-28:
Woohoo! Thank you.

jstaf, 2016-07-29:
Pull request sent. After doing some testing, it appears that only LAPACK 3.6.0 fix the issues on Linux.