simbody/simbody #519: convertRotationToQuaternion flips the sign of the quaternion elements

mitkof6, 2016-08-02:
It seems that in some cases almost identical Euler angles are reproduced by quaternion with flip signs. 

For example:

 [0.003 -0.803  0.588 -0.099] ->  [173.508  9.350  72.959]
 [0.003  0.803  -0.588  0.099] -> [ 172.953  8.941  72.979]

In case that someone is calculating numerical derivatives, will get large values because of the sign change. I find that when I use the following implementation I avoid this kind of problem when I have to convert a rotation matrix to quaternion representation.

```
SimTK::Quaternion convertMatrixToQuaternion(const SimTK::Rotation & R)
{
```
   ```
    double qw, qx, qy, qz;
    double tr = R[0][0] + R[1][1] + R[2][2];
    if (tr > 0)
    {
        double S = sqrt(tr + 1.0) * 2; // S=4*qw
        qw = 0.25 * S;
        qx = (R[2][1] - R[1][2]) / S;
        qy = (R[0][2] - R[2][0]) / S;
        qz = (R[1][0] - R[0][1]) / S;
    }
    else if ((R[0][0] > R[1][1]) && (R[0][0] > R[2][2]))
    {
        double S = sqrt(1.0 + R[0][0] - R[1][1] - R[2][2]) * 2; // S=4*qx
        qw = (R[2][1] - R[1][2]) / S;
        qx = 0.25 * S;
        qy = (R[0][1] + R[1][0]) / S;
        qz = (R[0][2] + R[2][0]) / S;
    }
    else if (R[1][1] > R[2][2])
    {
        double S = sqrt(1.0 + R[1][1] - R[0][0] - R[2][2]) * 2; // S=4*qy
        qw = (R[0][2] - R[2][0]) / S;
        qx = (R[0][1] + R[1][0]) / S;
        qy = 0.25 * S;
        qz = (R[1][2] + R[2][1]) / S;
    }
    else
    {
        double S = sqrt(1.0 + R[2][2] - R[0][0] - R[1][1]) * 2; // S=4*qz
        qw = (R[1][0] - R[0][1]) / S;
        qx = (R[0][2] + R[2][0]) / S;
        qy = (R[1][2] + R[2][1]) / S;
        qz = 0.25 * S;
    }
    SimTK::Quaternion q(qw, qx, qy, qz);
    q.normalize();
```

```
    return q;
} 
```

sherm1, 2016-08-02:
Hi, Dimitar. Can you be more specific about what derivative you are attempting to calculate? The main utility of quaternions is that they are smooth under differentiation, while Euler angle are discontinuous. I'm not sure how a Rotation-to-Quaternion conversion method relates to differentiation in this case. Regards, Sherm

mitkof6, 2016-08-03:
Hi Sherm,

I have a quaternion set where I perform interpolation and then I compute first and second order derivatives with finite difference. The problem is that these quaternions originate from a rotation matrix, which I compute and then I use `convertRotationToQuaternion `. Unfortunately, for some unknown reason the current implementation of `convertRotationToQuaternion ` flips the sign of adjustment rotation matrices ([image](https://dl.dropboxusercontent.com/u/50567323/sign-flip.png)) while the final Euler angles are correct, this sign change causes problems in the numerical differentiation. You can see the effect of the two implementation ([image](https://dl.dropboxusercontent.com/u/50567323/orientation.png)) on the angular velocity.

The two implementations seems very similar, my guess is [scale](https://github.com/simbody/simbody/blob/3dd798950b7cd8ba83e9338f0875c9647c8d09d4/SimTKcommon/Mechanics/src/Rotation.cpp#L671) this line of code, which I don't understand.

Best

chrisdembia, 2016-08-03:
I think @sherm1 understood the problem but wanted to know what you are using the derivatives for. For optimization? For a PD control term?

sherm1, 2016-08-03:
There is no unique mapping from Rotation to Quaternion, because a quaternion and its negative represent the same orientation. To avoid arbitrariness, the Simbody `convertRotationToQuaternion()` uses that scale line to generate the ["canonical"](https://simtk.org/api_docs/simbody/latest/singletonSimTK_1_1Quaternion__.html#details) quaternion, that is, the one that has its first element positive. That means the conversion from Rotation to Quaternion will be non-smooth as the first element passes through zero which is what you are seeing.

However, I don't believe that your version of the conversion code fixes the problem, it just moves it elsewhere. It also contains several other "if" clauses that may be traversed differently if you attempt differentiation by finite differences. Your test case likely just didn't hit those cases. It *is* possible to differentiate code like that, but to do so you must ensure that it follows the same path through any conditional statements.

What I don't understand is why you are doing the differentiation this way. You didn't answer my question about what derivative you are trying to calculate, that is: partial(some_calculation)/partial(some_variables). The reason I ask is that you *can* compute partial(quaternion)/partial(q) by finite differencing since quaternions are smooth, but you can't differentiate the conversion from rotation matrix to quaternion that way because of the if statements unless you ensure they follow the same path.

mitkof6, 2016-08-03:
Thanks for the comments. I care for the derivative of the quaternion wrt to time, which I need in order to compute the angular velocity and acceleration. I ensure that I can evaluate an orientation (use smooth quaternion splines, which I populate) on the unit length quaternion so that it always represents a rotation.

For now it seems that the aforementioned implementation does the trick. Do you know of a better approach to differentiate a quaternion wrt time or a rotation matrix?

sherm1, 2016-08-03:
I think I'm still unclear on what you're doing, Dimitar. Simbody calculates all those derivatives analytically but I think you already know that so you must be doing something else.

mitkof6, 2016-08-03:
Sorry. I estimate these quantities from measurements, and I don't extract them from simulation. I reconstruct the spatial properties of a body from the measured marker trajectory (>=3). For a given set of marker positions at n and n + 1 I compute the transition transformation. Thus if I multiply all transformation (with known initial condition) I know the desired orientation that I want to achieve. Unfortunately, when I track this goal I need the angular velocity and acceleration (PD law) of the marker set, so that is why I use quartenions (due to its interpolation) and I differentiate them wrt to time so that I can use the aforementioned formulas. The same can be achieved with rotational matricies, but they are more difficult to interpolate than quaternions.

sherm1, 2016-08-03:
I think I get it now. You know Rotation matrices R_GB[n] and R_GB[n+1] where R_GB is the orientation of a body B's frame in the ground frame. You would like to smoothly interpolate between those two orientations. So you convert the Rotations to Quaternions Q_GB[n] and Q_GB[n+1] to facilitate interpolation. But there is some arbitrariness to conversion from R to Q, because +/-Q represent the same orientation (quaternions cycle every 4π). The two conversions are independent of one another, so they can't coordinate to favor continuity. Instead, Simbody's conversion routine always chooses the representation in which the first entry (the "scalar" term `cos(a/2)`) is positive. In your problem there are steps n and n+1 where the rotation angle `a` passes through +/-π while going from n to n+1, which would cause `cos(a/2)` to change sign, except that Simbody canonicalizes it by moving the rotation angle into the [-π,+π] interval. The reason the rotation angle can get so big is that you are performing this operation in the Ground frame. I presume the amount of rotation between steps n and n+1 would actually be much, much smaller than π.

As I said earlier, I am not convinced that removing the canonicalization line from the conversion fixes the problem. I think it just moves it elsewhere. You still have the problem that the two conversion are done independently, so they aren't biased to choosing the closest pair of quaternions. I speculate that there must still be certain configurations in which the pair of quaternions will be on opposite sides of the quaternion cycle.

I would suggest this approach to solving the problem: don't attempt to do these calculations in the Ground frame. In that frame the rotation angles are arbitrarily large. Instead, work locally in the B[n] frame to move to the B[n+1] frame. That is, the quaternion at B[n] would be (1,0,0,0). The small rotation from B[n] to B[n+1] will result in small-angle quaternions that are never near the +/-π boundary. You can then reliably calculate the derivatives along the path from n to n+1, and since you already know the derivatives at B[n] you can add those in to get the ground frame angular velocity & acceleration.

A possibly easier (if less elegant) approach would be to post-process the quaternion pair you have after conversion to make sure you are taking the shortest path in the representation from Q[n] to Q[n+1] (you might have to flip the sign on one to bring them closer). I think you should do that regardless of which conversion routine you use because I think they will all have discontinuities somewhere.

mitkof6, 2016-08-04:
Thank you Sherm, for the the detailed explanation. 

> : don't attempt to do these calculations in the Ground frame. In that frame the rotation angles are arbitrarily large. Instead, work locally in the B[n] frame to move to the B[n+1] frame.

I compute the optimal transformation that maps two point set from n to n+1, that is T_GB = {R, t} as you correctly pointed out. Can you be more specific how I can work in local frame instead (procedure)? Should I express the marker positions in body frame and then compute the relative transition? Please note that I also have a translation component. 

Also, I see your point that maybe the problem will be translated elsewhere (in my implementation), but I was curious is there any practical reason why you should make a canonical quaternion?

Regards

mitkof6, 2016-08-04:
> Can you be more specific how I can work in local frame instead (procedure)? 

I know T_GB(n) = T_GB(n, n-1) * ... * T(1, 0) * T(0), where T(0) is the initial pose of the body wrt to the ground frame and T(n, n-1) is the transition from n to n-1. Do you actually, mean that I should compute my derivatives ignoring T(0) and then at the end I multiply by T(0) my orientations and as for the angular velocity and acceleration they are zero?

sherm1, 2016-08-04:
Let Bn be the body frame at step n, Bn+1 that same frame but at the next sample point. We would like to get the transform T_BnBn+1, that is, the *relative* pose of Bn+1 as seen from Bn. That is presumably a small rotation (and small translation) so should be easy to work with.

From mocap you could get the transforms `T_GBn` and `T_GBn+1`. Then `T_BnBn+1=T_BnG*T_GBn+1` where `T_BnG=~T_GBn` (~ means inverse here; Simbody has that operator).

Now you can consider Bn as a fixed frame (temporarily) and interpolate from Bn to Bn+1 using a quaternion starting at (1,0,0,0) [no rotation] and ending at the relative rotation between Bn and Bn+1 (that is, what I would call R_BnBn+1). That will be a well-behaved quaternion because of the small angles involved. You can differentiate it to calculate angular velocity & acceleration of Bn+1 in Bn.

Assuming you already know Bn's velocity and acceleration in Ground, you can now take the relative velocity & acceleration, add those to the known Bn values to find the ground velocity and acceleration of Bn+1, and of the interpolated values between Bn and Bn+1.

sherm1, 2016-08-04:
> is there any practical reason why you should make a canonical quaternion?

Yes. It is to turn a non-deterministic result into a deterministic one. In most cases it ensures that a pair of quaternions that are nearby have similar representations, but as you've seen it can fail for large rotations when they wrap around at 180 degrees. But that non-determinism doesn't go away if you ignore it!

mitkof6, 2016-08-04:
Thank you Sherm, it seems that the discontinuity reemerged, I will let you know when I fix it.

Best

FCL-M, 2021-04-01:
Hi All, 

is this issue solved? If yes, I'm very interested as I have exactly the same issue: a series of quaternions obtained from measurements that result is enormous outliers when calculating the angular velocity. just canonising the quaternions doe not solve the problem.... The body I'm looking at has 3 threefold symmetry so there are 6 orientations that virtually look alike. So some indeterminism is in place (when limiting the maximum rotation angle between frames to pi/2 the problem is not solved, still flipping of the body ref frame occurs)...

thanks in advance

sherm1, 2021-04-01:
Hi, @FCL-M. Please read the entire discussion above to see the cause and possible solutions for this problem. There is no magic solution to be found by changing how rotation matrices are converted to quaternions -- there is inherent arbitrariness there that cannot be eliminated. However by changing frames you can avoid the problem.

I'm going to close this issue since everything is working as it should.

closed 2021-04-01 by sherm1