simbody/simbody #520: CMAES Optimizer get stuck for high dimension problem

mitkof6, 2016-08-04:
I am trying to perform Anderson F.C., el al., A Dynamic Optimization Solution for Vertical Jumping, and to find a solution through optimization. There is something that me and some students want to test. I took a code written from Carmichael and adapted it so I can have the same criterion as in the original paper. I choose CMAES, because the other optimization algorithms does not find a good solution (local minimum). The source code can be downloaded from [github](https://github.com/mitkof6/skyjump) and it uses OpenSim3.3.

In my implementation I have 24 muscles and I also choose the number of control nodes per muscle (so I can find the muscle control signal). When I have 9 control points I have 24 * 9 = 216 parameters and CMAES, initializes and runs normally. I wanted to add more control points and when I set p > 10 CMAES gets stuck in the initialization, and the objective function is not called at all. Is this a limitation to the algorithm?

Also, because I am lazy sometimes, I find that brute force can give you a good solution and does not get stuck in local minimum. Are you interested in a pure random optimization as an alternative algorithm to compliment Simbody optimization algorithms?

chrisdembia, 2016-08-04:
Cool project; thank you for sharing.

Dimitar, could you make the title of this issue more specific? Is this a Simbody issue, or a CMAES issue? That is, if you use other CMAES implementations, do they have the same problem? What is p?

mitkof6, 2016-08-04:
I can't figure out where it gets stuck, but I will have a look again. I have used Simobody's CMAES. 

p is the number of control nodes per muscle, so when I increase p the algorithm gets stuck in the initialization.

mitkof6, 2016-10-18:
It seems that a bad initialization of the parameter space, and the default standard deviation of CMAES can cause this problem. A proper initialization of the CMAES will not cause problems, even if there are many parameters.

chrisdembia, 2016-10-18:
Alternatively, it may help to scale the problem better? Perhaps the best fix is to add more documentation about choosing the parameters of the CMAES algorithm.

carmichaelong, 2016-11-01:
Was it the case that your initial guess had many parameters that were near their bounds and/or your standard deviation was large compared to the range of the bounds of your parameters? Since CMAES has to sample a distribution to generate a population, sometimes it can sample outside of the bounds. In this case, it must *resample* the distribution (or else you'd get some bias in your sampling if you just clamp these guesses). With many parameters, this means it becomes more and more unlikely to sample a valid distribution, which means you can get caught in initialization (or really at any step along the way, but mainly at initialization).

ece7048, 2016-11-03:
I am using an optimization in the same problem with a 36 points in each muscle and runs well. For each of the initial values of parameters, you have to set a good set up of the "init_stepsize" for the convolution matrix with the: opt.setAdvancedRealOption("init_stepsize", X);. The X value is depending  the initial values of the parameters and the limit boundaries. For example if you used 0.5 for one of the parameter value of muscle exitation, then you could put X=0.1 or 0.01 or 0.001 etc. but if you put 0.95 you will able to put X=0.1 but if it give a value bigger than 1 (which is the upper boundary) it has to resample all the parameters because it out of boundaries of the spatial sample. This is my estimation for the code until this time.

ece7048, 2016-11-03:
The stucks happend because it tries to found a correct sample and  this it may will  take a lot of time with respect the case.

chrisdembia, 2016-11-03:
Thank you @ece7048 . This all sounds correct. It might be good to add this to the documentation. Perhaps here: https://github.com/simbody/simbody/blob/master/SimTKmath/include/simmath/Optimizer.h#L319

mitkof6, 2016-12-06:
@chrisdembia Correct me if I am wrong but it seems that the init_stepsize is the same for all variables. Since this parameter is essential, this may be problematic in case that the parameters under optimization have different bounds (lower and upper). A solution would be to transform the variables in the region [0, 1] and use the same std or enable the user to specify an array of std for each parameter. What do you think?

chrisdembia, 2016-12-06:
> Correct me if I am wrong but it seems that the init_stepsize is the same for all variables

You are correct; the code is [here](https://github.com/simbody/simbody/blob/master/SimTKmath/Optimizers/src/CMAESOptimizer.cpp#L160).

The way that we have used CMA is to scale the parameters to achieve the desired effective standard deviation for each parameter. You are right though that this is not the ideal interface. You could also imagine the optimizer interface handling the scaling of parameters (which should usually be done anyway).

The SimTK::Optimizer interface does not really make it easy to specify a different standard deviation for each parameter ([here are the knobs available](https://github.com/simbody/simbody/blob/master/SimTKmath/include/simmath/Optimizer.h#L388)).

mitkof6, 2016-12-06:
@chrisdembia I have implemented a Vector set, get advanced option and fixed this issue. I will see if any other optimization parameter should be exposed and will post a pull request.

ece7048, 2016-12-06:
I am not sure but a solution It may be to normalized all the parameters to be  in space of [0,1] and then do the fix std deviation.

carmichaelong, 2016-12-06:
> normalized all the parameters to be in space of [0,1] and then do the fix std deviation.

This is a very good way to deal with this (both specifically for CMA and for other optimizers).

> enable the user to specify an array of std for each parameter

I have found this also can work pretty well. Looking at the code snippet that @chrisdembia mentioned, it looks like you just have to be careful about dealing with the `init_stepsizeArray` and `stddev`. Remember that in CMA, there is a constant in front of the covariance matrix that also scales the problem.