simbody/simbody #522: Some simbody 3.5.3 tests fail on Ubuntu 14.04 LTS

wagglefoot, 2016-08-30:
Following 

`ctest -j8`

we get

```
98% tests passed, 5 tests failed out of 202

Total Test time (real) =  32.44 sec

The following tests FAILED:
	162 - TestReverseMobilizersStatic (Failed)
	174 - GazeboReactionForceWithAppliedForceRigidStatic (Failed)
	175 - TestCustomConstraints (Failed)
	176 - TestCustomConstraintsStatic (Failed)
	190 - GazeboReactionForceWithAppliedForceCompliantStatic (Failed)
Errors while running CTest
```

This is on a Ubuntu machine running 14.04 with `cmake version 3.2.2`.

chrisdembia, 2016-08-30:
Thank you for sharing. What compiler (and version) did you use? It is suspicious that the static version of some tests fail, but the shared version passes. Is this from a fresh build directory? Did you perhaps update any software on your computer between when you created the build directory and now? What do the tests output? `ctest -j8 --output-on-failure`

sherm1, 2016-08-30:
I notice these are the "static" versions of the tests (i.e. linked with static libraries). I don't think we have tried those for a long time. Do you get the same failures with the dynamic tests? I believe the static tests require some `#define` magic -- maybe those particular tests don't have it.

wagglefoot, 2016-08-30:
@chrisdembia this all happened within the space of a few hours and I have the most recent versions of cmake and swig (not that it is used). And they were updated today before I downloaded the source (today).

Alas;

```
clang --version
Ubuntu clang version 3.4-1ubuntu3 (tags/RELEASE_34/final) (based on LLVM 3.4)
Target: x86_64-pc-linux-gnu
Thread model: posix
```

```
gcc --version
gcc (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

clnsmith, 2016-10-14:
Hi all,

I'm building on Scientific Linux 6 using GCC 4.8.2 and Lapack 3.6.1 and I am getting the same failed tests:

The following tests FAILED:
	164 - GazeboReactionForceWithAppliedForceRigidStatic (Failed)
	172 - TestReverseMobilizersStatic (Failed)
	182 - GazeboReactionForceWithAppliedForceCompliantStatic (Failed)
Errors while running CTest

The shared library versions of the same tests all pass. All OpenSim tests pass as well. So I am not too worried about this. 

Just FYI, I originally tried compiling with GCC 4.4.7 built everything ok, but I got close to 20 failed tests (static and shared).  

  1/202 Test #172: TestReverseMobilizersStatic ..........................***Failed    0.10 sec
*** TEST PIN ***

Vels B:
  Fwd:   ~[~[-2.69705,-0.486375,1.2204],~[1.64348,-3.55569,2.21496]]
  Rev1: ~[~[-2.69705,-0.486375,1.2204],~[1.64348,-3.55569,2.21496]]
  Rev2: ~[~[-2.69705,-0.486375,1.2204],~[1.64348,-3.55569,2.21496]]
Vels A:
  Fwd:   ~[~[0,0,0],~[0,0,0]]
  Rev1: ~[~[8.88178e-16,3.33067e-16,0],~[1.33227e-15,0,0]]
  Rev2: ~[~[8.88178e-16,3.33067e-16,0],~[1.33227e-15,0,0]]
Accels B:
  Fwd:   ~[~[1.21637,-0.909969,0.229318],~[9.50996,10.1207,16.9542]]
  -Rev1: 1.09342e-13
  -Rev2: 1.09342e-13
Reacs BA:
  Fwd:   ~[~[511495,1.34172e+06,1.66519e+06],~[212758,-642673,624990]]
  Rev1: ~[~[511495,1.34172e+06,1.66519e+06],~[212758,-642673,624990]]
  Rev2: ~[~[511495,1.34172e+06,1.66519e+06],~[212758,-642673,624990]]

*** TEST PLANAR ***

rev2A.getU()=~[3,0.3,0.4]
Vels B:
  Fwd:   ~[~[-2.69705,-0.486375,1.2204],~[1.84134,-3.905,2.51301]]
  Rev2: ~[~[-2.69705,-0.486375,1.2204],~[1.84134,-3.905,2.51301]]
Vels A:
  Fwd:   ~[~[0,0,0],~[0,0,0]]
  Rev2: ~[~[8.88178e-16,3.33067e-16,0],~[-1.77636e-15,5.32907e-15,0]]
Accels B:
  Fwd:  ~[~[0.236582,-1.53248,-0.0668399],~[12.6116,3.17118,21.1439]]
  Rev2: ~[~[0.236582,-1.53248,-0.0668399],~[12.6116,3.17118,21.1439]]
Accels A:
  Fwd:  ~[~[0.41671,-1.5,-0.148348],~[4.48626,-0.0868,3.34765]]
  Rev2: ~[~[0.41671,-1.5,-0.148348],~[4.48626,-0.0868,3.34765]]
Reacs BA:
  Fwd AB:~[~[-83517.8,306091,-62591.3],~[-2877.48,-520.246,1298.48]] p_BA: ~[2.12132,-7.77817,-0]
  Fwd:   ~[~[72800.6,-290436,45113.6],~[2877.48,520.246,-1298.48]]
  Rev2:  ~[~[72800.6,-290436,45113.6],~[2877.48,520.246,-1298.48]]
  Fwd-Rev2:~[~[-1.27839e-07,-4.49363e-08,-1.34096e-08],~[-5.12227e-09,1.31695e-08,-3.84171e-09]]

*** TEST ELLIPSOID ***

rev2A.getU()=~[3,0.3,0.4]
Vels B:
  Fwd:   ~[~[0.0312865,-3.03711,-0.157978],~[2.74741,0.807566,-7.26806]]
  Rev2: ~[~[0.0312865,-3.03711,-0.157978],~[2.74741,0.807566,-7.26806]]
Vels A:
  Fwd:   ~[~[0,0,0],~[0,0,0]]
  Rev2: ~[~[7.21645e-16,4.44089e-15,8.32667e-16],~[1.77636e-14,1.11022e-16,-1.42109e-14]]
Accels B:
  Fwd:  ~[~[0.194,2.59397,1.25633],~[-28.3365,1.22785,20.8966]]
  Rev2: ~[~[0.194,2.59397,1.25633],~[-28.3365,1.22785,20.8966]]
Accels A:
  Fwd:  ~[~[1.20227,5.06982,3.82959],~[-21.5864,8.61864,-5.9483]]
  Rev2: ~[~[1.20227,5.06982,3.82959],~[-21.5864,8.61864,-5.9483]]
[... 67 lines cut ...]

GazeboReactionForce done. real/CPU ms: 523.4 / 523.1

  3/202 Test #191: TestCustomConstraints ................................   Passed    0.81 sec
  4/202 Test #182: GazeboReactionForceWithAppliedForceCompliantStatic ...***Failed    0.81 sec
Starting test GazeboReactionForce ...
  runOnce ...
LOW ACCURACY
Test with order 1 integator SemiExplicitEuler2, Accuracy=0.01, MaxStepSize=0.0005
t=4.9995
  joint1 a=~[-4.35438e-06,8.784e-05,-2.23917e-06] w=~[-9.31524e-14,2.39569e-13,-2.03739e-06]
  joint2 qerr=0.00401764 u=-5.90581e-13
  joint3 qerr=0.0025165 u=7.90508e-13
  Reaction 2p=~[~[-25.0009,200.882,-0.00132132],~[0.026352,0.00130637,-300]]
  Reaction 2c=~[~[25.0007,-200.882,0.101766],~[-1.23164,-0.00130637,299.997]]
  Reaction 3p=~[~[-24.9999,25.165,-0.102528],~[0.205273,0.000217727,-49.9996]]
  Reaction 3c=~[~[24.9996,-25.165,0.165439],~[-0.331097,-0.000217727,49.9989]]
  runOnce done. real/CPU ms: 729.4 / 729.0
Test failed due to exception: SimTK Exception thrown at GazeboReactionForceWithAppliedForceCompliant.cpp:327:
  Internal bug detected: Test values should have been numerically equivalent at tolerance=0.5.
  (Assertion 'SimTK::Test::numericallyEqual((reaction3.reactionOnParentInParent),(SpatialVec(Vec3(-25,0,0), Vec3(0,0,-50))),1,(0.5))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

GazeboReactionForce done. real/CPU ms: 732.9 / 732.5

sherm1, 2016-10-14:
It's disturbing to me that there would be *any* difference between the static and shared-library builds.

chrisdembia, 2016-10-14:
@clnsmith are you trying to build Simbody's master branch? Or 3.5.x? Just curious; this doesn't affect my understanding of what's wrong (I have no clue what's wrong).

clnsmith, 2016-10-14:
3.5.3 

The code is from "Download Package" from "OpenSim Source Code" https://simtk.org/frs/index.php?group_id=91

chrisdembia, 2016-10-14:
Thanks.

Two static tests fail for me on macOS 10.12, Apple Clang 8.0.0, latest master branch.

```
The following tests FAILED:
        168 - GazeboReactionForceWithAppliedForceCompliantStatic (Failed)
        170 - GazeboReactionForceWithAppliedForceRigidStatic (Failed)
```