simbody/simbody #523: Compilation failure with macOS 10.12 SDK

scpeters, 2016-09-20:
A build failure of simbody 3.5.3 was reported in https://github.com/osrf/homebrew-simulation/issues/135 ([full details here](https://gist.github.com/bwithbroe/9247eaf889cdaf3e0a97f9df27682af6)) that can be summarized with the following error message:

~~~
In file included from /tmp/simbody-20160920-28673-1o5ma3f/simbody-Simbody-3.5.3/SimTKcommon/src/Timing.cpp:33:
/tmp/simbody-20160920-28673-1o5ma3f/simbody-Simbody-3.5.3/
SimTKcommon/./include/SimTKcommon/internal/Timing.h:88:18:
error: typedef redefinition with different types ('long' vs 'enum clockid_t')
    typedef long clockid_t;
                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/
Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: 
note: previous definition is here
} clockid_t;
^
~~~

They were using macOS 10.11 but with the macOS 10.12 SDK, which appears to define the `clockid_t` type. This will probably cause the same build failures on macOS 10.12 when it is released. Thanks to @bwithbroe and @uframer for reporting this issue.

chrisdembia, 2016-09-20:
Thank you for reporting this.

> This will probably cause the same build failures on macOS 10.12 when it is released.

10.12 was released today!

@sherm1 I can try to fix this. Let me know if you have any suggestions on the fix.

chrisdembia, 2016-09-21:
Related:
- https://github.com/Homebrew/homebrew-core/issues/2674
- https://github.com/Homebrew/homebrew-core/issues/1957
- https://github.com/civetweb/civetweb/pull/356/files

sherm1, 2016-09-21:
Hi, @chrisdembia. Thanks for volunteering to fix this! Looks like I had to fake `clockid_t` and `clock_gettime()` for earlier OSX versions (in Timing.h). If you can figure out what magic `#ifdef` determines whether OS 10.12's SDK is in use, those definitions could be skipped.

closed 2016-09-22 by sherm1

chrisdembia, 2016-09-22:
Thank you for merging #524. Do you think we should make a new release (3.5.4), so that a released version of Simbody supports macOS 10.12?

sherm1, 2016-09-23:
> Do you think we should make a new release (3.5.4)

To do that would require a backport to the simbody-3.5 branch since #524 went into the master (3.6) version. That would allow C++03 compilations on 10.12 -- is that still useful? Otherwise we could consider just including that in a 3.6 release with the other new goodies there.

chrisdembia, 2016-09-23:
I am fine with having this bug fix in a 3.6 release. My reason for suggesting 3.5.4 instead was that I thought a 3.6 release was not ready yet, and it would be nice to have the bug fix release somewhat soon. But if you think the 3.6 release could be soon, then we do not need to backport.

bwithbroe, 2016-09-26:
Do you have a rough release schedule for 3.6?

sherm1, 2016-09-26:
> Do you have a rough release schedule for 3.6?

Not yet. @chrisdembia and I need to discuss to decide whether 3.6 is really ready to go now. Otherwise the 3.5.4 options sounds good - how fast do you need it?

scpeters, 2016-09-26:
Also related: https://github.com/Homebrew/brew/pull/1078

chrisdembia, 2016-09-26:
Thank you for sharing, @scpeters. Is the idea that Simbody could use a CMake variable `HAVE_CLOCK_GETTIME` to decide if we should define `clock_gettime()`? To make good use of this variable, wouldn't Simbody's CMake scripts have to detect if they were being run as part of Homebrew?

scpeters, 2016-09-26:
I was just posting that for reference; I think it won't solve the issue for people building simbody from source without homebrew, but it may resolve the issue of people building simbody 3.5.3 with homebrew with xcode 8 on macOS 10.11. Maybe not actually, since it says it addresses lazy linking but it probably won't fix the compilation errors?

chrisdembia, 2016-09-26:
Ah good point...

bwithbroe, 2016-09-29:
@chrisdembia - I'm working on a project that requires simbody as a dependency for something else. For now I can use a lab machine (and the other members of the team can run it fine, since they already compiled simbody). Sooner rather than later would be nice but I don't have any hard deadlines.

chrisdembia, 2016-10-01:
@scpeters and @bwithbroe: Simbody 3.5.4 has been released. This version of Simbody incorporates the fix for Xcode 8 and MacOS10.12.sdk.

daveminh, 2016-10-02:
I just compiled SimBody using cmake for the first time. 

It was on Mac OS X with Xcode Version 8.0 (8A218a) and CMAKE_OSX_SYSROOT set as /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk. 

I got this issue:

Minh-IIT-MBP:simbody-build dminh$ ./ExamplePendulum
dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
  Referenced from: /Users/dminh/Applications/simbody-build/libSimTKsimbody.3.5.dylib (which was built for Mac OS X 10.12)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _clock_gettime
  Referenced from: /Users/dminh/Applications/simbody-build/libSimTKsimbody.3.5.dylib (which was built for Mac OS X 10.12)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
  Referenced from: /Users/dminh/Applications/simbody-build/./simbody-visualizer (which was built for Mac OS X 10.12)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _clock_gettime
  Referenced from: /Users/dminh/Applications/simbody-build/./simbody-visualizer (which was built for Mac OS X 10.12)
  Expected in: /usr/lib/libSystem.B.dylib

Trace/BPT trap: 5

chrisdembia, 2016-10-02:
Thanks for trying this out @daveminh. 

What was your `CMAKE_OSX_DEPLOYMENT_TARGET`? Make sure it was `10.11`. Since the error message says `...built for Mac OS X 10.12`, it seems like your `CMAKE_OSX_DEPLOYMENT_TARGET` was `10.12`.

daveminh, 2016-10-02:
Okay thanks. I just updated to Mac OS X 10.12 and it worked. I didn't set CMAKE_OSX_DEPLOYMENT_TARGET to any value. so I guess it was 10.12.

PR #524 (2016-09-22): Only define clock_gettime() for macOS < 10.12
This PR fixes #523: OSX previously did not provide `clock_gettime()`, but macOS 10.12 now does provide this function. This PR causes Simbody's version of `clock_gettime()` to only appear if using an older version of OSX.

### Background on OSX SDKs, etc.

There are two versions to consider here: the MacOSX SDK version used when compiling the source code, and the version of the OS on a user's computer (called the DEPLOYMENT_TARGET). Both the SDK and the OS version use the same numbering scheme (e.g., 10.10, 10.11, 10.12). A single SDK can be used to [... cut]