simbody/simbody #531: README typo

kidzik, 2016-10-08:
There is

    sudo apt-get install libsimbody-dev libsimbody-doc

should be

    sudo apt-get install libsimbody-dev simbody-doc

scpeters, 2016-10-08:
We host Ubuntu trusty packages at [packages.osrfoundation.org](http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/s/simbody/), and at the time they were built it was called `libsimbody-doc`. Now simbody is in the [official Ubuntu repositories](http://packages.ubuntu.com/source/xenial/simbody) and it is called `simbody-doc`, as you say.

kidzik, 2016-10-08:
Sure, I didn't want to complain, I am just in the process of getting into OpenSim and trying tutorials so when I found a minor confusing type I decided to report/note here.

chrisdembia, 2016-10-08:
Nice catch @kidzik. Thanks for the background, @scpeters. I've submitted a PR to fix this.

scpeters, 2016-10-08:
Yes, thanks for posting @kidzik !

PR #532 (2016-10-08): [readme] libsimbody-doc -> simbody-doc
This PR fixes #531: the name of the `simbody-doc` Ubuntu package was out of date.
closed 2016-10-08 by sherm1