simbody/simbody #535: Add conda as an install option

moorepants, 2016-10-25:
I've built Simbody for [conda](http://conda.pydata.org), a cross platform package manager, through the [conda-forge](https://conda-forge.github.io/) build system. Now, anyone one using Windows, Mac, or Linux can easily install binaries of Simbody if they have conda installed, using identical  commands. This shows how to install and run one of the examples:

```
$ conda config --add channels conda-forge
$ conda install simbody
Fetching package metadata ...........
Solving package specifications: ..........

Package plan for installation in environment /home/moorepants/miniconda3/envs/simbody:

The following NEW packages will be INSTALLED:

    blas:        1.1-openblas            conda-forge
    freeglut:    2.8.1-0                            
    libgfortran: 3.0.0-1                            
    openblas:    0.2.18-5                conda-forge
    simbody:     3.5.3-blas_openblas_201 conda-forge [blas_openblas]

Linking packages ...
[      COMPLETE      ]|#######################################################################################################################################################################################| 100%
$ $CONDA_PREFIX/lib/simbody/examples/ExampleGears
```

Compiling custom code that links to Simbody simply needs to include `$CONDA_PREFIX/lib` and `$CONDA_PREFIX/include/simbody` in the (c)make files. For example, the example in the readme can be run like so:

```
$ cmake .
-- Configuring done
-- Generating done
-- Build files have been written to: /home/moorepants/simbody
$ make
Scanning dependencies of target pendulum
[ 50%] Building CXX object CMakeFiles/pendulum.dir/pendulum.cpp.o
[100%] Linking CXX executable pendulum
[100%] Built target pendulum
$ ./pendulum 
```

With CMakeLists.txt as something like:

```
cmake_minimum_required(VERSION 2.8)
project(simbody_pendulum)

add_executable(pendulum pendulum.cpp)

include_directories($ENV{CONDA_PREFIX}/include/simbody)
link_directories($ENV{CONDA_PREFIX}/lib)
target_link_libraries(pendulum
	$ENV{CONDA_PREFIX}/lib/libSimTKsimbody.so
	$ENV{CONDA_PREFIX}/lib/libSimTKcommon.so
	$ENV{CONDA_PREFIX}/lib/libSimTKmath.so)
```
and `pendulum.cpp` with the C++ source code for the example.

The build recipe is here: https://github.com/conda-forge/simbody-feedstock and the binaries are found here: https://anaconda.org/conda-forge/simbody/.

Maybe you all would like to add this as an install option to your documentation?

Note that I'm still working out a minor issue with the Windows build: https://github.com/conda-forge/simbody-feedstock/pull/4

sherm1, 2016-10-25:
That's awesome, @moorepants! Thanks!

sherm1, 2016-10-25:
Please consider submitting a PR to add this.

chrisdembia, 2016-10-25:
@moorepants thank you. I agree; we should add mention of conda to the README.

> With CMakeLists.txt as something like:

The CMakeLists should be simpler. Specifically, `CONDA_PREFIX` should not appear. The CMakeLists should look as follows:

```
cmake_minimum_required(VERSION 2.8)
project(simbody_pendulum)

find_package(Simbody REQUIRED)

add_executable(pendulum pendulum.cpp)

include_directories(${Simbody_INCLUDE_DIR})
target_link_libraries(pendulum SimTKsimbody)
```

Then, CMake should be invoked like this:

```
cmake <path-to-source-directory> -DCMAKE_PREFIX_PATH=$CONDA_PREFIX
```

The `find_package()` command will search in the `CMAKE_PREFIX_PATH` for files like `SimbodyConfig.cmake` (located in lib/cmake).

moorepants, 2016-10-25:
Thanks. I don't know how to use cmake properly. The above was just the first thing I got to work :)

I'll try yours.

moorepants, 2016-10-25:
I get this error:

```
moorepants@garuda:simbody$ cmake . -DCMAKE_PREFIX_PATH=$CONDA_PREFIX
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "FindSimbody.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Simbody", but
  CMake did not find one.

  Could not find a package configuration file provided by "Simbody" with any
  of the following names:

    SimbodyConfig.cmake
    simbody-config.cmake

  Add the installation prefix of "Simbody" to CMAKE_PREFIX_PATH or set
  "Simbody_DIR" to a directory containing one of the above files.  If
  "Simbody" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/moorepants/simbody/CMakeFiles/CMakeOutput.log".
```

moorepants, 2016-10-25:
Note that my simbody build is 3.5.3.

moorepants, 2016-10-25:
When I submit a PR should I branch off of 3.5.4 or the HEAD of master?

chrisdembia, 2016-10-25:
think the change should be present in both the master and simbody-3.5 branches. If you make a PR into master; I will backport your change to the simbody-3.5 branch.

moorepants, 2016-10-25:
Ok

chrisdembia, 2016-10-25:
Hmm...Is there a SimbodyConfig.cmake inside $CONDA_PREFIX/lib/cmake? Possibly in lib/cmake/simbody or lib/simbody/cmake.

Cool stuff by the way; thanks for all that work! I think language agnostic package managers are better than packages managers for each language. So I like conda more than some of the attempts to make c++-specific package managers.

closed 2017-01-30 by moorepants