simbody/simbody #537: MultibodyGraphMaker incorrectly handles an intermediate massless body

aseth1, 2016-11-14:
This issue was first raised in OpenSim [here](https://github.com/opensim-org/opensim-core/issues/1388)

The translation of a body (pelvis) is defined by two sliders. The first slider mobilizes a massless body and the second slider mobilizes the full mass of the pelvis. Two massive bodies (right and left thighs) mobilized by pins are connected to the pelvis.  

The MultigraphMaker introduces two FreeJoints and two slave bodies (a pair for each thigh)  which seems incorrect :
```
MULTIBODY GRAPH
---------------

10 BODIES:
 0  0: ground mass=inf mob=-1 master=-1
  jointsAsParent=[ 0 7 8]         jointsAsChild=[]        slaves=[]
 1  1: Platform mass=1 mob=0 master=-1
  jointsAsParent=[ 1]     jointsAsChild=[ 0]      slaves=[]
 2  2: Pelvis_massless mass=0 mob=1 master=-1
  jointsAsParent=[ 2]     jointsAsChild=[ 1]      slaves=[]
 3  3: Pelvis mass=0.5 mob=2 master=-1
  jointsAsParent=[ 3 4]   jointsAsChild=[ 2]      slaves=[]
 4  1: LeftThigh mass=0.05 mob=3 master=-1
  jointsAsParent=[ 5]     jointsAsChild=[ 3 7]    slaves=[ 8]
 5  1: RightThigh mass=0.05 mob=5 master=-1
  jointsAsParent=[ 6]     jointsAsChild=[ 4 8]    slaves=[ 9]
 6  2: LeftShank mass=0.05 mob=4 master=-1
  jointsAsParent=[]       jointsAsChild=[ 5]      slaves=[]
 7  2: RightShank mass=0.05 mob=6 master=-1
  jointsAsParent=[]       jointsAsChild=[ 6]      slaves=[]
 8  4: #LeftThigh_slave_1 mass=nan mob=7 master=4
  jointsAsParent=[]       jointsAsChild=[ 3]      slaves=[]
 9  4: #RightThigh_slave_1 mass=nan mob=8 master=5
  jointsAsParent=[]       jointsAsChild=[ 4]      slaves=[]

9 JOINTS:
 0  0:     PlatformToGround               ground->Platform              WeldJoint loopc=-1
 1  1: PelvisMasslessToPlatform             Platform->Pelvis_massless      SliderJoint loopc=-1
 2  2: PelvisToPelvisMassless      Pelvis_massless->Pelvis               SliderJoint loopc=-1
 3  7:       LThighToPelvis               Pelvis->LeftThigh              PinJoint loopc=-1
 4  8:       RThighToPelvis               Pelvis->RightThigh             PinJoint loopc=-1
 5  4:       LShankToLThigh            LeftThigh->LeftShank              PinJoint loopc=-1
 6  6:       RShankToRThigh           RightThigh->RightShank             PinJoint loopc=-1
 7  3:    #ground_LeftThigh               ground->LeftThigh             FreeJoint loopc=-1  ADDED BASE JOINT
 8  5:   #ground_RightThigh               ground->RightThigh            FreeJoint loopc=-1  ADDED BASE JOINT

9 MOBILIZERS:
 0  1:     PlatformToGround               ground->Platform              WeldJoint  0
 1  2: PelvisMasslessToPlatform             Platform->Pelvis_massless      SliderJoint  1
 2  3: PelvisToPelvisMassless      Pelvis_massless->Pelvis               SliderJoint  2
 3  1:    #ground_LeftThigh               ground->LeftThigh             FreeJoint  7
 4  2:       LShankToLThigh            LeftThigh->LeftShank              PinJoint  5
 5  1:   #ground_RightThigh               ground->RightThigh            FreeJoint  8
 6  2:       RShankToRThigh           RightThigh->RightShank             PinJoint  6
 7  4:       LThighToPelvis               Pelvis->#LeftThigh_slave_1     PinJoint  3
 8  4:       RThighToPelvis               Pelvis->#RightThigh_slave_1    PinJoint  4

0 LOOP CONSTRAINTS:

----- END OF MULTIBODY GRAPH.
```

To verify that the issue is related to the initial massless slider body,  I added mass to the first slider body to yield the expected graph:
```
7 JOINTS:
 0  0:     PlatformToGround               ground->Platform              WeldJoint loopc=-1
 1  1: PelvisMasslessToPlatform             Platform->Pelvis_massless      SliderJoint loopc=-1
 2  2: PelvisToPelvisMassless      Pelvis_massless->Pelvis               SliderJoint loopc=-1
 3  3:       LThighToPelvis               Pelvis->LeftThigh              PinJoint loopc=-1
 4  4:       RThighToPelvis               Pelvis->RightThigh             PinJoint loopc=-1
 5  5:       LShankToLThigh            LeftThigh->LeftShank              PinJoint loopc=-1
 6  6:       RShankToRThigh           RightThigh->RightShank             PinJoint loopc=-1

7 MOBILIZERS:
 0  1:     PlatformToGround               ground->Platform              WeldJoint  0
 1  2: PelvisMasslessToPlatform             Platform->Pelvis_massless      SliderJoint  1
 2  3: PelvisToPelvisMassless      Pelvis_massless->Pelvis               SliderJoint  2
 3  4:       LThighToPelvis               Pelvis->LeftThigh              PinJoint  3
 4  4:       RThighToPelvis               Pelvis->RightThigh             PinJoint  4
 5  5:       LShankToLThigh            LeftThigh->LeftShank              PinJoint  5
 6  5:       RShankToRThigh           RightThigh->RightShank             PinJoint  6

0 LOOP CONSTRAINTS:
```

sherm1, 2016-12-13:
Thanks @aseth1 -- definitely looks wrong. I will investigate (soon, hopefully!).

aseth1, 2016-12-13:
That would be great! Thank you @sherm1

sherm1, 2017-12-01:
Fixed in #592.

closed 2017-12-01 by sherm1