simbody/simbody #543: Segmentation fault when creating Inertia object

david-haber, 2016-12-23:
We ran into a segmentation fault when executing the following code on Debian 8.5 and Simbody 3.4 (installed from apt-get).

rigid_test.cc:
```
#include "Simbody.h"

int main() {
  SimTK::Inertia in(1.0);
  return 0;
}
```

We used the following compiler command:

`g++ -std=c++11 -std=gnu++11 -isystem /usr/include/simbody -Wall -g -Wfatal-errors -c -o rigid_test.o rigid_test.cc
g++ -L/usr/lib/ -lSimTKsimbody -lSimTKmath -lSimTKcommon  rigid_test.o lib.a -o rigid_test`

and receive the following error:

```
Running rigid_test: Segmentation fault
../mk/Makefile.inc:65: recipe for target 'rigid_test.run' failed
make: *** [rigid_test.run] Error 139

0x00007ffff6bfc401 in ?? () from /usr/lib/x86_64-linux-gnu/libSimTKcommon.so.3.4
(gdb) bt
#0  0x00007ffff6bfc401 in ?? () from /usr/lib/x86_64-linux-gnu/libSimTKcommon.so.3.4
#1  0x00007ffff7dea9ba in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe6b8, env=env@entry=0x7fffffffe6c8) at dl-init.c:78
#2  0x00007ffff7deaaa3 in call_init (env=0x7fffffffe6c8, argv=0x7fffffffe6b8, argc=1, l=<optimized out>) at dl-init.c:36
#3  _dl_init (main_map=0x7ffff7ffe1a8, argc=1, argv=0x7fffffffe6b8, env=0x7fffffffe6c8) at dl-init.c:126
#4  0x00007ffff7ddd1ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000001 in ?? ()
#6  0x00007fffffffe8af in ?? ()
#7  0x0000000000000000 in ?? ()
```

Our compiler version is: g++ (Debian 4.9.2-10) 4.9.2

This code seems quite trivial so we are wondering what is going on.

sherm1, 2016-12-23:
The apt-gettable binary for Simbody 3.4 was built using C++03. The gcc ABIs for C++03 and C++11 are not compatible, so it is *possible* that specifying c++11 here is the problem -- try it without that. If you want to use Simbody with C++11 you'll need to build Simbody from source with -std=c++11 using the most recent release, 3.5.4 which you can get [here](https://github.com/simbody/simbody/releases). Build instructions are provided in the `README.md` file.

However, the compile command you used may also be causing a problem. On Ubuntu 16.04 (a Debian descendant) I obtained Simbody with this command:
```
 sudo apt-get install libsimbody-dev simbody-doc
```
which gave me Simbody version 3.5.3, which I believe is also a C++03 binary.

I then compiled this version of your code in a file t.cpp:
```c++
#include <iostream>
#include "Simbody.h"

int main() {
  SimTK::Inertia in(1.0);
  std::cout << in << std::endl;
  return 0;
}
```
with this command line:
```
g++ -I/usr/include/simbody t.cpp -lSimTKsimbody -lSimTKmath -lSimTKcommon
``` 
(using g++ 5.4.0) and got the expected output when I ran `a.out` afterwards:
```
[1,0,0]
[0,1,0]
[0,0,1]
```
But ... this also worked fine when I compiled it with `-std=c++11` or `-std=c++14`. I don't know what to make of that, but I suggest you try using the simplified compile line here to see if there may be some problem other than ABI compatibility.

chrisdembia, 2016-12-24:
What if you use a later version of Simbody from debian (seems like 3.5 is in the testing and unstable distributions (https://packages.debian.org/search?keywords=simbody&searchon=names&suite=all&section=all).