simbody/simbody #544: Segmentation fault in realizeTopology

david-haber, 2016-12-23:
We get a segmentation fault in realizeTopology() when executing the following code on Debian 8.5 and with Simbody 3.5.3 built from source.

minimum_example0_test.cc:

```
#include "Simbody.h"

#include <cstdio>

#include "dcmotor_simplified.hh"

int main() {
    SimTK::MultibodySystem        system;
    SimTK::SimbodyMatterSubsystem matter(system);
    SimTK::GeneralForceSubsystem  forces(system);

    SimTK::MobilizedBody::Pin rotor(matter.Ground(), SimTK::Body::Rigid());

    std::cout << "Before motor" << std::endl;
    DCMotorTest motor(forces, rotor);

    std::cout << "Before realizeTopology" << std::endl;
    system.realizeTopology();

    std::cout << "After realizeTopology" << std::endl;

    return 0;
}
```

dcmotor_simplified.hh:

```
#pragma once

#include "Simbody.h"

class DCMotorTest : public SimTK::Force::Custom {
public:
	DCMotorTest(SimTK::GeneralForceSubsystem &forces, const SimTK::MobilizedBody &mb);
};
```

dcmotor_simplified.cc:

```
#include "dcmotor_simplified.hh"

#include <cmath>

class MotorImplTest : public SimTK::Force::Custom::Implementation {
public:
	MotorImplTest(SimTK::SimbodyMatterSubsystem &f) : m_matter(f){};

	MotorImplTest(const SimTK::MobilizedBody &mb) : m_matter(mb.getMatterSubsystem()), m_mobod(mb) {}

	void calcForce(const SimTK::State &state,
	               SimTK::Vector_< SimTK::SpatialVec > & /*bodyForces*/,
	               SimTK::Vector_< SimTK::Vec3 > & /*particleForces*/,
	               SimTK::Vector &mobilityForces) const {}

	SimTK::Real calcPotentialEnergy(const SimTK::State & /*state*/) const { return 0; }

	void setVoltage(SimTK::State &state, SimTK::Real v) const {
		std::cerr << "m_voltageIX in set Voltage" << m_voltageIx << std::endl;
		state.updDiscreteVariable(SimTK::SubsystemIndex(0), m_voltageIx);
	}

	void realizeTopology(SimTK::State &state) const {
		std::cout << "Called realizeTopology in MotorImpl" << std::endl;
		m_voltageIx =
		    m_matter.allocateDiscreteVariable(state, SimTK::Stage::Acceleration, new SimTK::Value< SimTK::Real >(0.0));
		std::cout << "m_voltageIx: " << m_voltageIx << std::endl;
	}

private:
	const SimTK::SimbodyMatterSubsystem &m_matter;
	SimTK::MobilizedBody                 m_mobod;
	mutable SimTK::DiscreteVariableIndex m_voltageIx;
};

DCMotorTest::DCMotorTest(SimTK::GeneralForceSubsystem &forces, const SimTK::MobilizedBody &mb)
    : SimTK::Force::Custom(forces, new MotorImplTest(mb)) {}

void Foo(SimTK::State &state) {
	SimTK::SimbodyMatterSubsystem f;
	MotorImplTest                 t(f);
	t.setVoltage(state, 1.0);
}
```

We used the following commands for compilation:

`g++ -std=c++11 -std=gnu++11 -isystem /usr/local/include/simbody -I/home/xxx/repos/exp-simmc -Wall -g  -Wfatal-errors -c -o dcmotor_simplified.o dcmotor_simplified.cc`
`ar -v -q lib.a dcmotor_simplified.o
ar: creating lib.a
a - dcmotor_simplified.o`
`g++ -std=c++11 -std=gnu++11 -isystem /usr/local/include/simbody -I/home/xxx/repos/exp-simmc -Wall -g  -Wfatal-errors -c -o minimum_example0_test.o minimum_example0_test.cc`
`g++ -L/local/usr/lib/ -lSimTKsimbody -lSimTKmath -lSimTKcommon  minimum_example0_test.o lib.a -o minimum_example0_test`

This is the output:

```
Running minimum_example0_test: Before motor
Before realizeTopology
Segmentation fault
../mk/Makefile.inc:65: recipe for target 'minimum_example0_test.run' failed
make: *** [minimum_example0_test.run] Error 139

(gdb) bt
#0  SimTK::CacheEntryInfo::registerWithPrerequisites (this=0x61c518, stateImpl=...) at /home/dh/repos/vendor-simbody/SimTKcommon/Simulation/src/State.cpp:1009
#1  0x00007ffff7afc3cb in SimTK::StateImpl::allocateCacheEntryWithPrerequisites (this=0x615090, subsys=..., earliest=..., latest=..., latest@entry=..., qPre=qPre@entry=true, uPre=uPre@entry=false, zPre=false, discreteVars=..., cacheEntries=..., value=0x61cff0)
    at /home/dh/repos/vendor-simbody/SimTKcommon/Simulation/include/SimTKcommon/internal/StateImpl.h:1213
#2  0x00007ffff7aee687 in allocateCacheEntryWithPrerequisites (value=<optimized out>, cacheEntries=..., discreteVars=..., z=false, u=false, q=true, latest=..., earliest=..., subsys=..., this=0x614ff0) at /home/dh/repos/vendor-simbody/SimTKcommon/Simulation/include/SimTKcommon/internal/StateImpl.h:2470
#3  SimbodyMatterSubsystemRep::realizeSubsystemTopologyImpl (this=0x616770, s=...) at /home/dh/repos/vendor-simbody/Simbody/src/SimbodyMatterSubsystemRep.cpp:394
#4  0x00007ffff6de3f92 in SimTK::Subsystem::Guts::realizeSubsystemTopology (this=0x616770, s=...) at /home/dh/repos/vendor-simbody/SimTKcommon/Simulation/src/Subsystem.cpp:209
#5  0x00007ffff7991298 in SimTK::MultibodySystemRep::realizeTopologyImpl (this=0x614ef0, s=...) at /home/dh/repos/vendor-simbody/Simbody/src/MultibodySystem.cpp:185
#6  0x00007ffff6deb62d in SimTK::System::Guts::realizeTopology (this=0x614ef0) at /home/dh/repos/vendor-simbody/SimTKcommon/Simulation/src/System.cpp:358
#7  0x0000000000404108 in main () at minimum_example0_test.cc:38
```

Our compiler version is: g++ (Debian 4.9.2-10) 4.9.2

Could anyone help?

traversaro, 2016-12-23:
I may be wrong, but this seems to be related to a mismatch between the Simbody ABI of the compiled version of Simbody and the ABI as seen by your code. 

I don't know if Simbody 3.5.3 can be compiled with C++98, but just to be sure, with which option did you compiled Simbody 3.5.3 from source? If for some reason Simbody 3.5.3 is compiled using C++98, this could cause problems due to the mismatch of the (default) ABI between C++11 and C++98 in GCC 4.x , see https://gcc.gnu.org/wiki/Cxx11AbiCompatibility .  This has been solved in GCC 5.

sherm1, 2016-12-23:
The method `registerWithPrerequisites()` does not exist in Simbody 3.5.3, and line State.cpp:1009 doesn't exist either since that file has 883 lines in that release. It looks to me that you are actually linked with the Simbody master branch code, which has version number 3.6. That suggests some confusion about what is being built, or possibly that the linker is finding libraries other than the ones you intended. Although you should be able to use 3.6, you may want to build with a released version (3.5.4 is the most current) and make sure that you are getting what you expect. 

Also please see my comments in issue #543. You definitely want to get that working first!