simbody/simbody #545: Debian installation of Simbody 3.5.3 doesn't include visualizer or examples

sherm1, 2016-12-23:
On Ubuntu 16.04 I installed Simbody with
```
sudo apt-get install libsimbody-dev simbody-doc
```
The installation worked fine when I used the libraries (see issue #543). However, I couldn't find the `simbody-visualizer` executable or the example programs so I couldn't do anything with it other than link.

Also the documents installed in /usr/share/doc/simbody were still gzipped after installation.

@j-rivero and @chrisdembia, is this expected behavior?

chrisdembia, 2016-12-24:
What about `sudo apt-get install libsimbody`? Or maybe `sudo apt-get install libsimbody3.5v5`? See [here](https://packages.debian.org/search?keywords=simbody&searchon=names&suite=all&section=all). I suspect this would not fix the gzipped documents though.

sherm1, 2016-12-24:
Thanks, Chris. I tried those. Apparently `libsimbody` doesn't exist, and when I asked for `libsimbody3.5v5` it said "already installed" so I think it must have come along with `libsimbody-dev`.

sherm1, 2016-12-25:
> Try installing simbody-doc

Thanks, Steve. I did install that along with libsimbody-dev. It provided the gzipped documenation I mentioned but apparently not the examples.

scpeters, 2016-12-26:
Sorry, I didn't read carefully enough!

scpeters, 2016-12-26:
I think this is the debian metadata for what gets installed by the `simbody-doc` package:

* https://anonscm.debian.org/cgit/debian-science/packages/simbody.git/tree/debian/simbody-doc.install

YuriOsokin, 2017-06-04:
Hello, will this problem be fixed in the near future?

sherm1, 2017-06-04:
> Hello, will this problem be fixed in the near future?

Likely not unless @j-rivero is working on it. Building from source is the best option at the moment.

moorepants, 2017-06-04:
It is also possible to install using conda on a Debian system. The conda packages include the visualizer.

chrisdembia, 2017-06-04:
Thanks @moorepants. @YuriOsokin, see https://github.com/simbody/simbody#windows-mac-and-linux-using-conda

chrisdembia, 2018-11-20:
I just installed a fresh copy of 18.04 and had no issue building the visualizer with the following commands:

```
   10  sudo apt install git
   12  git clone https://github.com/simbody/simbody
   14  cd simbody/
   15  mkdir build
   17  sudo apt install cmake
   19  cd build/
   21  sudo apt install liblapack-dev freeglut3-dev libxi-dev libxmu-dev doxygen
   22  cmake ../
   24  make simbody-visualizer
   25  ./simbody-visualizer 
```