simbody/simbody #550: Issue running simbody on a headless server

spMohanty, 2017-02-20:
Hi,

So I have been trying to run simbody (via OpenSim) on a headless server, and I keep getting this error :
```
(opensim-rl) ubuntu@ip-172-30-0-203:~/SDF/mohanty/osim-rl-grader/worker_dir$ xpra start :108
(opensim-rl) ubuntu@ip-172-30-0-203:~/SDF/mohanty/osim-rl-grader/worker_dir$ Entering daemon mode; any further errors will be reported to:
  /home/ubuntu/.xpra/:108.log

(opensim-rl) ubuntu@ip-172-30-0-203:~/SDF/mohanty/osim-rl-grader/worker_dir$
(opensim-rl) ubuntu@ip-172-30-0-203:~/SDF/mohanty/osim-rl-grader/worker_dir$ DISPLAY=:108 python simulate.py 127.0.0.1 6379 4f2b61e1aaf03d3283f135febbe225a4___c56b7ae556 alksjdhlj lkjlkjlkj lkjlkjk
127.0.0.1 6379 4f2b61e1aaf03d3283f135febbe225a4___c56b7ae556 alksjdhlj lkjlkjlkj lkjlkjk
Updating Model file from 30000 to latest format...
Loaded model gait9dof18musc_Thelen_BigSpheres.osim from file /home/ubuntu/anaconda2/envs/opensim-rl/lib/python2.7/site-packages/osim/env/../models/gait9dof18musc.osim
pelvis
femur_r
tibia_r
talus_r
calcn_r
toes_r
femur_l
tibia_l
talus_l
calcn_l
toes_l
torso
head
Movie Name
0.0029046360869
0.00562152964994
0.00849470496178
0.0132510904223
0.0175715312362
0.0190468281507
std::exception in 'bool OpenSim::Manager::integrate(SimTK::State &,double)': SimTK Exception thrown at VisualizerProtocol.cpp:378:
  Error detected by Simbody method VisualizerProtocol: An attempt to write() 1 bytes to pipe 5 failed with errno=32 (Broken pipe).
  (Required condition 'status!=-1' was not met.)

-500
terminate called after throwing an instance of 'SimTK::Exception::ErrorCheck'
  what():  SimTK Exception thrown at VisualizerProtocol.cpp:372:
  Error detected by Simbody method VisualizerProtocol: An attempt to write() 1 bytes to pipe 5 failed with errno=32 (Broken pipe).
  (Required condition 'status!=-1' was not met.)

Aborted (core dumped)
```

I tried using both xvfb and xpra, the issue still remains.

The goal is to use a slightly modified version of simbody-visualiser to automatically save the movie at a particular location (which it obtains from an env variable)

Any help or pointers will be really appreciated.

chrisdembia, 2017-02-20:
Seems like the same issue that #548 tries to address: destructing the visualizer causes that exception to be thrown if the simbody-visualizer has already been closed. Consider trying the change in #548.

As for modifying simbody-visualizer to save the movie, I cannot help. You can alternatively save the states to a file (using OpenSim's `Manager::getStatesStorage()`) and then write a new program that can visualize the previous simulation.

spMohanty, 2017-02-21:
Hi @chrisdembia,
Thanks for the quick reply.

As it turns out, simply by starting xvpr using : 
```
    xpra --xvfb="Xorg -noreset -nolisten tcp +extension GLX \
    -config /etc/xpra/xorg.conf \
    +extension RANDR +extension RENDER -logfile ${HOME}/.xpra/Xorg-10.log" \
    start :100    
```

and then passing the new DISPLAY variable worked for me.
Although it does occassionally crash now and then.

And as for the saving of the movie, I directly edited simbody-visualizer.cpp to save the movie everytime to a filename that it can obtain from an env variable. And then I replace the bundled simbody-visualizer binary with the new compiled binary. Seems to work for me.

Thanks for the help though

chrisdembia, 2017-02-21:
Great. I am closing this issue then. Please re-open if necessary.

closed 2017-02-21 by chrisdembia

spMohanty, 2017-02-21:
Btw, I am still getting stray broken pipe errors now and then :( !!
I pulled the referenced commit, and recompiled the binary, still the same. About 1 in 10 runs crash.

sherm1, 2017-02-21:
@spMohanty, the visualizer has been plagued by pipe problems on Ubuntu for ages. If you can figure out why that would be great!