simbody/simbody #556: simbody 3.5.3 compiling error in "usr/local/include/.." when building example on ubuntu 14.04

jiangfeiyi, 2017-03-28:
I just installed simbody 3.5.3 on ubuntu 14.04, and tried to build example:
~/simbody/examples/ExampleGears.cpp
using g++:
`$ g++ ExampleGears.cpp`

Then I got the error as below:

> /usr/local/include/SimTKcommon/internal/MeasureImplementation.h:577:25: error: call of overloaded ‘Value(double&)’ is ambiguous
                         ? this->getSubsystem().allocateCacheEntry
                         ^
/usr/local/include/SimTKcommon/internal/MeasureImplementation.h:577:25: note: candidates are:
In file included from /usr/local/include/SimTKcommon/basics.h:55:0,
                 from /usr/local/include/SimTKcommon.h:32,
                 from /usr/local/include/Simbody.h:38,
                 from ExampleGears.cpp:24:
/usr/local/include/SimTKcommon/internal/Value.h:126:5: note: SimTK::Value<T>::Value(const SimTK::Value<T>&) [with T = double]
     Value(const Value& value) : m_thing(value.m_thing) {}
     ^
/usr/local/include/SimTKcommon/internal/Value.h:120:14: note: SimTK::Value<T>::Value(T) [with T = double]
     explicit Value(const T&& value) : m_thing(std::move(value)) {}
              ^
/usr/local/include/SimTKcommon/internal/Value.h:115:14: note: SimTK::Value<T>::Value(const T&) [with T = double]
     explicit Value(const T& value) : m_thing(value) {}

I have no idea what cause this compiler error.
Is there anybody got some ideas about it?
Thanks for your help.

chrisdembia, 2017-03-28:
Could you provide the complete command you used? Did you use the `-std=c++11` flag?

jiangfeiyi, 2017-03-29:
thanks very much.
I solved this by using cmake.

closed 2017-03-29 by jiangfeiyi

chrisdembia, 2017-03-29:
+1 for CMake!