simbody/simbody #558: Add relative location from libSimTKsimbody.(so|dylib) to simbody-visualizer to the visualizer search path

chrisdembia, 2017-03-31:
PR #64 cleverly inserted the build-time install location of the `simbody-visualizer` to the visualizer search path. However, for OpenSim users who get a binary distribution of OpenSim, this build-time absolute path will be incorrect. This is mostly a problem on macOS (on Windows, `simbody-visualizer` will be on the PATH for OpenSim users; on Linux, hopefully OpenSim will be installed into common system-wide directories), but is also a problem for Linux users installing into non-standard directories. 

The specific use cases I am thinking of are OpenSim MATLAB and Python bindings.

When using Simbody's shared libraries, the code that spawns the visualizer is always in libSimTKsimbody.(so|dylib). We also always know the relative location of the simbody-visualizer from libSimTKsimbody.(so|dylib) (so long as the installation is not reorganized). If we know the location of libSimTKsimbody.(so|dylib), then we can find the simbody-visualizer by combining this known location with the relative path of simbody-visualizer from SimTKsimbody. Fortunately, on most UNIX operating systems, it is possible to obtain the location of the shared library that has been loaded (http://stackoverflow.com/questions/1681060/library-path-when-dynamically-loaded):

```cpp
#include <dlfcn.h>
__attribute__((constructor))
void on_load(void) {
    Dl_info dl_info;
    dladdr((void *)on_load, &dl_info);
    fprintf(stderr, "module %s loaded\n", dl_info.dli_fname);
}
```

I have tested this mechanism on macOS, and I have gotten it to work with OpenSim MATLAB and Python bindings.

@sherm1 , would you be interested in a PR that adds `getParentDir(dl_info.dli_fname) + "/" + rel_path_of_simbody_visualizer_from_lib_dir` (pseudocode) to the visualizer search path on platforms where `dlfcn.h` is available?

On Windows, or when using static libraries (on any platform), I don't think it's possible to use this mechanism. Therefore, for Windows, it would be nice if the search path included the PATH environment variable. This would allow OpenSim to decrease its reliance on the OPENSIM_HOME environment variable, which we are hoping we will no longer need to set.

chrisdembia, 2017-04-03:
@sherm1 if you would prefer that I do not make this change in Simbody, I can put it in OpenSim instead.

sherm1, 2017-04-03:
(Sorry for the reply delay, Chris.) This sounds like a generally useful capability. Would it make sense to add a method to Simbody's `Pathname` class that would take a function address and return the pathname of the library it was loaded from? It could return some indication of whether the operation is supported and if so we could call it and add the result to the default visualizer search path.

chrisdembia, 2017-04-03:
Ah good idea. I will do that. Thank you.

PR #559 (2017-04-04): Visualizer relative search path
This PR fixes #558: the path from the SimTKsimbody library to the simbody-visualizer is added as a search path in VisualizerProtocol. To facilitate this, I added the function `Pathname::getFunctionLibraryDirectory()`. I also added a small test, and updated the documentation.

I tested this locally on both macOS and Ubuntu 14.04. I was able to rename the install directory and successfully get ChainExample to spawn the visualizer; this behavior was not possible before this PR. I also tested this PR with OpenSim on macOS and was able to verify that OpenSim's MATLAB [... cut]
closed 2017-04-04 by sherm1