simbody/simbody #561: Getting more done in GitHub with ZenHub

shahidul56, 2017-04-11:
Hola! @shahidul56 has created a [ZenHub](http://www.zenhub.com) account for the **simbody** organization. ZenHub is the only project management tool integrated natively in GitHub – created specifically for fast-moving, software-driven teams.

----

#### How do I use ZenHub?

To get set up with ZenHub, all you have to do is **[download the browser extension](https://www.zenhub.com?utm_source=ZHOnboarding)** and log in with your GitHub account. Once you do, you’ll get access to ZenHub’s complete feature-set immediately.

#### What can ZenHub do?

ZenHub adds a series of enhancements directly inside the GitHub UI:

- Real-time, customizable task boards for GitHub issues;
- Multi-Repository burndown charts, estimates, and velocity tracking based on GitHub Milestones;
- Personal to-do lists and task prioritization;
- Time-saving shortcuts – like a quick repo switcher, a “Move issue” button, and much more.

### [Add ZenHub to GitHub](https://www.zenhub.com?utm_source=ZHOnboarding)

_Still curious? See [more ZenHub features](https://www.zenhub.com/features?utm_source=ZHOnboarding) or read [user reviews](https://chrome.google.com/webstore/detail/zenhub-for-github/ogcgkffhplmphkaahpmffcafajaocjbd/reviews). This issue was written by your friendly ZenHub bot, posted by request from @shahidul56._

[image]

chrisdembia, 2017-09-20:
@sherm1 did you have anything to do with this?  I'm not sure how someone who is not part of the simbody organization could create a ZenHub account for it. Perhaps we should close this issue.

sherm1, 2017-09-20:
Nope. Above it says "This issue was written by your friendly ZenHub bot, posted by request from @shahidul56." Looks like spam. Thanks for pointing it out. I don't think it actually authorized a ZenHub account -- the image above is boilerplate, not Simbody.

closed 2017-09-20 by sherm1