simbody/simbody #563: Simbody crashing over and over again

spMohanty, 2017-05-22:
Hi,

I keep getting this error multiple times when running in visualization mode.
```
Traceback (most recent call last):
  File "/home/ubuntu/SDF/osim-rl-grader/worker_dir/simulate.py", line 38, in <module>
    env = RunEnv(True)
  File "/home/ubuntu/osim-rl/osim/env/run.py", line 65, in __init__
    manager.integrate(state)
  File "/home/ubuntu/anaconda2/envs/opensim-rl/lib/python2.7/site-packages/opensim/simulation.py", line 38437, in integrate
    return _simulation.Manager_integrate(self, s, dtFirst)
RuntimeError: std::exception in 'bool OpenSim::Manager::integrate(SimTK::State &,double)': SimTK Exception thrown at VisualizerProtocol.cpp:490:
  Error detected by Simbody method VisualizerProtocol: An attempt to write() 11160 bytes to pipe 6 failed with errno=32 (Broken pipe).
  (Required condition 'status!=-1' was not met.)

terminate called after throwing an instance of 'SimTK::Exception::ErrorCheck'
  what():  SimTK Exception thrown at VisualizerProtocol.cpp:372:
  Error detected by Simbody method VisualizerProtocol: An attempt to write() 1 bytes to pipe 6 failed with errno=32 (Broken pipe).
  (Required condition 'status!=-1' was not met.)

Aborted (core dumped)
```

Anyone familiar with possible reasons why ?
I am using a slightly modified version of simbody (https://github.com/spMohanty/simbody) to be able to force generation of frames on every render.

will really appreciate any insights.

Thanks,

chrisdembia, 2017-05-22:
Thank you for reporting @spMohanty . Does this occur in the middle of a simulation, or at the end? Are you sure this error occurs without your modified version of Simbody?

spMohanty, 2017-05-22:
Hi @chrisdembia,
Thanks for the really quick response. 

And I was stuck with this problem for quite sometime, so I forgot to add some important context, that I have been trying to get it to run with a `xpra` fake display server. I have updated the main question to include this now. 

And I don't think it has anything to do with my modifications, as my fork of the repository runs perfectly fine when I run it locally.

Thanks

chrisdembia, 2017-05-22:
> And I don't think it has anything to do with my modifications, as my fork of the repository runs perfectly fine when I run it locally.

Could you provide more details about the exact scenario where the issue occurs? You are trying to render the frames on a machine other than the one performing the simulation?

spMohanty, 2017-05-22:
@chrisdembia : No, the simulations run a headless server which doesnot have a physical display attached. 
The goal is to be able to generate an animation (GIF) for every simulation. So I started by forcing the saving of all the frames in every simulation. But that did not work out of the box, as glut could not find a display.

So, I setup a fake display using xpra (https://wiki.archlinux.org/index.php/Xpra) on `localhost:100`.
Then I forced the `DISPLAY` variable to point to `:100` and in many cases everything works as expected, while in many cases it just crashes with the referenced error. 

If there is a better way to programmatically generate an animation of the simulation on a headless server, I would be happy to try that instead.

sherm1, 2017-05-22:
Is this different than what you were seeing in #550 ?

spMohanty, 2017-05-23:
@sherm1 : Looking at my own description of the problem from a few months ago, it seems this issue is tied up with both xvfb and xpra. I was frustrated in getting it to run with xpra using a particular xpra config, but it has been really unstable even with that.
And thanks, I will try to investigate more on the cause of the broken pipe issues, and will report back on this thread if I find anything interesting.
Also, any directed insights on the topic will be really helpful and appreciated.
Thanks

chrisdembia, 2017-12-15:
I'm getting the same error on Ubuntu 16.04:

```
terminate called after throwing an instance of 'SimTK::Exception::ErrorCheck'
  what():  SimTK Exception thrown at VisualizerProtocol.cpp:436:
  Error detected by Simbody method VisualizerProtocol: An attempt to write() 1 bytes to pipe 6 failed with errno=32 (Broken pipe).
  (Required condition 'status!=-1' was not met.)
```

sherm1, 2017-12-15:
I would really like to understand this. The Simbody visualizer has had intermittent problems with pipe writing on Ubuntu for ages. :cry:

ShravanTata, 2018-11-29:
Is this issue fixed yet? I am having similar crashes on Ubuntu 16.04

sherm1, 2018-11-29:
This still is not fixed. We haven't been able to reproduce it reliably in an environment where it can be debugged. Any help would be greatly appreciated!

ShravanTata, 2018-11-29:
Sure, I would be happy help debug the issue. Would a small example that would produce this issue on my local machine be sufficient to start with?

sherm1, 2018-11-29:
That might be a good start. Would be great if it occurred in a Debug build, or if you could add some instrumentation to Simbody to reveal what's happening. The pipe code Simbody uses to talk with the visualizer _seems_ very straightforward.

cmoxiv, 2019-08-31:
A bit late to the party. The same issue happens with me when running osim-rl on a virtual machine running Ubuntu 18. I thought it may be a virtual machine issue. It works fine for a while but sometimes when terminating the training it goes into producing this error. I have to restart the virtual machine to get it running again.

frederico-klein, 2022-02-21:
So I've been hit with this and I unfortunately managed to cause it rather consistently. 

This happens when I try to use Xming 32bit (or vcXsrv 64-bit) with Windows 10 to display output from a docker instance containing OpenSim. 

To get the docker: `docker pull mysablehats/opensim-rt:main` or follow the instructions from this repository https://github.com/frederico-klein/docker-opensimrt/tree/main 

To recreate the bug, simply run the docker with Xming and appropriate DISPLAY settings (I used as the DISPLAY, the IP from the Windows Subsystem from Linux, which you can get from running `ipconfig`, another option is to open Xming's log file and look which ip it is using) and run: ` ./TestLowerLimbIMUIKFromFile` 

The default failure behavior is actually just to end the program silently, which is really odd. To get the sigpipe, you either need to add this to `/opensimrt/OpenSimRT/IMU/applications/OnlineLowerLimbIMUIK.cpp`

    #include <signal.h>
    signal(SIGPIPE, SIG_IGN);

(from here https://stackoverflow.com/questions/108183/how-to-prevent-sigpipes-or-handle-them-properly) and build it again (you can use my build.sh script) or run the file using gdb. 

I believe the crashing is due to this is this error. Note that this is a very strange error and it took me a while to find. It works sometimes and a very similar demo from the same project `./TestUpperLimbIMUIKFromFile` always works (!).

Output when it works:
```
$ gdb TestLowerLimbIMUIKFromFile
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from TestLowerLimbIMUIKFromFile...done.
(gdb) b 115
Breakpoint 1 at 0x22142: file /opensimrt/OpenSimRT/IMU/tests/TestLowerLimbIMUIKFromFile.cpp, line 115.
(gdb) run
Starting program: /opensimrt/build/TestLowerLimbIMUIKFromFile
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Loaded model model_scaled from file /opensimrt/data/gait1992/scale/model_scaled.osim
[New Thread 0x7f3f4508e700 (LWP 509)]
Recording Static Pose...
[New Thread 0x7f3f4488d700 (LWP 523)]

Thread 1 "TestLowerLimbIM" hit Breakpoint 1, run () at /opensimrt/OpenSimRT/IMU/tests/TestLowerLimbIMUIKFromFile.cpp:116
116                 visualizer.update(pose.q);
```

Output from gdb when it fails:

````
(gdb) b 115
Breakpoint 1 at 0x22142: file /opensimrt/OpenSimRT/IMU/tests/TestLowerLimbIMUIKFromFile.cpp, line 115.
(gdb) run
Starting program: /opensimrt/build/TestLowerLimbIMUIKFromFile
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Loaded model model_scaled from file /opensimrt/data/gait1992/scale/model_scaled.osim
[New Thread 0x7f26dd2ef700 (LWP 463)]
Recording Static Pose...
[New Thread 0x7f26dcaee700 (LWP 477)]

Thread 1 "TestLowerLimbIM" hit Breakpoint 1, run ()
    at /opensimrt/OpenSimRT/IMU/tests/TestLowerLimbIMUIKFromFile.cpp:116
116                 visualizer.update(pose.q);
(gdb) n
[Thread 0x7f26dcaee700 (LWP 477) exited]

Thread 1 "TestLowerLimbIM" received signal SIGPIPE, Broken pipe.
0x00007f26dfc93257 in __GI___libc_write (fd=6, buf=0x7fffb669dcb3, nbytes=1) at ../sysdeps/unix/sysv/linux/write.c:27
27      ../sysdeps/unix/sysv/linux/write.c: No such file or directory.
````

Note that this docker implementation runs this code just fine **100% of the time in Debian 10**. 

If you need any more information, please ask. This was only tested in one windows system and it may me harder to replicate.

sherm1, 2022-02-21:
This sounds like an issue with OpenSim and should be reported there:
GitHub: https://github.com/opensim-org/opensim-core
Forum: https://simtk.org/plugins/phpBB/indexPhpbb.php?f=91

closed 2022-02-21 by sherm1