simbody/simbody #564: Tests and examples do not run

YuriOsokin, 2017-06-05:
Hello,

I have downloaded the source code, build it with standard options with "make". It went well. Then I ran the "./SimbodyInstallTest". It opened the visualizer, draw the pendulum and stopped. Closing the visualizer window lead to following error in terminal "freeglut  ERROR:  Function <glutPostRedisplay> called without first calling 'glutInit'".
In example Pendulum in the SimBody build folder the visualizer window hangs up too. Closing the window leads to the following error: "freeglut (simbody-visualizer): ERROR: Function <glutPostRedisplay> called with no current window defined." Then it prints the backtrace and then memory map.  The backtrace as is follows:
*** Error in `simbody-visualizer': corrupted double-linked list: 0x09bdc218 ***
======= Backtrace: =========
/lib/i386-linux-gnu/libc.so.6(+0x67377)[0xb6fbb377]
/lib/i386-linux-gnu/libc.so.6(+0x6d2f7)[0xb6fc12f7]
/lib/i386-linux-gnu/libc.so.6(+0x6e326)[0xb6fc2326]
/usr/lib/i386-linux-gnu/mesa/libGL.so.1(+0x4090f)[0xb767e90f]
/usr/lib/i386-linux-gnu/mesa/libGL.so.1(glXDestroyContext+0x111)[0xb76530b1]
/usr/lib/i386-linux-gnu/libglut.so.3(fgCloseWindow+0x44)[0xb7744364]
/usr/lib/i386-linux-gnu/libglut.so.3(fgDestroyWindow+0xa8)[0xb7742b18]
/usr/lib/i386-linux-gnu/libglut.so.3(fgDestroyMenu+0xef)[0xb7742cbf]
/usr/lib/i386-linux-gnu/libglut.so.3(fgDestroyStructure+0x30)[0xb7742d70]
/usr/lib/i386-linux-gnu/libglut.so.3(fgDeinitialize+0x70)[0xb773c100]
/usr/lib/i386-linux-gnu/libglut.so.3(fgError+0xdf)[0xb773ea6f]
/usr/lib/i386-linux-gnu/libglut.so.3(glutPostRedisplay+0x75)[0xb77346b5]
simbody-visualizer(_Z14listenForInputPv+0x2d69)[0x8059f99]
/lib/i386-linux-gnu/libpthread.so.0(+0x6295)[0xb7303295]
/lib/i386-linux-gnu/libc.so.6(clone+0x6e)[0xb703aeee]

Ubuntu 16.04, 32 bit
cmake version 3.5.1
GNU make 4.1
/usr/bin/cc and /usr/bin/c++ are standard Ubuntu compilers version 5.4.0

Thanks, Yuri

sherm1, 2017-06-05:
Strange. `glutInit()` is called first in `main()` of simbody-visualizer -- you can see it [here](https://github.com/simbody/simbody/blob/master/Simbody/Visualizer/simbody-visualizer/simbody-visualizer.cpp#L2731).

The only unusual thing I see here is the 32-bit Ubuntu. I suspect it has been a while since any of us ran anything but 64 bit. Has anyone built 32 bit successfully recently?

chrisdembia, 2017-06-05:
> The only unusual thing I see here is the 32-bit Ubuntu. I suspect it has been a while since any of us ran anything but 64 bit. Has anyone built 32 bit successfully recently?

I have never tried Simbody on 32-bit Ubuntu.